EmixamPP/linux-enable-ir-emitter #1: Service initialization failed with `Unable to open a file descriptor for /dev/video2`

m4rtins, 2020-10-27:
This guide worked perfectly, so thank you! 

I have one issue left, though: on boot, the service fails to start with the message 
`Unable to open a file descriptor for /dev/video2`.
I guess this is due to `enable-ir-emitter.service` starting before the webcam is properly detected by the system. I am unsure how to keep the service from starting, until the webcam is detected, as my technical knowledge of systemd is fairly limited. I tried adding
`after=systemd-udev-settle.service", but this did not solve the issue.
Interestingly, the IR-Blaster sometimes still works fine after a reboot when logging in in sddm, but not reliably.

EmixamPP, 2020-10-27:
Try by adding the field "Requires". Respecting the order of "Requires" before "After".
[Unit]
Description=enable ir emitter
Requires=systemd-udev-settle.service
After=systemd-udev-settle.service

m4rtins, 2020-10-27:
I am using Manjaro and according to [this](https://www.freedesktop.org/software/systemd/man/systemd-udev-settle.service.html), the usage of udev-settle is not recommended anyway. I have tried to use udev-rules in the meantime following the first answer in [this question](https://serverfault.com/questions/916179/start-webcam-app-with-systemd-on-linux). It did not work as well, but adding ´Requires=...´ seems to have done the trick. 

So in total, I have a file webcam.rules in /etc/udev/rules.d:
```
KERNEL=="video2", SYMLINK="video2", TAG+="systemd"
```

And enable-ir-emitter.service reads:
```
`[Unit]
Description=enable ir emitter
Requires=dev-video2.device
After=dev-video2.device

[Service]
ExecStart=/usr/local/bin/enable-ir-emitter

[Install]
WantedBy=multi-user.target suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target`
```

m4rtins, 2020-10-29:
One additional Note: the IR-Blaster also did not work upon resume from hibernate, as the service was applied during hibernation, but not on resume. Adding the corresponding services to After fixed this.

m4rtins, 2020-10-29:
Issue was solved (Sorry for the double closure)

closed 2020-10-29 by m4rtins

EmixamPP, 2020-10-29:
@m4rtins thank you for this remark, I will add the line 'After=multi-user.target suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target` to the file.