EmixamPP/linux-enable-ir-emitter #2: "launch improvement of the recognition of Windows Hello" not found

orclex, 2021-01-05:
I'd like to improve to config for webcam on Lenovo X1C7. I've installed virtualbox and Windows 10 20.04 as guest and captured USB traffic.
Unfortunately I do not find step 8):

> In the VM, go in the security options tab of the Windows settings and click on launch an improvement of the recognition of Windows Hello. A pop up is displayed and asks if we want to start the test, you can go back to Wireshark without having to go any further.

Maybe it's because I cannot setup Windows Hello Face? 
[image]

This webcam should be compatible to Windows Hello but I think it's the problem with the driver from virtual box...
How did you solve this?

However, I could capture some USB traffic:

```
Setup Data
    bmRequestType: 0x21
    bRequest: 1
    wValue: 0x0200
    wIndex: 1 (0x0001)
    wLength: 48
    Data Fragment: 01000207151605000000000000000000200000600900f40b0000c0e1e400070000000000000000000000000000000000

```
but I'm not sure if these are the default values and they will change if I launch an **improvement of the recognition of Windows Hello**?

EmixamPP, 2021-01-05:
I think it's unlikely that Windows will send a packet to the camera for its infrared part if Windows Hello says it can't find compatible hardware. I also think that the problem comes from VirtualBox and not from your computer since they are designed to run on Windows. Personally I used VMWare https://www.vmware.com/products/workstation-player/workstation-player-evaluation.html (free for non-commercial use) it directly recognized my camera with a popup that proposes me to connect it to the VM.

But did you try the script without changing the parameters ? This script uses the settings for my Lenovo T15, so our computers are quite close.  Or look in the `config.txt` file, there is a configuration for "older" Lenovo.

orclex, 2021-02-01:
Now I tried capturing USB data from VMware VM with Windows 10.

I changed:

```
    __u8 data[9] = {0x01,0x03,0x01,0x00,0x00,0x00,0x00,0x00,0x00};
    .unit = 0x0a, //2 first symbols of wIndex
```

I can compile and start it. But how does I know that it is doing what it should do?

EmixamPP, 2021-02-01:
Um, well when you want to use your infrared camera (not the standard one), you have the emitter that should be blinking. Step 13 which says: "If you don't have an error code, try if it works with ffplay /dev/videoX", normally you are supposed to see yourself, even in low light and the image will blink because the emitter is flashing. 

I hope this will help you. Are you going to use it with Howdy? If yes, don't forget to set the dark_threshold to 100.

orclex, 2021-02-02:
Well I have the standard camera on _/dev/video0_ and the infrared camera on _/dev/video2_.
_lsusb_ finds this camera:

`Bus 001 Device 002: ID 04f2:b67c Chicony Electronics Co., Ltd Integrated Camera`

I also used your filter in wireshark (is this correct for my camera?):
`usb.transfer_type == 0x02 && usb.bmRequestType == 0x21`

In VMware red led for IR is flashing on accessing Windows Hello camera.

In wireshark I captured many packages with a bigger data fragment so I think these are not the correct ones.
I also captured a control package and I changed above values, compiled and run (no status output?).
I can start `ffmepg /dev/video2` and I can see myself in very low light with the red led turned on (_no flashing_?!).

Looks good so far?

But with `howdy test` or `howdy add` there is no red led activated - only the main camera is running.
Is something wrong?

Thanks!

EmixamPP, 2021-02-02:
In howdy config don't forget to set "device_path" to "/dev/video2" but I guess you did ?

Think about maybe restart your pc to see if it's flashing. Maybe the Windows instruction conflicted with the Linux one, it already happened to me when I was looking for the right package. Then if it works without flashing ...

And note that if you put the computer in standby, restart etc..., the instruction to turn on the emitter at the same time as the camera is forgotten. So either execute it by hand each time, or look at the "Launch the script at each startup (with systemd)" part of my tutorial.

By the way, can you give the reference of your pc ? I've seen a lot of people with a Chicony camera use these settings (see `config.txt`):
__u8 data[2] = {0x02, 0x19};
.selector = 0x0e,
.size = 2,

orclex, 2021-02-03:
You're right, I had _/dev/video0_ in howdy config - didn't know that it needs the IR camera instead of the normal camera.
But it also does not work. On starting `howdy test` I can see the red led for one second, then it's turned off. I got the same for `howdy add` or `sudo su`. The first one registers my face but with sudo the red led turns on for a second, then turns off while the led for normal camera keeps turned on for a few more seconds. Then howdy breaks with `Face detection timeout reached`....
`howdy test` shows a very dark image after turning off red IR led. I think this is why it cannot detect my face...

I have a **Lenovo X1 Carbon Gen7**.

I also tried your config but the binary fails to start and I tried the other configuration with
```
__u8 data[2] = {0x02, 0x19};
.selector = 0x0e,
.size = 2,
``` 
The binary starts and I have the same result as with my data values (IR turnes on for one second but then turns off again).

EmixamPP, 2021-02-03:
In howdy config you have :
```
# The number of seconds to search before timing out
timeout = 4
```
Look if the value of timeout is superior to 1, for me it's on 4

orclex, 2021-02-03:
unfortunately it already is set to 4...

EmixamPP, 2021-02-03:
Try with a 2 instead of a 1 in your `__u8 data [9] = {0x01,0x03,0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00};` !?

For me, it's that you haven't found the right instruction yet, but I'm not sure of anything :/ 
What's your distro ?

EmixamPP, 2021-02-25:
I saw that you opened a door on Howdy's repo, I put it here as a shortcut link in case it helps: boltgolt/howdy#439.

EmixamPP, 2021-05-03:
Today I published a python script that tries to automatically setup your emitter. Have a look at the readme, we never know !

closed 2021-05-27 by EmixamPP