EmixamPP/linux-enable-ir-emitter #9: Can you please  help me? I am a complete noob when it comes to cameras

winux1, 2021-05-09:
Hello!
I am using a Dell Inspiron 15 5567 laptop with a windows hello compatible camera. I would like to make it work on linux (Arch Linux or Ubuntu) I followed every step in your guide and I don't get any errors but my IR emitter not turning on. Here is the output of `lsusb`:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0cf3:e005 Qualcomm Atheros Communications 
Bus 001 Device 004: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 003: ID 0bda:58c2 Realtek Semiconductor Corp. Integrated_Webcam_HD
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Can you please help me?

Thanks for your help in advance!

EmixamPP, 2021-05-09:
Hello, 
I guess your infrared camera is on /dev/video1 ? 

Before going any further, I would advise you to use the automatic configuration script. It's unlikely to work, as your camera doesn't seem to be the same as the ones people have shared, as they always had their infrared camera on /dev/video2.

So you will have to modify the file `auto-enable-ir-emitter.c`. At line 75, change /dev/video2 to /dev/video1, save and run `cd auto && python auto-config.py`, refer to the readme. 

If none of the configurations that the script will test for you work, we will try to do the manipulations in wireshark and the windows vm together. 

One last question (a bit silly sorry), are you sure your camera (visible by your Linux kernel at least) is infrared? If you turn on your camera (e.g. with ffplay /dev/video1 or 2), the image is in grayscale ?

winux1, 2021-05-09:
Thanks for helping me! I don't know witch one of the two is my IR camera but only /dev/video0 works with ffplay. If I use howdy the image is grayscale but with ffplay it is colourised.
I tried to run the auto  script and I got this output:
/dev/video0:
```
Compilation of the C script ...
Trying all know infrared camera configuration ...
Configuration #0 ...
Ioctl error code: -1, errno: 2
The device does not support the given control or the specified extension unit could not be found.
Configuration #0 does not work
Configuration #1 ...
Ioctl error code: -1, errno: 2
The device does not support the given control or the specified extension unit could not be found.
Configuration #1 does not work
Configuration #2 ...
Ioctl error code: -1, errno: 2
The device does not support the given control or the specified extension unit could not be found.
Configuration #2 does not work
Configuration #3 ...
Ioctl error code: -1, errno: 2
The device does not support the given control or the specified extension unit could not be found.
Configuration #3 does not work
```
/dev/video1:
```
Compilation of the C script ...
Trying all know infrared camera configuration ...
Configuration #0 ...
Ioctl error code: -1, errno: 25
Configuration #0 does not work
Configuration #1 ...
Ioctl error code: -1, errno: 25
Configuration #1 does not work
Configuration #2 ...
Ioctl error code: -1, errno: 25
Configuration #2 does not work
Configuration #3 ...
Ioctl error code: -1, errno: 25
Configuration #3 does not work
```
If we have to do the manipulation with wireshark do I need to use a windows wm or I am good with dual-boot?

EmixamPP, 2021-05-09:
"but with ffplay it is colourised" bad news, I think it's not a infrerad camera. 
"I am good with dual-boot" yes of course ! 

And I presume that `ffplay /dev/video1` exit with "Inappropriate ioctl for device" ? 
Do you see the infrared emitter flashing under Windows ? 

Unfortunately, I think your kernel does not support your infrared camera ... 
Unless drivers are provided by the camera manufacturer for Linux, this is beyond my skills. 
Are you running kernel 5.12 ? This is the latest, so your camera may be taken over.

winux1, 2021-05-09:
Yes ffplay  exits with the exact error. I see the IR led flashing under windows I am running kernel 5.8 but now I am running an update to 5.12.2

winux1, 2021-05-09:
Yes ffplay /dev/video1 exits with  Inappropriate ioctl for device. Under windows the IR emitter is flashing. I am on kernel 5.12.2-051202-generic

EmixamPP, 2021-05-09:
What makes me think that your infrared camera is not recognised is that people usually have their normal camera on /dev/video0 and 1, and their infrared camera on 2 and 3. (In each case the camera usage file is the first of the two).

Eventually your camera has the possibility to be used in both modes. 

I'm not an expert on infrared cameras, but I'd be surprised if the facial recognition would work on a standard rgb colour image. Because technically it's the corpses with your infrared emission that are supposed to be more visible than the rest of the background. 

Would you show what the command `v4l2-ctl -d /dev/video0 --list-formats-ext` returns?

<details>
<summary>For example my standard camera return that</summary>

```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
```            
</details>

<details>
<summary>And my infrared camera return that</summary>

```   
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 640x360
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
```   
</details>

winux1, 2021-05-09:
Thanks for the explanation. 
For me `v4l2-ctl -d /dev/video0 --list-formats-ext` returns this:
```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 340x340
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 340x340
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
```
And `v4l2-ctl -d /dev/video1 --list-formats-ext` returns this:
```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```

EmixamPP, 2021-05-11:
Have you tested Ubuntu? Some time ago Lenovo released a communication saying that they would fully support this distro (the infrared emitter is not yet part of it, I had discussed it with a project dev). 

I don't think it will make a difference, but you never know.

winux1, 2021-05-12:
Thanks for the advice. I am currently running Ubuntu no difference so far unfortunately.

EmixamPP, 2021-09-16:
Hello ! Still no /dev/video2 device recognized by the kernel ? 

I am working on the  v3.0.0 on the dev branch. It could be that your infrared emitter is on /dev/video0. And this new version doesn't need Windows or anything else. If an instruction exists to turn on the emitter on /dev/video0, it will be found! If you want to test :
1. clone the git
2. checkout on dev
3. execute `sudo linux-enable-ir-emitter configure` and look at the infrared emitter and answer the questions asked in the terminal

EmixamPP, 2021-09-16:
(I've edited my previous message)

winux1, 2021-09-20:
Hi,
I still don’t have /dev/video2 I tried the development script. The only difference is that the white light started blinking but the IR emitter still not

Do you have any new idea?

EmixamPP, 2021-09-20:
Can you past the output of `sudo linux-enable-ir-emitter configure -d /dev/video0 --verbose` ?

EmixamPP, 2021-09-22:
The 3.1.0 have been released can you test this ? `sudo linux-enable-ir-emitter -v -d /dev/video0 configure`

winux1, 2021-09-23:
Hi,
Sorry for the late reply!
Here is a section of the output:
```bash
DEBUG: control: [77, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Ioctl error code: -1, errno: 84
Illegal byte sequence.
```
I hope we can figure out something together!

EmixamPP, 2021-09-23:
Can you test again with the version 3.1.1 (it has just been released) and paste the ouput of `linux-enable-ir-emitter -v -d /dev/video0 configure`. The option `-v` provide more information in order to understand your situation.

winux1, 2021-09-24:
No success so far. Here is the output:
```
sudo linux-enable-ir-emitter -v -d /dev/video0 configure
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
INFO: Warning to do not kill the processus !
DEBUG: unit: 4, selector: 10, curr control: ['0', '8', '0', '187', '187', '187', '187', '187'], max control: ['255', '255', '255', '255', '255', '255', '255', '255'], res control: ['1', '0', '0', '0', '0', '0', '0', '0']
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [1, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [2, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [3, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [4, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [5, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 7, selector: 2, curr control: ['1'], max control: ['6'], res control: ['1']
DEBUG: control: [2]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [3]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [4]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [5]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [6]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 7, selector: 3, curr control: ['0'], max control: ['5'], res control: ['1']
DEBUG: control: [1]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [2]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [3]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [4]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [5]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 7, selector: 4, curr control: ['0', '0'], max control: ['8', '0'], res control: ['4', '0']
DEBUG: control: [4, 0]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: [8, 0]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
ERROR: The configuration has failed.
INFO: Do not hesitate to open an issue on GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter
```

EmixamPP, 2021-09-24:
Unfortunately, I am unable to help you more. The software has tested everything possible for /dev/video0.

salah7670, 2021-12-24:
Facing the same issue on Ubuntu 20.04 Inspiron 5378. Any updates?

winux1, 2021-12-26:
Unfortunetly the camera is still not recognized properly by the linux kernel

EmixamPP, 2022-06-20:
The v4 is out for all distros, it improves the research algorithm. So you can still try it if you want.

winux1, 2022-06-21:
I will try it out as soon as possible

sarvasana, 2022-06-30:
Since v4 the emitter on my lenovo yoga slim 7 is no longer working.

winux1, 2022-07-04:
With v4 I don't even see the white LED blinking unfortunately

EmixamPP, 2023-06-21:
Closing because we can't do anything until this infrared camera is recognized by the kernel :confused:

MrWinux, 2023-10-29:
Hi,
I decided to revisit tis issue (I lost my original account). Today I was able to get my camera to work under vmware by installing [this driver](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=tn71g&lwp=rt) under windows so I captured some packets while setting up the face recognition. Could you please take a look at it? 
[camera.pcapng.zip](https://github.com/EmixamPP/linux-enable-ir-emitter/files/13198747/camera.pcapng.zip)
I also found this ancient article (https://petermortimer.de/dell-xps-ir-camera.html)  which suggests that the camere is recognised as `/dev/video1` on some dell machines but I wasn't able to get a raw image.

EmixamPP, 2023-11-04:
Hi, I'm pretty sure that no camera is on /dev/video1 considering your previous comment:

> Thanks for the explanation. For me `v4l2-ctl -d /dev/video0 --list-formats-ext` returns this:
>
> ```
> ioctl: VIDIOC_ENUM_FMT
> 	Type: Video Capture
>
> 	[0]: 'MJPG' (Motion-JPEG, compressed)
> 		Size: [...]

MrWinux, 2023-11-04:
That makes sense. I still find it interesting that the resolution of the infrared camera is listed in the output for `dev/video0`:
```
Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
```
Do you have any idea what I could try? Also were you able to take a look at the capture I made?

Thanks for your help in advance!

EmixamPP, 2023-11-09:
I will try to take a look soon

EmixamPP, 2023-12-25:
Hey, please try the v6 #154: `sudo linux-enable-ir-emitter -d /dev/video0 --width 640 --height 480 configure` :crossed_fingers:

MrWinux, 2023-12-25:
Merry Christmas @EmixamPP!

Unfortunately I had no luck. I still see myself in color and the white LED is blinking.

EmixamPP, 2023-12-30:
And is the resolution of the window is 640x480px?
You can execute `linux-enable-ir-emitter -d /dev/video0 --width 640 --height 480 test`

EmixamPP, 2023-12-30:
Maybe also try 640x360

EmixamPP, 2024-01-17:
Did you find time to test? (uninstall (see master branch readme) and reinstall the latest v6 version)

EmixamPP, 2024-02-02:
Closing the issue for inactivity, but please feel free to comment to reopen it 😄

closed 2024-02-02 by EmixamPP

MrWinux, 2026-02-14:
Hi,
My part number is 0JCXG0
https://www.parts-people.com/index.php?action=item&id=23458

SeeleVolleri, 2026-02-19:
Hi, you can try that patch out with `sudo v4l2-ctl -d /dev/video0 --set-ctrl 'band_stop_filter=0'`
Please remember that [PID need you manually edit](https://github.com/SeeleVolleri/research_0jcxg0/blob/ce6300e3eed88f6ea20fc06ce27d2515e6c1edfa/0002-media-uvcvideo-add-Dell-0JCXG0-IR-camera-device-iden.patch#L39).

MrWinux, 2026-02-20:
I just created this little toolset using your research: https://github.com/MrWinux/rtk-ir-tools
It works erfectly with Howdy