EmixamPP/linux-enable-ir-emitter #11: IR emitters work but keep red and get small frame. As windows hello, it should blink

BackMountainDevil, 2021-06-03:
# Start
## general

- Device: XiaoXinPro-13ARE 2020
- Situation: video0 work without IR. video2 work with IR keep red.
The python script auto-config.py not work. I have read your README and run script. The last part I have not tried but I will in your way or other way.

The device name and path I got in windwos 10's Device Manager.
- Integrated Camera   USB\VID_04F2&PID_B67C&MI_00\7&26fcf372&1&0000
- Integrated IR Camera  USB\VID_04F2&PID_B67C&MI_02\7&26FCF372&1&0002

### work

- /dev/video0: max 1280x720 30fps. White light always on
- /dev/video2: max 640x360 30 fps. Both white light and red light always on.

we used to talk about this in [hodwy IR emitters not working on Dell Inspiron 5567 #543](https://github.com/boltgolt/howdy/issues/543)

### unwork

- /dev/video1
[video4linux2,v4l2 @ 0x7fe514000c80] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device /dev/video1: Inappropriate ioctl for device

- /dev/video3
/dev/video1: Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x7fc310000c80] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device /dev/video3: Inappropriate ioctl for device

## detail bash output
### show all camera

```bash
$ v4l2-ctl --list-devices
Integrated Camera: Integrated C (usb-0000:03:00.4-2):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
```

### ffplay test

/dev/video0 and /dev/video2 play well. Other fail to open.

#### arch

```bash
$ ffplay /dev/video0
ffplay version n4.4 Copyright (c) 2003-2021 the FFmpeg developers
  built with gcc 10.2.0 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-shared --enable-version3
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
Input #0, video4linux2,v4l2, from '/dev/video0':B sq=    0B f=0/0   
  Duration: N/A, start: 5144.686043, bitrate: 147456 kb/s
  Stream #0:0: Video: rawvideo (YUY2 / 0x32595559), yuyv422, 1280x720, 147456 kb/s, 10 fps, 10 tbr, 1000k tbn, 1000k tbc
Switch subtitle stream from #-1 to #-1 vq=    0KB sq=    0B f=0/0   
5147.67 M-V:  0.000 fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0

$ ffplay /dev/video1
ffplay version n4.4 Copyright (c) 2003-2021 the FFmpeg developers
  built with gcc 10.2.0 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-shared --enable-version3
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
[... 233 lines cut ...]
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.067s (15.000 fps)

$ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

$ v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 640x360
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)

$ v4l2-ctl -d /dev/video3 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture
```

# End

EmixamPP, 2021-06-03:
Wow, this is a very complete issue.

(First of all, your frame is smaller with /dev/video2 is normal because the resolution of your infrared camera does not go any higher. But it's really not a problem. Rare are the ir webcam embedded in full hd.)

The fact that `auto-config.py` doesn't find a configuration for your camera shows that nobody has shared with me a configuration that works with your computer. 

(Just in case you could give me the output of the program?)

So you'll have to follow the `Setting up the script` part of the tutorial. This will help you set up a script to make your infrared emitter work properly (like under Windows). Since you have a dual boot, you don't have to go through a VM as I say in the tutorial, you just have to install Wireshark for Windows. But remember to write down everything that wireshark shows you so you don't have to go back. Or save the results of Wireshark and reopen it in Wireshark for Linux.  

Don't hesitate to ask me for more information or help.

EmixamPP, 2021-06-05:
Today someone has just shared a new configuration. Try the automatic script again, we never know.

BackMountainDevil, 2021-06-05:
Thanks for `supdrewin`' config, which does make it work. Now it can blink.    
And Now it comes few problems.
- set autostart ok but fail in fact

If I start the service, it does work.

# Detail

Then I follw the script to make it autostart. For test, I shutdown and start my pc, test with `ffplay /dev/video2`. No blink or  red. Start the `auto-config.py` again then it blink. So do the same to autostart the IR emitter. Shutdown and start my pc again, the same result and do the same  step. Shutdown and start my pc again, no blink no red.  

```bash
$ sudo systemctl status  enable-ir-emitter
× enable-ir-emitter.service - enable ir emitter
     Loaded: loaded (/etc/systemd/system/enable-ir-emitter.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sat 2021-06-05 22:16:47 CST; 2min 37s ago
    Process: 384 ExecStart=/usr/local/bin/enable-ir-emitter -dataSize 9 -data 0x01 0x03 0x02 0x00 0x00 0x00 0x00 0x00 0x00 -unit 0x0a -selector 0x06 (code=exited, status=>
   Main PID: 384 (code=exited, status=1/FAILURE)
        CPU: 2ms

6月 05 22:16:47 arch systemd[1]: Started enable ir emitter.
6月 05 22:16:47 arch enable-ir-emitter[384]: Unable to open a file descriptor for /dev/video2
6月 05 22:16:47 arch systemd[1]: enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE

[kearney@arch system]$ sudo systemctl start  enable-ir-emitter
[kearney@arch system]$ sudo systemctl status  enable-ir-emitter
○ enable-ir-emitter.service - enable ir emitter
     Loaded: loaded (/etc/systemd/system/enable-ir-emitter.service; enabled; vendor preset: disabled)
     Active: inactive (dead) since Sat 2021-06-05 22:19:45 CST; 1s ago
    Process: 1713 ExecStart=/usr/local/bin/enable-ir-emitter -dataSize 9 -data 0x01 0x03 0x02 0x00 0x00 0x00 0x00 0x00 0x00 -unit 0x0a -selector 0x06 (code=exited, status>
   Main PID: 1713 (code=exited, status=0/SUCCESS)
        CPU: 1ms

6月 05 22:19:45 arch systemd[1]: Started enable ir emitter.
6月 05 22:19:45 arch systemd[1]: enable-ir-emitter.service: Deactivated successfully.
lines 1-9/9 (END)...skipping...
○ enable-ir-emitter.service - enable ir emitter
     Loaded: loaded (/etc/systemd/system/enable-ir-emitter.service; enabled; vendor preset: disabled)
     Active: inactive (dead) since Sat 2021-06-05 22:19:45 CST; 1s ago
    Process: 1713 ExecStart=/usr/local/bin/enable-ir-emitter -dataSize 9 -data 0x01 0x03 0x02 0x00 0x00 0x00 0x00 0x00 0x00 -unit 0x0a -selector 0x06 (code=exited, status>
   Main PID: 1713 (code=exited, status=0/SUCCESS)
        CPU: 1ms

6月 05 22:19:45 arch systemd[1]: Started enable ir emitter.
6月 05 22:19:45 arch systemd[1]: enable-ir-emitter.service: Deactivated successfully.
```
Now the IR blink red. If shutdown, it need to be started by myself. I look through the auto-config.py and find the function `def create_systemd(command)`. 

```python
    system("sudo cp enable-ir-emitter /usr/local/bin")
    system("sudo cp enable-ir-emitter.service /etc/systemd/system")
    system("sudo systemctl enable --now enable-ir-emitter")
```

I have checked the file existed in where you them be.

BackMountainDevil, 2021-06-05:
Thanks for `supdrewin`' config, which does make it work. Now it can blink.    
And Now it comes few problems.
- set autostart ok but fail in fact

If I start the service, it does work.

# Detail

Then I follw the script to make it autostart. For test, I shutdown and start my pc, test with `ffplay /dev/video2`. No blink or  red. Start the `auto-config.py` again then it blink. So do the same to autostart the IR emitter. Shutdown and start my pc again, the same result and do the same  step. Shutdown and start my pc again, no blink no red.  

```bash
$ sudo systemctl status  enable-ir-emitter
× enable-ir-emitter.service - enable ir emitter
     Loaded: loaded (/etc/systemd/system/enable-ir-emitter.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sat 2021-06-05 22:16:47 CST; 2min 37s ago
    Process: 384 ExecStart=/usr/local/bin/enable-ir-emitter -dataSize 9 -data 0x01 0x03 0x02 0x00 0x00 0x00 0x00 0x00 0x00 -unit 0x0a -selector 0x06 (code=exited, status=>
   Main PID: 384 (code=exited, status=1/FAILURE)
        CPU: 2ms

6月 05 22:16:47 arch systemd[1]: Started enable ir emitter.
6月 05 22:16:47 arch enable-ir-emitter[384]: Unable to open a file descriptor for /dev/video2
6月 05 22:16:47 arch systemd[1]: enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE

[kearney@arch system]$ sudo systemctl start  enable-ir-emitter
[kearney@arch system]$ sudo systemctl status  enable-ir-emitter
○ enable-ir-emitter.service - enable ir emitter
     Loaded: loaded (/etc/systemd/system/enable-ir-emitter.service; enabled; vendor preset: disabled)
     Active: inactive (dead) since Sat 2021-06-05 22:19:45 CST; 1s ago
    Process: 1713 ExecStart=/usr/local/bin/enable-ir-emitter -dataSize 9 -data 0x01 0x03 0x02 0x00 0x00 0x00 0x00 0x00 0x00 -unit 0x0a -selector 0x06 (code=exited, status>
   Main PID: 1713 (code=exited, status=0/SUCCESS)
        CPU: 1ms

6月 05 22:19:45 arch systemd[1]: Started enable ir emitter.
6月 05 22:19:45 arch systemd[1]: enable-ir-emitter.service: Deactivated successfully.
lines 1-9/9 (END)...skipping...
○ enable-ir-emitter.service - enable ir emitter
     Loaded: loaded (/etc/systemd/system/enable-ir-emitter.service; enabled; vendor preset: disabled)
     Active: inactive (dead) since Sat 2021-06-05 22:19:45 CST; 1s ago
    Process: 1713 ExecStart=/usr/local/bin/enable-ir-emitter -dataSize 9 -data 0x01 0x03 0x02 0x00 0x00 0x00 0x00 0x00 0x00 -unit 0x0a -selector 0x06 (code=exited, status>
   Main PID: 1713 (code=exited, status=0/SUCCESS)
        CPU: 1ms

6月 05 22:19:45 arch systemd[1]: Started enable ir emitter.
6月 05 22:19:45 arch systemd[1]: enable-ir-emitter.service: Deactivated successfully.
```
Now the IR blink red. If shutdown, it need to be started by myself. I look through the auto-config.py and find the function `def create_systemd(command)`. 

```python
    system("sudo cp enable-ir-emitter /usr/local/bin")
    system("sudo cp enable-ir-emitter.service /etc/systemd/system")
    system("sudo systemctl enable --now enable-ir-emitter")
```

I have checked the file existed in where you them be. 

# solution

I look through the file in `/etc/systemd/system/`  and then edit `/etc/systemd/system/enable-ir-emitter.service` to add a new line. reboot and log in  then test fine except in user log in

```txt
[Service]
Type=one-shot
ExecStart=/usr/local/bin/enable-ir-emitter -dataSize 9 -data 0x01 0x03 0x02 0x00 0x00 0x00 0x00 0x00 0x00 -unit 0x0a -selector 0x06
Restart=always
```

EmixamPP, 2021-06-05:
From what I understand, it is at startup, the script fails because it is not given access to the camera. 
Can you try this : https://github.com/EmixamPP/linux-enable-ir-emitter/issues/1#issuecomment-717458823

supdrewin, 2021-06-05:
It's recommend to try following steps:
1. Disable the previous service
`sudo systemctl stop enable-ir-emitter`
`sudo systemctl disable enable-ir-emitter`
2. Clear the old files
`sudo rm /etc/systemd/system/enable-ir-emitter.service`
`sudo rm /usr/local/bin/enable-ir-emitter`
3. Receive the sources
`rm -r Path/To/Git/Clone/Directory`
`git clone https://github.com/EmixamPP/linux-enable-ir-emitter.git` 
4. Then build it again
`cd ./linux-enable-ir-emitter/auto`
`python3 auto-config.py`
_then press 'y' twice_
Finally, use `sudo howdy test` to try if the config correctly work, and then shutdown the machine and remove power adapter.
When you restart and reaching the login screen, check if the emitter blinking.

_noticed that chicony-ir-toggle should be completely remove if it has ever been installed_

BackMountainDevil, 2021-06-06:
@supdrewin 's steps woork.   
It seems that I use the old config build it. Thank you guys.

closed 2021-06-06 by BackMountainDevil