EmixamPP/linux-enable-ir-emitter #18: Lenovo X1 Carbon Gen 9

giacomoi, 2021-06-10:
I followed the instructions to get the right parameters for a new laptop (VM + Wireshark) and found a set of parameters that work with enable-ir-emitter (i.e., that do not give any error) on my new Lenovo X1 Carbon, Gen. 9.
.
However, the IR emitters still do not work, even after running './enable-ir-emitter' or 'sudo ./enable-ir-emitter'
I also tried setting the 'frame_width = 340' and 'frame_height = 340' in the howdy config file, but I suspect the IR camera forces the values back to 640x360.

The confic values I found are:
- data: 0xff 0x00 0x00 0x00 0x00 0x00 0x00 0x00
  unit: '0x0a'
  selector: '0x0a'

EmixamPP, 2021-06-10:
I think you may have found the instruction to disable de ir blaster. If so, try to find something that resembles your data, like different from one value and always on 0x0a and 0x0a.

Have you continued to look for other configurations that work without errors ?

And, have you try the `auto-config.py` script that I mention in the readme ?

giacomoi, 2021-06-10:
I tried at least another dozen data buffers with 0x0a + 0x0a but with no success. 
I'll try more, perhaps using the auto-config.py script, and post updates here. Thanks for the feedback!

EmixamPP, 2021-06-10:
Maybe it's not 0x0a, 0x0a.

To reduce the number of possibilities, do as I said in the tutorial, just open the Windows Hello dialog, without launching the facial recognition training.

giacomoi, 2021-06-10:
Thanks! That helped and I found the right configuration! Now it works flawlessly.

The configuration for the Lenovo X1 Carbon Gen9 is:
- data: 0x01 0x03 0x02 0x00 0x00 0x00 0x00 0x00 0x00
  unit: '0x0e'
  selector: '0x06'

Please feel free to add it to your auto/config.yaml and/or config.txt

closed 2021-06-11 by EmixamPP