EmixamPP/linux-enable-ir-emitter #21: New utility software

EmixamPP, 2021-07-03:
I have just published on a new branch a new utility that allows, in addition to before, to capture packets from the bus with wireshark. We just have to open the Windows VM.

@renyuneyun I noticed that your AUR package is still not up to date with the changes we discussed over 2 weeks ago. So I think in the end, it would be easier if I manage this AUR package.

EmixamPP, 2021-07-10:
The master branch has been updated !

EmixamPP, 2021-07-28:
@renyuneyun I saw that you orphaned the package and that someone updated it. Thank you for everything you've done!

komex, 2021-07-28:
Hi, @EmixamPP. If you wish, I may add you as maintainer to https://aur.archlinux.org/packages/linux-enable-ir-emitter.

renyuneyun, 2021-08-12:
@EmixamPP I got too busy recently and did not get time to keep track of everything. It's wonderful that @komex can take over it.
Thanks to both of your work!