EmixamPP/linux-enable-ir-emitter #29: AUR

EmixamPP, 2021-08-03:
> alerque - https://aur.archlinux.org/pkgbase/linux-enable-ir-emitter/
  This packaging in goofed up. It using a repository with no fixed version as it's source, and yet it doesn't have a pkgver() function. If it did it would be a VCS package, but that already exists elsewhere. This should be changed to only package tagged stable releases and use fixed source packages (whether tarballs or source builds are git archives, they need to be fixed with a checksum or specific SHA).
  
@komex, do you know how to handle it to comply with the guidelines ? 

I will also try to contact the maintainer of the "-git" package to update it  (#28 @FabioLolix).

FabioLolix, 2021-08-03:
You need to publish fixed release(s), after that is easy

komex, 2021-08-04:
Yes, you need to release a stable version. But there is a little problem with versioning. Currently in AUR `linux-enable-ir-emitter` has version `20210725.0-1`. So, you need to use this versioning style. IMHO, version with `20210804.0` will be OK. @FabioLolix, what do you think?

EmixamPP, 2021-08-04:
Or should I upgrade the package to release 2, and the version to 2.0.0? Would that work? Since the current realease is 1.

EmixamPP, 2021-08-04:
I've finished preparing the software and the PKGBUILD. Once I have your opinion on how to choose the version: I will push and produce the release. Then we'll just have to do the checksum and update AUR. Thank you again for your help !

komex, 2021-08-04:
I don't think version 2.0.0 will help, because 20210725 > 2. But 2.0 is more correct version. We may release a new version 2.0 and write a comment to force update package from 20210725.0-1 to 2.0.0.

EmixamPP, 2021-08-04:
Yes but pkgrel goes from 1 to 2. So the update system will be based on pkgrel instead of pkgver ? Or not at all ?

FabioLolix, 2021-08-04:
I'm away now so I'll be short, what you need to add is epoch=1 for making it an update

EmixamPP, 2021-08-04:
The AUR package has been updated.

closed 2021-08-04 by EmixamPP