EmixamPP/linux-enable-ir-emitter #31: No 2.0.1 tag

Eeems, 2021-08-09:
The PKGBUILD file is trying to pull source from the 2.0.1 tag, but it doesn't exist.

EmixamPP, 2021-08-09:
Yes it's normal, I haven't had time to update the AUR yet, only the PKGBUILD source. I'm going to do this very soon !

Eeems, 2021-08-09:
I run nightly builds which is how I found out. It might make more sense for your PKGBUILD in the repo to determine it's version from the git tags to allow building the latest source with the makefile.

EmixamPP, 2021-08-09:
I rushed to update the AUR, I hope everything is working properly. Sorry about that!

Normally, I would have a "-git" AUR version of this repo (which would then always be up to date and not based on the releases I publish). However, someone created it without giving me the rights to modify it. I'm waiting for an answer, otherwise in a couple of weeks I'll make a request for the package to be orphaned. Then another PKGBUILD source for this package will be available on this git.

Eeems, 2021-08-09:
When you get control over the -git version and have it working, could you ping me so I can update my nightly builds to point at it instead?

EmixamPP, 2021-08-09:
Yes, no problem! I'll leave this issue open, I'll send a new comment when it's good.

EmixamPP, 2021-08-12:
@Eeems here is the PKGBUILD you were waiting for <https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/AUR%20/git/PKGBUILD>.

Eeems, 2021-08-12:
Just updated to use it. Waiting for the build to finish and I'll let you know!

Eeems, 2021-08-12:
Looks like my concourse worker got stuck and I need to rebuild it. Might be a bit before I can validate.

Eeems, 2021-08-13:
@EmixamPP Looks like you accidentally added a space to the `AUR` folder name.

EmixamPP, 2021-08-13:
Well seen, it's merged.

Eeems, 2021-08-13:
That worked, thanks!

closed 2021-08-13 by Eeems