EmixamPP/linux-enable-ir-emitter #32: Directory for saved configuration

renyuneyun, 2021-08-12:
I was trying to get updated to how to use this software, and figured out a saved configuration file needs to be generated/created first. One way to do so is to execute `linux-enable-ir-emitter quick`.
When drilling down to the source code, it looks like the generated configuration is saved to `/usr/lib/linux-enable-ir-emitter/irConfig.yaml`.

In my experience, the `/usr/` directories are normally not modified by the user / user softwares, especially not for saving configuration files. Configuration files are usually saved under `/etc/` or `/var/lib/` (e.g. mysql/mariadb). Though I have to admit there are some exceptions, particularly for apps in early development (e.g. mycroft).

I'm not an expert in that either, and there are no mandatory rules for that. Therefore, I'm bringing this up for discussion, and wait to see if others have better suggestions.

EmixamPP, 2021-08-13:
At first, I wanted to save the configuration in the user's home. However, this posed problems with the fact that the `linux-enable-ir-emitter full` command must be run as administrator. 

I was inspired by the `howdy` package which saves directly the configuration file in its lib folder . 

And personally, I see no problem in placing this file in /etc/.

EmixamPP, 2021-08-30:
I've thought about it and I think that moving the file to `/etc` is not a good idea in our case. Because first of all, the file is not present during installation, it is generated during configurations and can be deleted. Secondly, the user/administrator is not supposed to modify it without going through the `linux-enable-ir-emitter manual` command which checks if the file has been corrupted or not (if it has been corrupted, a message is displayed).

EmixamPP, 2021-09-16:
The version on the dev branch no longer requires administrator rights. Would placing the file in the user's home be a good idea ? But then I would have to forbid running the command in administrator otherwise the home will be root's.

EmixamPP, 2021-09-21:
In fact it's not possible to use the user's home because the configuration file needs to be read during boot. So I've move the configuration file in /etc but the software continue to support the old location.

closed 2021-09-21 by EmixamPP