EmixamPP/linux-enable-ir-emitter #34: I have to run boot enable again after update

cchvuth, 2021-08-24:


EmixamPP, 2021-08-24:
After updating the software or the system ? In both cases, what is your distro ?

cchvuth, 2021-08-24:
I am using Manjaro KDE. It seems to happen after AUR updates that include linux-enable-ir-emitter. Thank you

EmixamPP, 2021-08-24:
Can you paste the return of `journalctl -u linux-enable-ir-emitter.service` ?

cchvuth, 2021-08-24:
vuth@vuth-xps159500 Linux 5.13.11-1-MANJARO x86_64 21.1.0 Pahvo
~ >>> journalctl -u linux-enable-ir-emitter.service                                                                
-- Journal begins at Mon 2021-04-05 12:08:29 CEST, ends at Tue 2021-08-24 14:58:46 CEST. --
Jul 23 18:24:51 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 23 18:24:52 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 23 18:24:52 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot acbbc7bb5a9548609169654675fd891e --
Jul 23 18:46:55 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 23 18:46:56 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 23 18:46:56 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot b8035f93ebf643f1ababb0964aae4c7e --
Jul 23 18:54:42 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 23 18:54:43 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 23 18:54:43 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
Jul 23 18:55:39 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 23 18:55:40 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 23 18:55:40 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot fe58a8d5887d4b138777ddfd453f39ae --
Jul 24 06:00:35 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 24 06:00:36 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 24 06:00:36 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
Jul 24 10:11:28 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 24 10:11:29 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 24 10:11:29 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot 4154cf6e0f1b47a9ad8291702cde9b45 --
Jul 24 16:04:09 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 24 16:04:10 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 24 16:04:10 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot cd3e60e7d5e845f98091222a19332978 --
Jul 25 04:19:32 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 25 04:19:35 vuth-xps159500 linux-enable-ir-emitter[902]: Unable to open a file descriptor for /dev/video2
Jul 25 04:19:36 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 25 04:19:36 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot 49608150adcf4b19b6b83a6c717f22c0 --
Jul 25 04:20:35 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 25 04:20:36 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 25 04:20:36 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
Jul 25 06:52:17 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 25 06:52:18 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 25 06:52:18 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot 4471d997b90c4986951d9d1ee2010da9 --
Jul 26 03:53:31 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 26 03:53:32 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 26 03:53:32 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot f442292469ca4a74aa712c045ec88fae --
Jul 26 03:55:45 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 26 03:55:46 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 26 03:55:46 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
-- Boot ce42123c04074cf1b042b97e38669c3b --
Jul 26 04:40:14 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 26 04:40:15 vuth-xps159500 linux-enable-ir-emitter[880]: Unable to open a file descriptor for /dev/video2
Jul 26 04:40:16 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 26 04:40:16 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
Jul 26 04:40:16 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Consumed 1.926s CPU time.
Jul 26 05:04:04 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 26 05:04:05 vuth-xps159500 linux-enable-ir-emitter[2611]: Unable to open a file descriptor for /dev/video2
Jul 26 05:04:06 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 26 05:04:06 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
Jul 26 05:04:06 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Consumed 1.755s CPU time.
Jul 26 06:29:32 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...
Jul 26 06:29:34 vuth-xps159500 linux-enable-ir-emitter[6686]: Unable to open a file descriptor for /dev/video2
Jul 26 06:29:36 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 26 06:29:36 vuth-xps159500 systemd[1]: Finished enable the infrared emitter.
Jul 26 06:29:36 vuth-xps159500 systemd[1]: linux-enable-ir-emitter.service: Consumed 2.829s CPU time.
Jul 26 07:22:57 vuth-xps159500 systemd[1]: Starting enable the infrared emitter...

EmixamPP, 2021-08-24:
Humm, I've no idea :/ .  It's strange because I don't remember having this problem.
And the other maintainers of the AUR package not having opened any issue about this,this leads me to think that they did not have this problem. 

I will leave this issue open, if the systemd service is disabled again; feel free to send a new comment !

cchvuth, 2021-08-24:
thank you, I'll post another comment if it happen again

cchvuth, 2021-08-28:
@EmixamPP thanks for your answer where can I find enable-ir-emitter.service?

EmixamPP, 2021-08-28:
Oh yes, sorry! I forgot that the software as it is today did not exist at the time of this issue. 

For me the problem is not with `linux-enable-ir-emitter` because of the `Unable to open a file descriptor for /dev/video2` error. This means that the program could not access the camera. 
The solution found in issue #1 ensures that the camera will be accessible when the program will be executed by systemd. I have not included this fix in the installation as the method used is not very recommended. 

So, here is the path to the file : `/usr/lib/systemd/system/linux-enable-ir-emitter.service`.

If it works for you, I think I'll include this fix automatically in my `install.bash` script.

cchvuth, 2021-08-28:
Hi unfortunately it didn't work. I added a new file called webcam.rules in /etc/udev/rules.d and edit /usr/lib/systemd/system/linux-enable-ir-emitter.service. But it didn't work

EmixamPP, 2021-08-28:
Can you try this, I have set up a restart of the service in case of an error :
```
[Unit]
Description=enable the infrared emitter
After=multi-user.target suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target

StartLimitInterval=10
StartLimitBurst=10

[Service]
Type=simple
ExecStart=/usr/bin/linux-enable-ir-emitter run

Restart=on-failure
RestartSec=1

[Install]
WantedBy=multi-user.target suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
```

cchvuth, 2021-08-28:
It is still not working. Anyways, when I try to activate IR.

vuth@vuth-xps159500 Linux 5.13.12-1-MANJARO x86_64 21.1.1 Pahvo
~ >>> sudo ls                                                                                                      
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1034) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (597) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
All frames were too dark, please check dark_threshold in config
Average darkness: [78.8554], Threshold: 50.0
Face detection image too dark
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1034) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (597) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
All frames were too dark, please check dark_threshold in config
Average darkness: [78.720924], Threshold: 50.0
Face detection image too dark
[sudo] password for vuth:

EmixamPP, 2021-08-28:
The opencv warnings are "normal", you can hide them by adding `export OPENCV_LOG_LEVEL=ERROR` to the file `~/.bashrc`.

Howdy show these message :
```
All frames were too dark, please check dark_threshold in config
Average darkness: [78.720924], Threshold: 50.0
Face detection image too dark
```
cause the camera does not detect a strong enough infrared signal, therefore the image is dark. Hence the importance of having an infrared emitter in front of our face.

Some infrared emitter are also not very efficient, so you need to change the value of `dark_threshold =` to `100` in `sudo howdy config`.

Any way, our file descriptor problem persist. Are you sure that another program using the camera does not start up? If it hogs the camera, linux-enable-ir-emitter cannot access it.

I have another question :everything works correctly if you only use (boot and reboot) Linux ?

cchvuth, 2021-08-28:
I just tried boot reboot linux, it didn't work either. so it's not related to dual boot with windows. :(. Maybe i'll try with a clean install of Manjaro

EmixamPP, 2021-08-28:
I developed this program using Manjaro, it is strange. Your distro is up to date (pacman + kernel) ? The issue #1 is about Manjaro, may be you have failed the manipulation ? I will recap in details :

Change `/usr/lib/systemd/system/linux-enable-ir-emitter.service` with :
```
[Unit]
Description=enable the infrared emitter
Requires=dev-video2.device
After=dev-video2.device multi-user.target suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target

[Service]
Type=oneshot
ExecStart=/usr/bin/linux-enable-ir-emitter run

[Install]
WantedBy=multi-user.target suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
```

Create a new file `/etc/udev/rules.d/webcam.rules`
```
KERNEL=="video2", SYMLINK="video2", TAG+="systemd"
```

Execute `sudo linux-enable-ir-emitter boot enable`

cchvuth, 2021-08-28:
It works with your last configs. Thank you so much!

EmixamPP, 2021-08-28:
Perfect ! With pleasure :smile:

EmixamPP, 2021-08-29:
@cchvuth could I ask you a favour ? Could you try changing the contents of `/usr/lib/systemd/system/linux-enable-ir-emitter.service`  with what is below ? Then rerun `sudo linux-enable-ir-emitter boot enable`. I'm trying to find a cleaner way to solve your problem and include it in the next software release. Thanks for your help!

```
[Unit]
Description=enable the infrared emitter
After=multi-user.target suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target

[Service]
Type=oneshot
ExecStartPre=/sbin/modprobe uvcvideo
ExecStart=/usr/bin/linux-enable-ir-emitter run

[Install]
WantedBy=multi-user.target suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
```

If you don't have the time (or if you don't want to do that), no problem, feel free to close issue.

cchvuth, 2021-08-29:
I tested your new config 3 times and it works. Thank you

EmixamPP, 2021-08-29:
Thank you very much for your participation !
I forgot to mention that you have to delete the file  `/etc/udev/rules.d/webcam.rules` .
If it was not done, can you try just one last time ?

cchvuth, 2021-08-29:
yup still works when deleted that rules

closed 2021-08-29 by EmixamPP