EmixamPP/linux-enable-ir-emitter #35: New release

EmixamPP, 2021-08-29:
I have published what looks like the future version of the programme on the <https://github.com/EmixamPP/linux-enable-ir-emitter/tree/pre-release> branch.

This new version currently includes :
* `linux-enable-ir-emitter fix` command : can reset the config file and uninstall chicony-ir-toggle
* I tried to fix the `unable to open a file descriptor` error caused during boot. The problem is that the camera is not yet initialized and therefore not yet accessible. I've added the line `ExecStartPre=/sbin/modprobe uvcvideo` to the .service file. Since I don't have this problem, I can't test, I hope it works.

What I might consider adding: move the config file to the `/etc` folder. 

My second announcement is that I am no longer using Manjaro and therefore I will not be able to properly maintain the AUR package. @komex will you take care of updating the package for new releases? (If so, don't worry I'll let you know when it's ready).

EmixamPP, 2021-09-01:
@komex could you add @supdrewin as maintainer of the package?
And if possible, in order to make things easier in the future:  make myself the main maintainer of the package so that I can add co-maintainers.

komex, 2021-09-01:
@EmixamPP, done

EmixamPP, 2021-09-01:
Thank you ! I have added you as a co-maintainer.

closed 2021-09-01 by EmixamPP