EmixamPP/linux-enable-ir-emitter #37: Possibility to control emitters during authentication?

boltgolt, 2021-09-02:
All Windows Hello cameras that i have come across have blinking IR emitters that do not necessarily sync with the camera framerate. Currently Howdy relies on analysing the darkness of frames to filter out these unlit frames. Filtering them out speeds up recognition as Howdy will only analyse well-lit frames that have a higher chance of success.

As you are much more knowledgable on the packets send to the camera from Windows Hello, do you think it would be possible to read the status of the emitters for each frame or even sync up this blinking with the camera?

EmixamPP, 2021-09-04:
I have the impression that the flashing of the emitter is only related to the hardware. I come to this conclusion because during the blinking no control query is send. 

However, I will do more in-depth analysis of packets returned by the camera. But I'm pretty busy for a week. 
I'll keep in touch soon as have more conclusive result.

boltgolt, 2021-09-07:
More in depth info about the camera stream would be amazing! Shame there's no way to control the camera directly

EmixamPP, 2021-09-16:
Unfortunately, I parsed, observed, tried to understand as much as I could of the packets (it's really not easy there are hundreds of them even on just 1 or 2 seconds of video capture) and I didn't find or understand anything conclusive. I have the impression that the infrared emitter is really a hardware device with almost no live interaction. 

So I undertook some research on the internet but the documentation on the subject is extremely poor. 

However, I managed to design the IR emitter configuration and found out that there is (at least on my emitter) the possibility to not flash it. And so the image is irradiated all the time. I don't know if this would be useful?

EmixamPP, 2021-09-18:
I tried to make sense of every instruction I could make on my camera, however apart from a couple of obvious ones I didn't see any visible change in brightness or recognition efficiency.   

Unless the manufacturer provides documentation I don't think it is possible to understand the handling of the infrared camera by its instructions. And again, I have the impression that this kind of hardware is only slightly programmable. (but all depends on the manufacturer)

boltgolt, 2021-09-20:
Thanks for your hard work on this! It's a shame that we can't have the control we want over devices we already own

I guess the not-great method of filtering out dark frames after capture is the way to go then. Again, thanks so much for looking into it!

closed 2021-09-20 by boltgolt