EmixamPP/linux-enable-ir-emitter #39: Your camera configuration is not in the database shared by the git community.

kapi117, 2021-09-06:
Output of `sudo linux-enable-ir-emitter full`, following the steps given [here](https://github.com/EmixamPP/linux-enable-ir-emitter/wiki/Semi-automatic-configuration)
```Please read and folow this tutorial : https://github.com/EmixamPP/linux-enable-ir-emitter/wiki/Semi-automatic-configuration. Press enter when you'r ready
The capturing is finished, make sure the camera is connected to the host os. Press enter when it's done
Ioctl error code: -1, errno: 2
The device does not support the given control or the specified extension unit could not be found.
Ioctl error code: -1, errno: 2
The device does not support the given control or the specified extension unit could not be found.
Did you see the ir emitter flashing ? Yes/No ? Yes
Your camera configuration is not in the database shared by the git community.
Could you please take 5 minutes to copy and paste the contents of 'linux-enable-ir-emitter manual' into a new issue: https://github.com/EmixamPP/linux-enable-ir-emitter/issues.
Thank you for the others !
```

Output of `sudo linux-enable-ir-emitter manual`
```
#Caution: any manual modification of this file may corrupt the operation of the>
#Please consult https://github.com/EmixamPP/linux-enable-ir-emitter/wiki/Manual>
#If you currupt the config file: execute 'linux-enable-ir-emitter fix config' t>

!!python/object:IrConfiguration.IrConfiguration
_data:
- '0x1'
- '0x0'
- '0x0'
- '0x0'
_selector: '0x9'
_unit: '0x7'
_videoPath: /dev/video2
```

EmixamPP, 2021-09-06:
Thank you for your participation in the project !

EmixamPP, 2021-09-06:
Are you sure it works? 
To be sure, put your computer into the sleep mode or reboot, then run `linux-enable-ir-emitter test`.
The emitter still blinking?

kapi117, 2021-09-06:
Yes, I've done full configuration again, put computer into the sleep mode and then run `sudo linux-enable-ir-emitter test` and the emitter is still blinking. (Btw, primarily I thought that message `your camera is not in the datebase` means that program is therefore unable to work properly).
However, `sudo howdy test` doesn't turn on the emitter.

kapi117, 2021-09-06:
The case can be that I had my emitter working properly but not with howdy (I don't know if it changes anything, but in **Cheese** application I could switch to IR camera and emitter was working properly even before using your script).

EmixamPP, 2021-09-06:
When you say it doesn't work with Howdy, what do you mean? Only the transmitter doesn't work? Because if your infrared camera works well with Cheese, then your configuration in Howdy is wrong. And indeed, it could be that your infrared emitter is working natively without needing my program. (hence the fact that what is in `linux-enable-ir-emitter manual` is not actually an instruction to turn on the emitter but something else and since my software is trying to turn on the emitter it thought it was because of this instruction).

kapi117, 2021-09-07:
Yes, perhaps it works natively. With 'doesn't work with howdy' I meant that howdy turned on only the LED indicating that camera is turned on while the emitter wasn't blinking. Now I reinstalled the Howdy and reconfigured (while installing, I'm using Ubuntu 20.04). And now only transmitter blinks, and the LED indicating camera turned on doesn't. Maybe that's fine, but in Windows Hello both are turned on and hence my uncertainty.

Sorry for confusion ;)

EmixamPP, 2021-09-11:
The `full` command uses false positives when the infrared camera is working beforehand. I'll try to find a way to detect them in order to ignore them. 

I have taken note of your remark about the message displayed, thank you!

EmixamPP, 2021-09-16:
I am working on the  v3.0.0 on the dev branch. And this new version doesn't need Windows or anything else. If an instruction exists to turn on the emitter, it will be found ! But to use this version your infrared camera should not working natively. Is this the case at this moment?

closed 2021-09-25 by EmixamPP