EmixamPP/linux-enable-ir-emitter #42: Device as Symlink

dav-is, 2021-09-21:
If I used a device which is a symlink `/dev/videohowdy -> video2 OR video6` I get this error:

```
Traceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 83, in <module>
    test.execute()
  File "/usr/lib64/linux-enable-ir-emitter/command/test.py", line 11, in execute
    exit_code = ExitCode.FAILURE if not ir_config else ir_config.triggerIr()
  File "/usr/lib64/linux-enable-ir-emitter/IrConfiguration.py", line 89, in triggerIr
    capture = cv2.VideoCapture(int(self.device[-1]))
ValueError: invalid literal for int() with base 10: 'y'
```

EmixamPP, 2021-09-22:
Unfortunately the /dev/videoX format is mandatory, because of the cv2 module of python (cv3 isn't in fedora deposit) and secondly I won't be able to know the product and vendor id of the camera.

closed 2021-09-22 by EmixamPP

dav-is, 2021-09-23:
Are you able to add support for multiple IR cameras?

EmixamPP, 2021-09-24:
Yes of course, but it will take some time because I need to think how to implement that. 

It will be released with version 3.2.0, if I can find the time it will be during the next week. If not, it will be the week after.

EmixamPP, 2021-09-25:
https://github.com/EmixamPP/linux-enable-ir-emitter/issues/46