EmixamPP/linux-enable-ir-emitter #43: Skips Adding Config Even Though It's Required

dav-is, 2021-09-21:
```
$ sudo linux-enable-ir-emitter configure 
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/videoio_registry.cpp (217) VideoBackendRegistry VIDEOIO: Enabled backends(9, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); MSMF(970); V4L2(960); CV_IMAGES(950); CV_MJPEG(940); FIREWIRE(930); UEYE(920)
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_gstreamer.cpp (831) open OpenCV | GStreamer: v4l2src device=/dev/video2 ! videoconvert ! appsink drop=true
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_gstreamer.cpp (864) open OpenCV | GStreamer: mode - MANUAL
[ WARN:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_gstreamer.cpp (1081) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? yes
ERROR: Your infrared camera is already working, skipping the configuration.

$ sudo linux-enable-ir-emitter test     
ERROR: No configuration is currently saved.
Traceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 83, in <module>
    test.execute()
  File "/usr/lib64/linux-enable-ir-emitter/command/test.py", line 12, in execute
    exitIfFileDescriptorError(exit_code, ir_config.device)
AttributeError: 'NoneType' object has no attribute 'device'
```

EmixamPP, 2021-09-21:
You need to restart your computer before cause you have uninstall and install the software.

dav-is, 2021-09-21:
That doesn't seem to work. 

If I answer `no` and then `yes` it adds the config file and works fine.

I can easily reproduce by running `sudo linux-enable-ir-emitter fix config` which deletes the config file and go through the steps gives above.

EmixamPP, 2021-09-21:
To use the configure command your infrared emitter need to don't work. So execute `limux-enable-ir-emitter boot disable` and delete the configuration file (in order to  be sure) . Put in sleep your computer then wakeup and normaly `howdy` will fail cause the ir emitter don'work. Login with your password and execute the configure command. After that every things should be fine.

EmixamPP, 2021-09-22:
If your problem emains you can stay on the 2.1.0 version ? ​`dnf install linux-enable-ir-emitter-2.1.0. I will try to investigate. But can you past the output of `linux-enable-ir-emitter configure --verborse`

EmixamPP, 2021-09-22:
I will fix these problems on the 3.1.0. The packages have been rollback to 2.1.0. So you can easily return to a stable version. Sorry for that !

dav-is, 2021-09-22:
There is no issue with 2.1.0 👍🏼 I have it completely working if I say `no` initially and then answer yes later in the setup, just seems to be a bug in the config generator

EmixamPP, 2021-09-22:
The 3.1.0 is released and should fix all problems (the compability after update too). Let me know if not.

dav-is, 2021-09-23:
I can't reproduce the issue anymore, thanks

closed 2021-09-23 by dav-is