EmixamPP/linux-enable-ir-emitter #44: The configuration has failed.

cchvuth, 2021-09-21:
- After the new update, the IR broke, so I run
- `systemctl status linux-enable-ir-emitter.service`
- I saw a message saying config file is corrupted, please run `linux-enable-ir-emitter fix config`
- I ran the command and run `linux-enable-ir-emitter configure`
- however, it said configuration has failed. I tried restarting my computer, linux-enable-ir-emitter boot disable and run config command again but still fail.
- Below is my log:

vuth@vuth-xps159500 Linux 5.14.2-1-MANJARO x86_64 21.1.3 Pahvo
~ >>> sudo linux-enable-ir-emitter configure                                                                       
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1034) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (597) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
All frames were too dark, please check dark_threshold in config
Average darkness: 0.0, Threshold: 100.0
Face detection image too dark
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1034) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (597) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
All frames were too dark, please check dark_threshold in config
Average darkness: 0.0, Threshold: 100.0
Face detection image too dark
[sudo] password for vuth: 
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1081) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
INFO: Warning to do not kill the processus !
ERROR: The configuration has failed.
INFO: Do not hesitate to open an issue on GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter
~ >>> sudo linux-enable-ir-emitter configure                                                                                                                                                                                            [1]
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1081) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
INFO: Warning to do not kill the processus !
ERROR: The configuration has failed.
INFO: Do not hesitate to open an issue on GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter

EmixamPP, 2021-09-22:
I broke the compability ... shit
Can you paste me the output of `linux-enable-ir-emitter configure --verbose`
You stay on the 2.1.0 version `dnf install linux-enable-ir-emitter-2.1.0`

cchvuth, 2021-09-22:
[sudo] password for vuth: 
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1081) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
INFO: Warning to do not kill the processus !
DEBUG: unit: 4, selector: 6, curr control: ['1', '3', '1', '0', '0', '0', '0', '0', '0'], max control: ['1', '3', '3', '0', '0', '0', '0', '0', '0']
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1081) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
DEBUG: unit: 4, selector: 6, curr control: ['1', '3', '1', '0', '0', '0', '0', '0', '0'], max control: ['1', '3', '3', '0', '0', '0', '0', '0', '0']
[ WARN:0] global /build/opencv/src/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1081) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
ERROR: The configuration has failed.
INFO: Do not hesitate to open an issue on GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter

EmixamPP, 2021-09-22:
Can you answer `yes` to the second question ?
And then execute  `linux-enable-ir-emitter test`. Does the ir emitter blink? 
(You have a Lenovo right ?)

cchvuth, 2021-09-22:
I tried and it's not blinking. I have dell xps 15 9500.

EmixamPP, 2021-09-22:
Sorry for this broken version ... 
Can your rollback on the 2.1.0 and execute the `quick` command and then paste the content of the `manual` command ? 
Thank your for your help.

EmixamPP, 2021-09-22:
@supdrewin are you able to rollback the AUR packages version to 2.0.1 ? This version seems to not work with Dell ... Thank you !

cchvuth, 2021-09-22:
yes I rolled back to 2.1.0 and it works with quick command. below is the content:

!!python/object:IrConfiguration.IrConfiguration
_data:
- '0x1'
- '0x3'
- '0x2'
- '0x0'
- '0x0'
- '0x0'
- '0x0'
- '0x0'
- '0x0'
_selector: '0x6'
_unit: '0x4'
_videoPath: /dev/video2

supdrewin, 2021-09-22:
> @supdrewin are you able to rollback the AUR packages version to 2.0.1 ? This version seems to not work with Dell ... Thank you !

OK! No problem, but the 2.1.0 works so I just rollback to 2.1.0.

EmixamPP, 2021-09-22:
Ok I see the problem ! I will fix that in the v3.1.0. Thank @cchvuth for your help ! When I ask you, can you test the 3.1.0 that will be on the dev branch ? In order to avoid other problems of this kind.

cchvuth, 2021-09-22:
yeah sure, please let me know

EmixamPP, 2021-09-22:
Ok now everything should be fine : <https://github.com/EmixamPP/linux-enable-ir-emitter/tree/dev>

Uninstall your current version, clone the git and checkout on dev, then you have to execute `sudo bash installer.sh install` in order to install this dev version. 

(you can execute `sudo bash installer.sh uninstall` to uninstall this manual installed version)

Thank you for your help !

cchvuth, 2021-09-22:
@EmixamPP  works

closed 2021-09-22 by EmixamPP

cchvuth, 2021-11-06:
@EmixamPP its not working again after the last update

EmixamPP, 2021-11-06:
What is the problem ?

cchvuth, 2021-11-06:
its not flashing IR on boot until I run linux-enable-ir-emitter test command

EmixamPP, 2021-11-06:
Have you re-executed `boot enable` ?

cchvuth, 2021-11-06:
ah yeah I just need to run it again and it works now

cchvuth, 2021-11-06:
somehow it got deactivated

EmixamPP, 2021-11-06:
You update it using AUR ?