EmixamPP/linux-enable-ir-emitter #45: Error after installation

anon1892, 2021-09-22:
Hello,

I'm on Ubuntu 20.04, and I'm unable to use your tool.
The installation process sucess (I think), and show me that : 

```bash
$ sudo bash installer.sh install
Requirement already satisfied: opencv-python in /usr/local/lib/python3.8/dist-packages (4.5.2.52)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (5.3.1)
Requirement already satisfied: numpy>=1.17.3 in /usr/local/lib/python3.8/dist-packages (from opencv-python) (1.20.3)
WARNING: Running pip as root will break packages and permissions. You should install packages reliably by using venv: https://pip.pypa.io/warnings/venv
WARNING: You are using pip version 21.1.2; however, version 21.2.4 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
make: Entering directory '/home/*****/linux-enable-ir-emitter/sources/uvc'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/*****/linux-enable-ir-emitter/sources/uvc'
install: creating directory '/usr/lib/linux-enable-ir-emitter'
install: creating directory '/usr/lib/linux-enable-ir-emitter/uvc'
'sources/uvc/get_query' -> '/usr/lib/linux-enable-ir-emitter/uvc/get_query'
'sources/uvc/len_query' -> '/usr/lib/linux-enable-ir-emitter/uvc/len_query'
'sources/uvc/set_query' -> '/usr/lib/linux-enable-ir-emitter/uvc/set_query'
'sources/uvc/execute_query.o' -> '/usr/lib/linux-enable-ir-emitter/uvc/execute_query.o'
install: creating directory '/usr/lib/linux-enable-ir-emitter/command'
'sources/command/boot.py' -> '/usr/lib/linux-enable-ir-emitter/command/boot.py'
'sources/command/configure.py' -> '/usr/lib/linux-enable-ir-emitter/command/configure.py'
'sources/command/fix.py' -> '/usr/lib/linux-enable-ir-emitter/command/fix.py'
'sources/command/__init__.py' -> '/usr/lib/linux-enable-ir-emitter/command/__init__.py'
'sources/command/manual.py' -> '/usr/lib/linux-enable-ir-emitter/command/manual.py'
'sources/command/run.py' -> '/usr/lib/linux-enable-ir-emitter/command/run.py'
'sources/command/test.py' -> '/usr/lib/linux-enable-ir-emitter/command/test.py'
'sources/globals.py' -> '/usr/lib/linux-enable-ir-emitter/globals.py'
'sources/IrConfiguration.py' -> '/usr/lib/linux-enable-ir-emitter/IrConfiguration.py'
'sources/linux-enable-ir-emitter.py' -> '/usr/lib/linux-enable-ir-emitter/linux-enable-ir-emitter.py'
'sources/linux-enable-ir-emitter.service' -> '/usr/lib/systemd/system/linux-enable-ir-emitter.service'
```

But when I try a command (`sudo linux-enable-ir-emitter -h`, or `sudo linux-enable-ir-emitter configure`), an error occur : 
```bash
Traceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 20, in <module>
    logging.basicConfig(encoding='utf-8', format='%(levelname)s: %(message)s', level=logging.INFO)
  File "/usr/lib/python3.8/logging/__init__.py", line 2009, in basicConfig
    raise ValueError('Unrecognised argument(s): %s' % keys)
ValueError: Unrecognised argument(s): encoding
```

Is it a bug in your program, or a misconfiguration in pip ? And how could I repair that ?
Thanks in advance

EmixamPP, 2021-09-22:
Yes is a problem with your Python implementation. But don't worry, I will adapt my software for this problem. 
I was just about to release version 3.1.0. 
I will send a new comment when I have pushed the commits.

EmixamPP, 2021-09-22:
The 3.1.0 is released and should fix this problem. Let me know if yes or not.

anon1892, 2021-09-23:
Hello,
Thank you very much for the fast answer. With the new version, it works... But for now I can't configure my ir emitter...
```bash
$ sudo linux-enable-ir-emitter configure
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
INFO: Warning to do not kill the processus !
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [5, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [6, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [7, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [8, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: control: [9, 8, 0, 187, 187, 187, 187, 187]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ?
```
It does that and never stop... I will try the manual configuration.

EmixamPP, 2021-09-23:
This is not for forever. You forgot to show me the first part of the output no ? 
If you reach the end you will find the right configuration. It requires patient for some device ^^

I can eventually think to add an option to skip a part of the test after many 'no'.

anon1892, 2021-09-23:
No, I sent you all the beginning of the output of this command.
I try up to `DEBUG: control: [102...`, but nothing more...
So I find some information with Wireshark, but I didn't understand how to write them in the conf file, it's quite confusing...

```bash
bmRequestType: 0x21
bRequest: SET CUR (0x01)
Control Selector: Unknown (0x0a)
Interface: 0x00
Entity: 0x04
wLength: 8
Current Value
bRequest: 1
wValue: 0x0a00
wIndex: 1024 (0x0400)
wLength: 8
Data Fragment: 019e000001000000
```

Thanks for your help

EmixamPP, 2021-09-23:
Hum there are some missing outputs the debug. I will look at that later. 
Yes, in fact you need to skip this control pattern (maximum 255 control of each pattern, so 255 'no' ha ha ha !) 
I will implement a skip system for pass a pattern after 5 or 10 'no'. I will try to publish the v3.1.1 for tomorrow. 

Your unit is 0x0a and your selector is 0x04
And the control value : 0x01 0x9e 0x00 0x00 0x01 0x00 0x00 0x00 (You can convert or not the hex value in decimal, but don't quote any value). I'm a bit busy and I forgot to update the wiki for the v3.0.0. I will tried to do that for tomorrow too ! 

I will keep you informed

EmixamPP, 2021-09-23:
But I think after these first 255 'no' you should find the right instruction.

anon1892, 2021-09-23:
Okay thanks

My configuration file look like that now : 
```
!!python/object:IrConfiguration.IrConfiguration
_control: 0x01 0x9e 0x00 0x00 0x01 0x00 0x00 0x00
_device: /dev/video2
_selector: 0x04
_unit: 0x0a
```

Now, `sudo linux-enable-ir-emitter test` didn't do any output (no more errors), but I only see the white LED, as usual, not the IR...
I will wait for the update, 255 "no" is too long 😅

EmixamPP, 2021-09-23:
Yes you need to respect the config example automatically generated but the manual command: one control index per line with an "- " before.

EmixamPP, 2021-09-23:
The 3.1.1 is out ! You will no longer need to input so many "no". If you have some problems please provide the ouput of `linux-enable-ir-emitter -v configure`. The option `-v` provide more information in order to understand your situation.

anon1892, 2021-09-25:
Thank you very much, it's working !

closed 2021-09-25 by anon1892

EmixamPP, 2021-09-25:
With pleasure ! If you like the project, don't hesitate to star the repository to support me :smile: