EmixamPP/linux-enable-ir-emitter #47: IR emitter working, but not after restart

joakimkingstrom, 2021-10-07:
Hi,

First of all a big thanks for creating this fix, I've got a Dell XPS 9510 and this fix works like a charm - until I restart the computer. After the restart it only starts to work again once I've run linux-enable-ir-emitter configure. Simply restarting the service doesn't work. I'm running Pop OS 21.04. 

Any ideas what might cause this?

Kind regards,
Joakim

EmixamPP, 2021-10-07:
Have you execute `sudo linux-enable-ir-emitter boot enable` ?

EmixamPP, 2021-10-07:
Can you please give me the following information :
https://github.com/EmixamPP/linux-enable-ir-emitter/wiki/Issues

joakimkingstrom, 2021-10-07:
Oh, sorry about that, I'd missed that! Please see below for the info:

linux-enable-ir-emitter boot status

>! joakim@pop-os:~$ sudo linux-enable-ir-emitter boot status
>! ● linux-enable-ir-emitter.service - enable the infrared emitter
>!      Loaded: loaded (/lib/systemd/system/linux-enable-ir-emitter.service; enabl>
>!      Active: inactive (dead) since Thu 2021-10-07 09:57:05 BST; 11min ago
>!     Process: 1719 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/S>
>!     Process: 1722 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, >
>!    Main PID: 1722 (code=exited, status=0/SUCCESS)
>! 
>! Oct 07 09:57:05 pop-os systemd[1]: Starting enable the infrared emitter...
>! Oct 07 09:57:05 pop-os systemd[1]: linux-enable-ir-emitter.service: Succeeded.
>! Oct 07 09:57:05 pop-os systemd[1]: Finished enable the infrared emitter.

journalctl -xeu linux-enable-ir-emitter.service

>! -- Boot 9c16e9b59c154784a8b115a095a80515 --
>! Oct 07 09:57:05 pop-os systemd[1]: Starting enable the infrared emitter...
>! ░░ Subject: A start job for unit linux-enable-ir-emitter.service has begun execution
>! ░░ Defined-By: systemd
>! ░░ Support: http://www.ubuntu.com/support
>! ░░ 
>! ░░ A start job for unit linux-enable-ir-emitter.service has begun execution.
>! ░░ 
>! ░░ The job identifier is 120.
>! Oct 07 09:57:05 pop-os systemd[1]: linux-enable-ir-emitter.service: Succeeded.
>! ░░ Subject: Unit succeeded
>! ░░ Defined-By: systemd
>! ░░ Support: http://www.ubuntu.com/support
>! ░░ 
>! ░░ The unit linux-enable-ir-emitter.service has successfully entered the 'dead' state.
>! Oct 07 09:57:05 pop-os systemd[1]: Finished enable the infrared emitter.
>! ░░ Subject: A start job for unit linux-enable-ir-emitter.service has finished successfully
>! ░░ Defined-By: systemd
>! ░░ Support: http://www.ubuntu.com/support
>! ░░ 
>! ░░ A start job for unit linux-enable-ir-emitter.service has finished successfully.
>! ░░ 
>! ░░ The job identifier is 120.

EmixamPP, 2021-10-07:
Hum that's weird, because `/sbin/modprobe uvcvideo` only ensure that the camera have been initialized by the system and therefore linux-enable-ir-emitter access it without error (if linux-enable-ir-emitter can't access to the camera, the return code is 126 https://github.com/EmixamPP/linux-enable-ir-emitter/wiki/Exit-code. Not 0 like in your case). On some distro linux-enable-ir-emitter can run proprely without execute `/sbin/modprobe uvcvideo`  which is usless but  without consequence so it's fine.

May be you can try to modify `/usr/lib/systemd/system/linux-enable-ir-emitter.service` by removing the line `ExecStartPre=/sbin/modprobe uvcvideo` ?

(by the way a markdown spoiler is that : https://gist.github.com/jbsulli/03df3cdce94ee97937ebda0ffef28287)

joakimkingstrom, 2021-10-07:
Btw, adding graphical.target or gdm3.service to the After field in the linux-enable-ir-emitter.service doesn't work either. It's as if this command needs to be run once my user is logged in.