EmixamPP/linux-enable-ir-emitter #57: Since the last version the boot enable option no longer works

logicito, 2021-11-03:
Today I tried with Ubuntu 21.10 and Fedora 35, the configuration process worked fine, IR leds activated and flashed OK, no issues.

Then activated boot enable, output was ok as well, nonetheless when booting the IR leds will never turn on, I reinstalled both distributions 2 times from scratch, same outcome.

This is very weird, because I have done this 20 times, always success at the first try, never a problem.

EmixamPP, 2021-11-03:
Can you please open an issue using the right template, in order to provide all necassary information.
You can juste copy paste the template on a new comment, no need to reopen a new issue :
<https://github.com/EmixamPP/linux-enable-ir-emitter/issues/new?assignees=EmixamPP&labels=&template=does-not-work-at-boot.md&title=>
Thank you.

logicito, 2021-11-04:
YES

**Output of `linux-enable-ir-emitter boot status`**
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/lib/systemd/system/linux-enable-ir-emitter.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Wed 2021-11-03 20:14:30 EDT; 1min 33s ago
    Process: 1550 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 1552 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 1552 (code=exited, status=0/SUCCESS)
        CPU: 1.882s

Nov 03 20:14:30 xps systemd[1]: Starting enable the infrared emitter...
Nov 03 20:14:30 xps systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Nov 03 20:14:30 xps systemd[1]: Finished enable the infrared emitter.
Nov 03 20:14:30 xps systemd[1]: linux-enable-ir-emitter.service: Consumed 1.882s CPU time.
```

**Ouput of `journalctl -xeu linux-enable-ir-emitter.service`**
<details><summary>configure command output</summary>

```
-- Journal begins at Wed 2021-11-03 19:47:41 EDT, ends at Wed 2021-11-03 20:22:54 EDT. --
Nov 03 20:12:52 xps systemd[1]: Starting enable the infrared emitter...
░░ Subject: A start job for unit linux-enable-ir-emitter.service has begun execution
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit linux-enable-ir-emitter.service has begun execution.
░░ 
░░ The job identifier is 3211.
Nov 03 20:12:52 xps systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit linux-enable-ir-emitter.service has successfully entered the 'dead' state.
Nov 03 20:12:52 xps systemd[1]: Finished enable the infrared emitter.
░░ Subject: A start job for unit linux-enable-ir-emitter.service has finished successfully
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit linux-enable-ir-emitter.service has finished successfully.
░░ 
░░ The job identifier is 3211.
Nov 03 20:12:52 xps systemd[1]: linux-enable-ir-emitter.service: Consumed 1.955s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit linux-enable-ir-emitter.service completed and consumed the indicated resources.
-- Boot 9df1c1c72e2f429ca9845f3f52f44d76 --
Nov 03 20:14:30 xps systemd[1]: Starting enable the infrared emitter...
░░ Subject: A start job for unit linux-enable-ir-emitter.service has begun execution
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit linux-enable-ir-emitter.service has begun execution.
░░ 
░░ The job identifier is 141.
Nov 03 20:14:30 xps systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit linux-enable-ir-emitter.service has successfully entered the 'dead' state.
Nov 03 20:14:30 xps systemd[1]: Finished enable the infrared emitter.
░░ Subject: A start job for unit linux-enable-ir-emitter.service has finished successfully
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit linux-enable-ir-emitter.service has finished successfully.
░░ 
░░ The job identifier is 141.
Nov 03 20:14:30 xps systemd[1]: linux-enable-ir-emitter.service: Consumed 1.882s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit linux-enable-ir-emitter.service completed and consumed the indicated resources.

```
</details>

(Specify if possible when the problem occurred)

**Additional info**
 - Distro: Ubuntu and Fedora
 - Version: 21.10 and 35
 - Version of linux-enable-ir-emitter: 3.2.0
 - How did you install linux-enable-ir-emitter: (bash and COPR when I installed on Fedora 35)

logicito, 2021-11-04:
> Can you please open an issue using the right template, in order to provide all necassary information.
> You can juste copy paste the template on a new comment, no need to reopen a new issue : [...]

Apologies, I just added the comments with the info requested

EmixamPP, 2021-11-04:
I will work on the systemd service. 
It seems that with the new distro versions, the access to /dev/video at boot is not totaly finished.
I will keep you in touch.

logicito, 2021-11-04:
Thank you very much

EmixamPP, 2021-11-04:
Can you please do the following steps :
1. clone the git 
2. checkout on `dev`
3. execute `sudo bash installer.sh reinstall`
4. execute `sudo linux-enable-ir-emitter boot enable`
5. shutdown (not reboot ; for test purpose)
6. restart the compute
Did that fix the problem ?

logicito, 2021-11-04:
Working on it........

logicito, 2021-11-04:
I am not an expert in git, I already close it, what do you mean by **checkout on dev**?

EmixamPP, 2021-11-04:
If you have already clone the git so execute :  `git checkout dev && git fetch && git update`.
By cloning I mean that you executed  `git clone https://github.com/EmixamPP/linux-enable-ir-emitter.git` not just downloaded a zip file with github.

logicito, 2021-11-04:
> If you have already clone the git so execute : `git checkout dev && git fetch && git update`.
> By cloning I mean that you executed `git clone https://github.com/EmixamPP/linux-enable-ir-emitter.git` not just downloaded a zip file through github.

I did execute git clone......
then cd linux......
git checkout dev OK
git fetch OK
git update ERROR: update is not a git command

EmixamPP, 2021-11-04:
Yes sorry I wrote "git update" instead of "git pull"

logicito, 2021-11-04:
ok, git pull = Already up to date

Now I will run the bash.....

EmixamPP, 2021-11-04:
This is not normal that it's already up to date, because I've push some modification. 
Are you sure that you have executed git checkout dev ?

Or delete and clone the git again in order to be sure that all is up to date.

logicito, 2021-11-04:
Yes, I did git checkout dev and was ok

**It worked after shutdown, also after a restart**

EmixamPP, 2021-11-04:
Nice it works ! Thank you for you help !

logicito, 2021-11-04:
Are you going to release a new version?

EmixamPP, 2021-11-04:
Yes, in a few hours at most.

logicito, 2021-11-04:
Thank you, how can I fully remove (purge) what I have today, including configuration, to test it again with the final release later?

EmixamPP, 2021-11-04:
`sudo bash installer.sh uninstall`

logicito, 2021-11-04:
Thank you, please ping me when the new version is out to test it

logicito, 2021-11-04:
Version 3.2.2 is working OK

Thank you very much for the quick fix and for maintaining such a great tool, without it, I cannot use Howdy

Great work!!

EmixamPP, 2021-11-04:
The COPR version will soon be finished being built.

logicito, 2021-11-04:
I noticed that now the boot service will be automatically enabled after configuration, is that the expected behavior?

EmixamPP, 2021-11-04:
Yes because I think everyone enables it directly anyway.

logicito, 2021-11-04:
Makes perfect sense, great addition!!

logicito, 2021-11-04:
I just tested it again, on a fresh Ubuntu 21.10 install, worked flawlessly!!

EmixamPP, 2021-11-04:
Thanks for all your testing !

logicito, 2021-11-04:
Anytime, my pleasure

logicito, 2021-11-04:
Unfortunately, there is still a bug, when I resume from Hibernation in Ubuntu 21.10 IR less will not turn on, before with 21.04 they did.

EmixamPP, 2021-11-04:
Systemd sucks so much ...

EmixamPP, 2021-11-04:
Can you try the new fixes ? There are on the `dev` branch, so you need to checkout and git fetch and update.  Or you can execute `sudo bash installer.sh update` (I've implemented this command in the 3.2.2) .

logicito, 2021-11-04:
I just have to do **sudo bash installer.sh update** and then reboot?

EmixamPP, 2021-11-04:
Reexecute `sudo linux-enable-ir-emitter boot enable` before

logicito, 2021-11-04:
This is the update output:

removed '/usr/bin/linux-enable-ir-emitter'
removed '/usr/lib/linux-enable-ir-emitter/Systemd.py'
removed '/usr/lib/linux-enable-ir-emitter/linux-enable-ir-emitter.py'
removed '/usr/lib/linux-enable-ir-emitter/__pycache__/globals.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/__pycache__/Systemd.cpython-39.pyc'
removed directory '/usr/lib/linux-enable-ir-emitter/__pycache__'
removed '/usr/lib/linux-enable-ir-emitter/command/run.py'
removed '/usr/lib/linux-enable-ir-emitter/command/configure.py'
removed '/usr/lib/linux-enable-ir-emitter/command/manual.py'
removed '/usr/lib/linux-enable-ir-emitter/command/fix.py'
removed '/usr/lib/linux-enable-ir-emitter/command/boot.py'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/boot.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/run.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/fix.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/__init__.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/configure.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/manual.cpython-39.pyc'
removed directory '/usr/lib/linux-enable-ir-emitter/command/__pycache__'
removed '/usr/lib/linux-enable-ir-emitter/command/__init__.py'
removed directory '/usr/lib/linux-enable-ir-emitter/command'
removed '/usr/lib/linux-enable-ir-emitter/globals.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/DriverGenerator.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/__pycache__/Driver.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/driver/__pycache__/__init__.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/driver/__pycache__/DriverGenerator.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/driver/__pycache__/DriverSerializer.cpython-39.pyc'
removed directory '/usr/lib/linux-enable-ir-emitter/driver/__pycache__'
removed '/usr/lib/linux-enable-ir-emitter/driver/DriverSerializer.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/__init__.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/Driver.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/uvc/execute_query.o'
removed '/usr/lib/linux-enable-ir-emitter/driver/uvc/set_query'
removed '/usr/lib/linux-enable-ir-emitter/driver/uvc/get_query'
removed '/usr/lib/linux-enable-ir-emitter/driver/uvc/len_query'
removed directory '/usr/lib/linux-enable-ir-emitter/driver/uvc'
removed directory '/usr/lib/linux-enable-ir-emitter/driver'
removed '/usr/lib/linux-enable-ir-emitter/IrConfiguration.py'
removed directory '/usr/lib/linux-enable-ir-emitter/'
removed '/etc/systemd/system/linux-enable-ir-emitter.service'
removed '/etc/udev/rules.d/99-linux-enable-ir-emitter.rules'
remote: Enumerating objects: 38, done.
remote: Counting objects: 100% (38/38), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 23 (delta 15), reused 22 (delta 14), pack-reused 0
Unpacking objects: 100% (23/23), 2.58 KiB | 661.00 KiB/s, done.
From https://github.com/EmixamPP/linux-enable-ir-emitter
   45b0590..1548988  master     -> origin/master
   ed170c8..a885f64  dev        -> origin/dev
Updating 45b0590..1548988
Fast-forward
 AUR/git/linux-enable-ir-emitter.install     | 2 +-
 AUR/release/linux-enable-ir-emitter.install | 2 +-
 COPR/linux-enable-ir-emitter.spec           | 3 ++-
 installer.sh                                | 2 +-
 sources/globals.py                          | 2 +-
 5 files changed, 6 insertions(+), 5 deletions(-)
Requirement already satisfied: opencv-python in /usr/local/lib/python3.9/dist-packages (4.5.4.58)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (5.3.1)
Requirement already satisfied: numpy>=1.19.3 in /usr/local/lib/python3.9/dist-packages (from opencv-python) (1.21.3)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
make: Entering directory '/home/brosenz/linux-enable-ir-emitter/sources/driver/uvc'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/brosenz/linux-enable-ir-emitter/sources/driver/uvc'
install: creating directory '/usr/lib/linux-enable-ir-emitter'
'sources/globals.py' -> '/usr/lib/linux-enable-ir-emitter/globals.py'
'sources/IrConfiguration.py' -> '/usr/lib/linux-enable-ir-emitter/IrConfiguration.py'
'sources/linux-enable-ir-emitter.py' -> '/usr/lib/linux-enable-ir-emitter/linux-enable-ir-emitter.py'
'sources/Systemd.py' -> '/usr/lib/linux-enable-ir-emitter/Systemd.py'
install: creating directory '/usr/lib/linux-enable-ir-emitter/command'
'sources/command/boot.py' -> '/usr/lib/linux-enable-ir-emitter/command/boot.py'
'sources/command/configure.py' -> '/usr/lib/linux-enable-ir-emitter/command/configure.py'
'sources/command/fix.py' -> '/usr/lib/linux-enable-ir-emitter/command/fix.py'
'sources/command/__init__.py' -> '/usr/lib/linux-enable-ir-emitter/command/__init__.py'
'sources/command/manual.py' -> '/usr/lib/linux-enable-ir-emitter/command/manual.py'
'sources/command/run.py' -> '/usr/lib/linux-enable-ir-emitter/command/run.py'
install: creating directory '/usr/lib/linux-enable-ir-emitter/driver'
'sources/driver/DriverGenerator.py' -> '/usr/lib/linux-enable-ir-emitter/driver/DriverGenerator.py'
'sources/driver/Driver.py' -> '/usr/lib/linux-enable-ir-emitter/driver/Driver.py'
[... cut ...]
'sources/driver/__init__.py' -> '/usr/lib/linux-enable-ir-emitter/driver/__init__.py'
install: creating directory '/usr/lib/linux-enable-ir-emitter/driver/uvc'
'sources/driver/uvc/get_query' -> '/usr/lib/linux-enable-ir-emitter/driver/uvc/get_query'
'sources/driver/uvc/len_query' -> '/usr/lib/linux-enable-ir-emitter/driver/uvc/len_query'
'sources/driver/uvc/set_query' -> '/usr/lib/linux-enable-ir-emitter/driver/uvc/set_query'
'sources/driver/uvc/execute_query.o' -> '/usr/lib/linux-enable-ir-emitter/driver/uvc/execute_query.o'

logicito, 2021-11-04:
rebooting and checking now....

EmixamPP, 2021-11-04:
Hum one detail : have you execute `git checkout dev` before `sudo bash installer.sh update` ? Seems your log I don't think so. Sorry I forget to mension that.

logicito, 2021-11-04:
I didn't, I thought it was one of the other, I will do it now

logicito, 2021-11-04:
New output:

brosenz@xps:~$ cd linux-enable-ir-emitter/
brosenz@xps:~/linux-enable-ir-emitter$ git checkout dev
Branch 'dev' set up to track remote branch 'dev' from 'origin'.
Switched to a new branch 'dev'
brosenz@xps:~/linux-enable-ir-emitter$ sudo bash installer.sh update
removed '/usr/bin/linux-enable-ir-emitter'
removed '/usr/lib/linux-enable-ir-emitter/Systemd.py'
removed '/usr/lib/linux-enable-ir-emitter/linux-enable-ir-emitter.py'
removed '/usr/lib/linux-enable-ir-emitter/__pycache__/globals.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/__pycache__/Systemd.cpython-39.pyc'
removed directory '/usr/lib/linux-enable-ir-emitter/__pycache__'
removed '/usr/lib/linux-enable-ir-emitter/command/run.py'
removed '/usr/lib/linux-enable-ir-emitter/command/configure.py'
removed '/usr/lib/linux-enable-ir-emitter/command/manual.py'
removed '/usr/lib/linux-enable-ir-emitter/command/fix.py'
removed '/usr/lib/linux-enable-ir-emitter/command/boot.py'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/boot.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/run.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/fix.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/__init__.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/configure.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/command/__pycache__/manual.cpython-39.pyc'
removed directory '/usr/lib/linux-enable-ir-emitter/command/__pycache__'
removed '/usr/lib/linux-enable-ir-emitter/command/__init__.py'
removed directory '/usr/lib/linux-enable-ir-emitter/command'
removed '/usr/lib/linux-enable-ir-emitter/globals.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/DriverGenerator.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/__pycache__/Driver.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/driver/__pycache__/__init__.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/driver/__pycache__/DriverGenerator.cpython-39.pyc'
removed '/usr/lib/linux-enable-ir-emitter/driver/__pycache__/DriverSerializer.cpython-39.pyc'
removed directory '/usr/lib/linux-enable-ir-emitter/driver/__pycache__'
removed '/usr/lib/linux-enable-ir-emitter/driver/DriverSerializer.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/__init__.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/Driver.py'
removed '/usr/lib/linux-enable-ir-emitter/driver/uvc/execute_query.o'
removed '/usr/lib/linux-enable-ir-emitter/driver/uvc/set_query'
removed '/usr/lib/linux-enable-ir-emitter/driver/uvc/get_query'
removed '/usr/lib/linux-enable-ir-emitter/driver/uvc/len_query'
removed directory '/usr/lib/linux-enable-ir-emitter/driver/uvc'
removed directory '/usr/lib/linux-enable-ir-emitter/driver'
removed '/usr/lib/linux-enable-ir-emitter/IrConfiguration.py'
removed directory '/usr/lib/linux-enable-ir-emitter/'
removed '/usr/lib/systemd/system/linux-enable-ir-emitter.service'
removed '/etc/udev/rules.d/99-linux-enable-ir-emitter.rules'
Already up to date.
Requirement already satisfied: opencv-python in /usr/local/lib/python3.9/dist-packages (4.5.4.58)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (5.3.1)
Requirement already satisfied: numpy>=1.19.3 in /usr/local/lib/python3.9/dist-packages (from opencv-python) (1.21.3)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
make: Entering directory '/home/brosenz/linux-enable-ir-emitter/sources/driver/uvc'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/brosenz/linux-enable-ir-emitter/sources/driver/uvc'
install: creating directory '/usr/lib/linux-enable-ir-emitter'
'sources/globals.py' -> '/usr/lib/linux-enable-ir-emitter/globals.py'
'sources/IrConfiguration.py' -> '/usr/lib/linux-enable-ir-emitter/IrConfiguration.py'
'sources/linux-enable-ir-emitter.py' -> '/usr/lib/linux-enable-ir-emitter/linux-enable-ir-emitter.py'
'sources/Systemd.py' -> '/usr/lib/linux-enable-ir-emitter/Systemd.py'
install: creating directory '/usr/lib/linux-enable-ir-emitter/command'
'sources/command/boot.py' -> '/usr/lib/linux-enable-ir-emitter/command/boot.py'
'sources/command/configure.py' -> '/usr/lib/linux-enable-ir-emitter/command/configure.py'
'sources/command/fix.py' -> '/usr/lib/linux-enable-ir-emitter/command/fix.py'
'sources/command/__init__.py' -> '/usr/lib/linux-enable-ir-emitter/command/__init__.py'
'sources/command/manual.py' -> '/usr/lib/linux-enable-ir-emitter/command/manual.py'
'sources/command/run.py' -> '/usr/lib/linux-enable-ir-emitter/command/run.py'
install: creating directory '/usr/lib/linux-enable-ir-emitter/driver'
'sources/driver/DriverGenerator.py' -> '/usr/lib/linux-enable-ir-emitter/driver/DriverGenerator.py'
'sources/driver/Driver.py' -> '/usr/lib/linux-enable-ir-emitter/driver/Driver.py'
'sources/driver/DriverSerializer.py' -> '/usr/lib/linux-enable-ir-emitter/driver/DriverSerializer.py'
'sources/driver/__init__.py' -> '/usr/lib/linux-enable-ir-emitter/driver/__init__.py'
install: creating directory '/usr/lib/linux-enable-ir-emitter/driver/uvc'
'sources/driver/uvc/get_query' -> '/usr/lib/linux-enable-ir-emitter/driver/uvc/get_query'
'sources/driver/uvc/len_query' -> '/usr/lib/linux-enable-ir-emitter/driver/uvc/len_query'
'sources/driver/uvc/set_query' -> '/usr/lib/linux-enable-ir-emitter/driver/uvc/set_query'
'sources/driver/uvc/execute_query.o' -> '/usr/lib/linux-enable-ir-emitter/driver/uvc/execute_query.o'
brosenz@xps:~/linux-enable-ir-emitter$

EmixamPP, 2021-11-04:
So reexecute `boot enable`, reboot, test if it works after hibernate. And in any case show me the `boot status` output. Thanks again

logicito, 2021-11-04:
It no longer boots, I get the following message after shutdown, I am stuck, I will have to reinstall from scratch:

/dev/nvme0n1p3: clean, 168481/6250496 files, 3434093/24999936 blocks (this is the swap partition)
[FAILED] Failed to start Hybrid Suspend+Hibernate
[DEPEND] Dependency failed for Hybrid Suspend+Hibernate

EmixamPP, 2021-11-04:
Can you show me the `boot status` ? Or it's impossible ? 
I will continue to work on this tomorrow ...

logicito, 2021-11-04:
The system is gone, I can't boot, neither login, sorry, I have to reinstall it

EmixamPP, 2021-11-04:
What the fuck, no I'm sorry for that. 
I've done a last commit before go to the bed, I hope it will be fix this huge issue.

logicito, 2021-11-04:
Thanks god I had Timeshift, repeated the sequence, it booted now, here is the output from boot status:

brosenz@xps:~$ cd linux-enable-ir-emitter/
brosenz@xps:~/linux-enable-ir-emitter$ sudo linux-enable-ir-emitter boot status
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Thu 2021-11-04 19:54:54 EDT; 41s ago
    Process: 794 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 794 (code=exited, status=0/SUCCESS)
        CPU: 1.462s

Nov 04 19:54:54 xps systemd[1]: Starting enable the infrared emitter...
Nov 04 19:54:54 xps systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Nov 04 19:54:54 xps systemd[1]: Finished enable the infrared emitter.
Nov 04 19:54:54 xps systemd[1]: linux-enable-ir-emitter.service: Consumed 1.462s CPU time.
brosenz@xps:~/linux-enable-ir-emitter$

logicito, 2021-11-04:
What I have to do ensure I get the last commit?

EmixamPP, 2021-11-04:
`sudo bash installer.sh update` and be on the `dev` branch.

logicito, 2021-11-04:
Hibernate didn't work, no IR LEDs

logicito, 2021-11-05:
I'll do it again, maybe before I didn't get the last commit

logicito, 2021-11-05:
I get this:

brosenz@xps:~/linux-enable-ir-emitter$ git checkout dev
Already on 'dev'
Your branch is up to date with 'origin/dev'.

logicito, 2021-11-05:
That means the last run had the last commit because I got the same message

logicito, 2021-11-05:
The last output from boot status is the most up to date

EmixamPP, 2021-11-05:
Can you try a last time and a clean install ? 
1. sudo bash installer.sh uninstall
2. delete de folder
3. git clone 
4. git checkout dev
5. sudo bash installer.sh install
6. configure and test if it works after hibernate

You can do that tomorrow, I will stop for now. Thank you !

logicito, 2021-11-05:
ok, I will do it now, super fast

logicito, 2021-11-05:
sorry, didn't work

logicito, 2021-11-05:
Output from boot status:

○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Thu 2021-11-04 20:09:02 EDT; 27min ago
    Process: 3456 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 3456 (code=exited, status=0/SUCCESS)
        CPU: 1.957s

Nov 04 20:09:02 xps systemd[1]: Starting enable the infrared emitter...
Nov 04 20:09:02 xps systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Nov 04 20:09:02 xps systemd[1]: Finished enable the infrared emitter.
Nov 04 20:09:02 xps systemd[1]: linux-enable-ir-emitter.service: Consumed 1.957s CPU time.

EmixamPP, 2021-11-05:
Hum OK there is à conflict. 
You should execute : 
1. Boot disable
2. Sudo rm /etc/systemd/system/linux-enable-ir-emitter.service (the service must be in /usr/lib/sysremd/system)
3. Boot enable 
4. Reboot and test if it work after hibernation.

logicito, 2021-11-05:
Testing now.....

logicito, 2021-11-05:
I got the following message at boot:

[image]

logicito, 2021-11-05:
Output from boot status:

○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/lib/systemd/system/linux-enable-ir-emitter.service; enabled; vendor preset: enabled)
     Active: inactive (dead)

EmixamPP, 2021-11-05:
OK now it should be fine, I think I've found the solution ! I've push it on the `dev` branch
I'm really sorry to have to make you do all these tests and failures. I'd have to set up several virtual machines too.

[38 more replies]

closed 2021-12-17 by EmixamPP