EmixamPP/linux-enable-ir-emitter #58: Not working on ThinkPad X1 Carbon after previously working

jb3, 2021-11-04:
NOTE: this is an odd one, originally it worked however I think after running some howdy test and maybe `linux-enable-ir-emitter test` and CTRL+Cing midway the configuration does not work anymore and I cannot get the IR emitter to work, even after a reboot.

**Which device did you used ?**
ThinkPad X1 Carbon Gen 9

**Ouput of `	`**
<details><summary>configure command output</summary>

```
INFO: Warning to do not kill the processus !
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (2075) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (1053) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 13, selector: 14, curr control: ['2', '20'], max control: ['2', '100'], res control: ['1', '10']
DEBUG: unit: 14, selector: 2, curr control: ['1', '0', '0', '0', '0', '0', '0', '16', '39', '0', '0', '0', '0', '0', '0'], max control: ['7', '0', '0', '0', '0', '0', '0', '200', '20', '5', '0', '0', '0', '0', '0'], res control: ['0', '0', '0', '0', '0', '0', '0', '1', '0', '0', '0', '0', '0', '0', '0']
DEBUG: unit: 14, selector: 6, curr control: ['1', '3', '1', '0', '0', '0', '0', '0', '0'], max control: ['1', '3', '3', '0', '0', '0', '0', '0', '0'], res control: ['0', '0', '1', '0', '0', '0', '0', '0', '0']
DEBUG: control: ['1', '3', '2', '0', '0', '0', '0', '0', '0']
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (2075) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (1053) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: ['1', '3', '3', '0', '0', '0', '0', '0', '0']
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (2075) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (1053) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.5.4/modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
ERROR: The configuration has failed.
INFO: Do not hesitate to open an issue on GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Integrated Camera: Integrated C (usb-0000:00:14.0-4):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
```

**Ouput of `v4l2-ctl -d /dev/video2 --list-formats-ext`**
<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Additional info**
 - Distro: Arch Linux
 - Version: 5.14.16-arch1-1
 - Version of linux-enable-ir-emitter: 3.2.2
 - How did you install linux-enable-ir-emitter: AUR

jb3, 2021-11-04:
This seems to have solved itself after some rebooting and config deleting.

closed 2021-11-04 by jb3

EmixamPP, 2021-11-04:
Feel free to reopen in case of problems!