EmixamPP/linux-enable-ir-emitter #60: Cannot run

benoe77, 2021-11-06:
yes
**Describe the bug**

the command doesn't start
**Who to reproduce**

After uninstalling the earlier version, updating and reinstalling the latest version, the command doesn't start at all with any argument:

output:

`linux-enable-ir-emitter -h
Traceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 9, in <module>
    from command import boot, fix, manual, run, configure
  File "/usr/lib/linux-enable-ir-emitter/command/boot.py", line 5, in <module>
    from Systemd import Systemd
  File "/usr/lib/linux-enable-ir-emitter/Systemd.py", line 19, in <module>
    class Systemd:
  File "/usr/lib/linux-enable-ir-emitter/Systemd.py", line 20, in Systemd
    def __init__(self, devices: list[str]) -> None:
TypeError: 'type' object is not subscriptable
`

**Additional info**
 - Distro: Ubuntu
 - Version: 20.04
 - Version of linux-enable-ir-emitter: the latest on 2021.11.06 
 - How did you install linux-enable-ir-emitter: 
bash

EmixamPP, 2021-11-06:
Which version of python do you use ? `python -V`
Do you have many installed version ?

benoe77, 2021-11-06:
`python -V
Python 2.7.18
`
I don't think I installed any other version of python

EmixamPP, 2021-11-06:
I used a feature of python3.9 and you'r under python 3.8. (3.9 is one year old and 3.8 is two years old)
I will update the software, in at most one hour I will push a fix. 

It's a painfull for developers that Ubuntu leaves old versions of Python on its LTS versions.

benoe77, 2021-11-06:
I also have this :)

`python3.8 -V
Python 3.8.10
`

EmixamPP, 2021-11-06:
You can execute `sudo bash installer.sh update`. I have update the software in order to support python3 version <3.9 (but  >3.6.2). I think it should be fine on any distro

Let me know if you have remaining problems

EmixamPP, 2021-11-06:
How I forgot to push ! Now you can :smile:

benoe77, 2021-11-06:
I confirm this version is working:

`linux-enable-ir-emitter 3.2.2
`

closed 2021-11-06 by benoe77