EmixamPP/linux-enable-ir-emitter #61: Not working on Ubuntu 20.04 (Razer Book 13)

williamd5, 2021-11-22:
**Describe the bug**

The following error is shown after running `install.sh`
```
install: cannot stat 'sources/autocomplete/%{name}': No such file or directory
```
Upon running `sudo linux-enable-ir-emitter configure`, the IR did not start flashing. I answered `No` about 10 times (and I don't have 10 cameras) and each time got:
```
Ioctl error code: -1, errno: 84
Illegal byte sequence.
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? No
```
I saw my regular camera indicator flash the last several times (but not the IR camera).
**How to reproduce**

I followed the instructions from the readme (version b2172cf)

**Additional info**
 - Distro: `Ubuntu`
 - Version: `20.04.3 LTS x86_64 with 5.10.0-1051-oem kernel`
 - Version of linux-enable-ir-emitter:  `3.2.3`
 - Version of python3: `3.8.10`
 - How did you install linux-enable-ir-emitter: bash

EmixamPP, 2021-11-22:
Can you please open an issue using this https://github.com/EmixamPP/linux-enable-ir-emitter/issues/new?assignees=&labels=&template=configuration-has-failed.md&title= template, I need more information to help you with the configuration.

> install: cannot stat 'sources/autocomplete/%{name}': No such file or directory

I've push a fix for this bug, this is not very important, it's the bash autocomplete that wasn't installed correctly.

> ... and I don't have 10 cameras ...

Yes don't worry it's normal

williamd5, 2021-11-22:
Here is the additional information:

<details><summary>`v4l2-ctl --list-devices`</summary>

```
Integrated Camera: Integrated C (usb-0000:00:14.0-1):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
```

Utility was not initially installed. Upon installation of v4l-utils, I got `sh: 0: getcwd() failed: No such file or directory` after processing triggers.
</details>

<details><summary>`v4l2-ctl -d /dev/video0 --list-formats-ext`</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
```

</details>

<details><summary>`/dev/video1`</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```

</details>

<details><summary>`/dev/video2` (I believe this is the IR camera)</summary>
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
</details>

<details><summary>`/dev/video3`</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```

</details>

EmixamPP, 2021-11-22:
You forgot to give me the ouptut of `linux-enable-ir-emitter --verbose configuration`

williamd5, 2021-11-22:
> You forgot to give me the ouptut of `linux-enable-ir-emitter --verbose configuration`

Sorry about that. I only included a small part in the original issue.

This gives me bad usage error.

```sh
usage: linux-enable-ir-emitter [-h] [-v] [-V] {run,configure,manual,boot,test,fix} ...
linux-enable-ir-emitter: error: argument command: invalid choice: 'configuration' (choose from 'run', 'configure', 'manual', 'boot', 'test', 'fix')
```
I ran it again

```
william@razer ~ ➜ sudo linux-enable-ir-emitter --verbose configure
[sudo] password for william: 
INFO: Warning to do not kill the process !
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? No
Ioctl error code: -1, errno: 84
Illegal byte sequence.
DEBUG: unit: 10, selector: 10, curr control: ['3', '8', '0', '187', '187', '187', '187', '187'], max control: ['255', '255', '255', '255', '255', '255', '255', '255'], res control: ['1', '0', '0', '0', '0', '0', '0', '0']
DEBUG: control: ['4', '8', '0', '187', '187', '187', '187', '187']
Ioctl error code: -1, errno: 84
Illegal byte sequence.
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? Yes
Ioctl error code: -1, errno: 84
Illegal byte sequence.
INFO: The configuration is completed with success.
INFO: The boot service have been enabled, please reboot.
```
And this time on the second prompt I did see the IR camera flash (it's possible that I missed it the previous try). Anyways, I am using this tool to try to resolve an issue with Howdy and I don't know what I am doing, so it's possible everything here is working fine.

Let me know if any additional information is needed, otherwise if you think everything should be working fine, feel free to close this. Thanks for quick responses.

EmixamPP, 2021-11-22:
If you answer "no" to the first question, it means that your emitter doesn't work natifively. 
Here is what you have to do :
1. go in the cloned directory and execute `sudo bash installer.sh uninstall`
2. `git fetch && git pull`
3. `sudo bash installer.sh install`
4.  shutdown and then start your laptop (not reboot)
5. `sudo linux-enable-ir-emitter --verbose configure` be patient and give me the output if the configuration failed

williamd5, 2021-11-22:
I got Howdy to work so it's possible I never even had a problem with the emitter. The LED is just very tiny and I have to look away from the screen to spot it so it's possible I just didn't see it.

EmixamPP, 2021-11-22:
You can use your phone camera in order to see infrared better

williamd5, 2021-11-22:
Okay, thanks for your help!

closed 2021-11-22 by williamd5

EmixamPP, 2021-11-22:
Wow, you did fast ! Every thing is fine now ? Your emitter works natively or not ?

You' re welcome !

williamd5, 2021-11-22:
Yes, I think everything is working fine now as I was able to get Howdy (facial recognition software) to work with my IR camera