EmixamPP/linux-enable-ir-emitter #63: Thinkpad X1 Yoga 4th Gen

Gooseheaded, 2022-01-10:
**Which device did you used ?**

The default device.

**Ouput of `linux-enable-ir-emitter --verbose configuration`**
<details><summary>configure command output</summary>

```
[gctrindade@goose-yoga ~]$ sudo linux-enable-ir-emitter --verbose configure
INFO: Warning to do not kill the process !
[ WARN:0@0.207] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.207] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.207] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? Yes
ERROR: Your infrared camera is already working, skipping the configuration.
ERROR: The configuration has failed.
INFO: Do not hesitate to open an issue on GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
[gctrindade@goose-yoga ~]$ 
```
</details>

**Output of `v4l2-ctl --list-devices`**
<details><summary>list devices output</summary>
```
[gctrindade@goose-yoga ~]$ v4l2-ctl --list-devices
Video Capture 3 (usb-0000:00:14.0-8):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```
</details>

**Ouput of `v4l2-ctl -d /dev/video2 --list-formats-ext`**
<details><summary>v4l2 device output</summary>

```
[gctrindade@goose-yoga ~]$ v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Additional info**
 - OS: Arch Linux x86_64 
 - Host: 20SA0004US ThinkPad X1 Yoga 4th 
 - Kernel: 5.15.13-arch1-1  
 - CPU: Intel i5-10210U (8) @ 4.200GHz 
 - GPU: Intel CometLake-U GT2 [UHD Graphics] 
 - Version of linux-enable-ir-emitter: 3.2.5
 - How did you install linux-enable-ir-emitter: AUR

EmixamPP, 2022-01-10:
I don't see the problem if you answered "yes" to the first "Did you see the ir emitter flashing (not just turn on) ? Yes/No ? Yes" question. This means that your infrared camera works because you saw the ir emitter blinking. 

I'm not sure that I understand your problem ? 
Or conversely, I'm not sure that you understand the aim of this software ?

Gooseheaded, 2022-01-11:
Sorry for the confusion!

When I run the `configure` command, the IR emitter does flash successfully.

However, if I run the `test` command, or the `run` command, I get a message saying "ERROR: No driver is currently saved."

Something similar happens with the `fix driver` command: "ERROR: No driver files to delete."

Is this normal?

EmixamPP, 2022-01-12:
> When I run the configure command, the IR emitter > does flash successfully.

So why have you answer yes to the question "Did you see the ir emitter flashing (not just turn on) ? Yes/No ?" 
You have to answer no until the emitter works. 

And yes the other command don't work because you haven't yet configured the camera.

EmixamPP, 2022-01-12:
> When I run the configure command, the IR emitter does flash successfully.

So, as the error message said, your infrared camera works natively. You don't to use this software. 

For which reason did you install linux-enable-ir-emitter    ? Let me know if I can help you

Gooseheaded, 2022-01-12:
I installed this originally because the command `howdy test` initially showed me a series of black-and-white lines, instead of the proper IR image. I then installed this package to see if that would fix the problem.

Anyway, I just ran `howdy test` again and it to works now.  :smile: Not sure if your package helped or if it was something else; but that doesn't matter anymore.

Thank you for your support. :+1:

closed 2022-01-12 by Gooseheaded