EmixamPP/linux-enable-ir-emitter #65: Lenovo Yoga14s (Slim 7) Configuration has failed

arkylin, 2022-01-25:
When I first configured this program, the infrared fill light flashed and red could be seen. 

After I selected enable and restarted, the fill light could not be used. 

After reinstalling the program, no red light is on when the configuration is found.

(I've tried it many times, but it doesn't seem to work)

**Camera**
```
Bus 003 Device 002: ID 04f2:b6cb Chicony Electronics Co., Ltd Integrated Camera
```

**Which device did you used ?**
Lenovo Yoga14s (Slim 7)

**Ouput of `linux-enable-ir-emitter --verbose configuration`**
<details><summary>configure command output</summary>

```
─>$ sudo linux-enable-ir-emitter -v configure
INFO: Warning to do not kill the process !
[ WARN:0@0.308] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.308] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.308] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 13, selector: 14, curr control: ['2', '20'], max control: ['2', '100'], res control: ['1', '10']
DEBUG: unit: 14, selector: 2, curr control: ['1', '0', '0', '0', '0', '0', '0', '16', '39', '0', '0', '0', '0', '0', '0'], max control: ['7', '0', '0', '0', '0', '0', '0', '200', '20', '5', '0', '0', '0', '0', '0'], res control: ['0', '0', '0', '0', '0', '0', '0', '1', '0', '0', '0', '0', '0', '0', '0']
DEBUG: unit: 14, selector: 6, curr control: ['1', '3', '1', '0', '0', '0', '0', '0', '0'], max control: ['1', '3', '3', '0', '0', '0', '0', '0', '0'], res control: ['0', '0', '1', '0', '0', '0', '0', '0', '0']
DEBUG: control: ['1', '3', '2', '0', '0', '0', '0', '0', '0']
[ WARN:0@6.688] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@6.688] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@6.688] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: control: ['1', '3', '3', '0', '0', '0', '0', '0', '0']
[ WARN:0@9.743] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@9.743] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@9.743] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
ERROR: The configuration has failed.
INFO: Do not hesitate to open an issue on GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki

```
</details>

**Output of `v4l2-ctl --list-devices`**
```
╰─>$ sudo  v4l2-ctl --list-devices
Integrated Camera: Integrated C (usb-0000:00:14.0-1):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/media0
        /dev/media1
```

**Ouput of `v4l2-ctl -d /dev/video2 --list-formats-ext`**
<details><summary>v4l2 device output</summary>

```
╰─>$ v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 640x360
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)

```
</details>

**Additional info**
 - Distro: Archlinux
 - Version: None
 - Version of linux-enable-ir-emitter:  linux-enable-ir-emitter 3.2.5
 - How did you install linux-enable-ir-emitter: AUR->GIT

EmixamPP, 2022-01-25:
> When I first configured this program, the infrared fill light flashed and red could be seen.

Could you show me the content of `sudo linux-enable-ir-emitter manual` ? 

>After I selected enable and restarted, the fill light could not be used.

And the output of `sudo linux-enable-ir-emitter boot status`

arkylin, 2022-01-25:
After I reinstalled the program, the red light never came on

**The content of sudo linux-enable-ir-emitter manual**
```
#Caution: any manual modification of this file may corrupt the operation of the program! You must therefore be very careful.
#Please consult https://github.com/EmixamPP/linux-enable-ir-emitter/wiki/Manual-configuration before.
#If you currupt the driver file: execute 'linux-enable-ir-emitter fix driver' to reset the file.

!!python/object:driver.Driver.Driver
_control:
- 0
_device: /dev/videoX
_selector: 0
_unit: 0
```
 **The output of sudo linux-enable-ir-emitter boot status**
```
ERROR: The boot service does not exists.
```

EmixamPP, 2022-01-25:
Try again after a cold boot. I mean: showdown your computer, remove the power cable, wait some seconds and then boot.

arkylin, 2022-01-25:
It's fine now. I wonder why it gets better after cold start. But there seems to be a new problem
```
INFO: Warning to do not kill the process !
[ WARN:0@0.099] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.099] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.099] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
[ WARN:0@5.545] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@5.545] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@5.545] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? y
INFO: The configuration is completed with success.
Traceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 101, in <module>
    configure.execute(args.device[0], args.limit[0], args.pipe_format)
  File "/usr/lib/linux-enable-ir-emitter/command/configure.py", line 29, in execute
    boot.execute("enable")
  File "/usr/lib/linux-enable-ir-emitter/command/boot.py", line 26, in execute
    exit_code = systemd.enable()
  File "/usr/lib/linux-enable-ir-emitter/Systemd.py", line 54, in enable
    self._create_udev_rule()
  File "/usr/lib/linux-enable-ir-emitter/Systemd.py", line 88, in _create_udev_rule
    vid = get_vid(self.device_number(device))
  File "/usr/lib/linux-enable-ir-emitter/Systemd.py", line 30, in device_number
    start_pos = re.search("[0-9]", device).start()
AttributeError: 'NoneType' object has no attribute 'start'

```

EmixamPP, 2022-01-25:
Could show me the content of "sudo linux-enable-ir-emitter manual" ? This is very weird.
If you execute "sudo linux-enable-ir-emitter boot enable", you see the same error ?

arkylin, 2022-01-25:
Oh, I know.
```
!!python/object:driver.Driver.Driver
_control:
- 0
_device: /dev/videoX
_selector: 0
_unit: 0
--- !!python/object:driver.Driver.Driver
_control:
- 1
- 10
_device: /dev/video2
_selector: 14
_unit: 13
```
The configuration file seems to be messy
When I delete the messy files, I can enable it

EmixamPP, 2022-01-25:
You right, the "dumb" configuration 
```
!!python/object:driver.Driver.Driver
_control:
- 0
_device: /dev/videoX
_selector: 0
_unit: 0
```
that has been generated when a told the first time to show me the content of `sudo linux-enable-ir-emitter manual` has been saved. 

In the past, I had set up a system to delete it if the user had not changed it, but I have since removed this system. (I don't remember why)
So, is everything working now ? If you like the project, don't hesitate to star the repository to support me, thank you !

arkylin, 2022-01-26:
It can work well, thank you very much!

closed 2022-01-26 by arkylin