EmixamPP/linux-enable-ir-emitter #67: Turning IR emitter continuously on

nahoj, 2022-04-23:
Hi,

I'm trying to do something a bit different from what this project does, namely to make the IR emitter of my IR camera always be on when the camera is on.

I have a Chicony integrated camera which I'm trying to use to monitor a room at night. At the moment, when I turn on the IR camera, the emitter flashes; this would be the correct behavior for facial recognition but of course for a continuous video stream, it makes the video flash too.

I understand this is not the main purpose of this project but if you could tell me how to tweak it to achieve this or had any insights, that would be super helpful.

Here are some tech infos:
```
$ lsusb|grep -i camera 
3:Bus 003 Device 003: ID 04f2:b684 Chicony Electronics Co., Ltd Chicony USB2.0 Camera

$ sudo linux-enable-ir-emitter configure
INFO: Warning to do not kill the process !
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? Yes
ERROR: Your infrared camera is already working, skipping the configuration.
ERROR: The configuration has failed.
INFO: Do not hesitate to open an issue on GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki

$ sudo linux-enable-ir-emitter run
CRITICAL: Cannot access to /dev/videoX.
```

EmixamPP, 2022-04-23:
Hello, with this version of linux-enable-ir-emitter I'm not sure that you will succeed to do this kind of things, because the driver generator isn't yet perfect (but it will be !). I'm working on the v4.0.0 which will be soon on the dev branch. 

I maybe have an idea for you, but in order to explain it to you properly, is your emitter works natively ? Because you inputted 'yes' to the first question ? Or you have already configured the camera  ?

However, I guess you think that the images/video will be continuously irradiated. Which is not the case (on my camera at least); an infrared is only sent on the first impulse.

nahoj, 2022-04-23:
> is your emitter works natively ? In order word, you didn't install this software to activate your emitter, it was working before.

Indeed, I didn't do anything to it before, it just works that way with tools such as Cheese, Webcamoid, etc.

> However, I guess you think that the images/video will be continuously irradiated. Which is not the case (on my camera at least); an infrared is only sent on the first impulse.

Ah, yes, I was in fact hoping to have this. Do you think this is something that just can't be done with this hardware? I'd also be interested in disabling the emitter entirely, to see if the IR camera is at all usable without it.

EmixamPP, 2022-04-23:
> Do you think this is something that just can't be done with this hardware?

I've explored everything that my infrared camera can do, and I also tried to do this, but without success. Nevertheless, maybe your camera can do this. 

I realize that explaining this properly in writing will take me hours. For perhaps not being clear enough in the end.
I see on your profile that you speak French (I am Belgian), if you wish, we could call each other so that I can explain it to you by sharing a screen, it will be easier for both of us. What is your level of urgency ? Unfortunately, I am personally quite busy. 

> I'd also be interested in disabling the emitter entirely, to see if the IR camera is at all usable without it.

It should be pretty straightforward with the v4 (the code is finished for a long time, but I still have to test it). Because I'm not sure if the v3 will work in the case where the emitter works natively. But you can always try to execute `linux-enable-ir-emitter configure`: where you input no to all questions until you find a query which disable the emitter, and then input yes.

nahoj, 2022-05-29:
So, you could guess from the delay that this wasn't super urgent for me. In fact, I don't think having the emitter consinuously on is a good idea anymore. After fiddling with the emitter for an hour my eyes were aching, and I wouldn't be surprised if it was actually not safe to have the emitter in sight for more than the few seconds of non-continuous emission required for authentication purposes.

I'll still report what I got from `linux-enable-ir-emitter configure`:

- Could not get to a usable state with the emitter continuously on or off.
- Most tries left things as is, namely camera working with the emitter flashing.
- A few tries close to the beginning left the emitter and camera unusable (warnings from linux-enable-ir-emitter and not working with video recording apps). Further tries (or a reboot) fixes it.
- Every time I finally selected "Yes", I got a stack trace.

<details>
  <summary>Full trace</summary>

```
➜  linux-enable-ir-emitter git:(master) ✗ sudo linux-enable-ir-emitter configure
INFO: Warning to do not kill the process !
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? No
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? No
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? No
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? No
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? No
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
[ WARN:0@37.169] global /io/opencv/modules/videoio/src/cap_v4l.cpp (1000) tryIoctl VIDEOIO(V4L2:/dev/video2): select() timeout.
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
[ WARN:0@53.747] global /io/opencv/modules/videoio/src/cap_v4l.cpp (1000) tryIoctl VIDEOIO(V4L2:/dev/video2): select() timeout.
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
[ WARN:0@69.936] global /io/opencv/modules/videoio/src/cap_v4l.cpp (1000) tryIoctl VIDEOIO(V4L2:/dev/video2): select() timeout.
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
[ WARN:0@83.683] global /io/opencv/modules/videoio/src/cap_v4l.cpp (1000) tryIoctl VIDEOIO(V4L2:/dev/video2): select() timeout.
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
[ WARN:0@98.624] global /io/opencv/modules/videoio/src/cap_v4l.cpp (1000) tryIoctl VIDEOIO(V4L2:/dev/video2): select() timeout.
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? no
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
[... 4 lines cut ...]
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? n
Did you see the ir emitter flashing (not just turn on) ? Yes/No ? y
[... cut ...]
  File "/usr/lib/linux-enable-ir-emitter/Systemd.py", line 54, in enable
    self._create_udev_rule()
  File "/usr/lib/linux-enable-ir-emitter/Systemd.py", line 88, in _create_udev_rule
    vid = get_vid(self.device_number(device))
  File "/usr/lib/linux-enable-ir-emitter/Systemd.py", line 30, in device_number
    start_pos = re.search("[0-9]", device).start()
AttributeError: 'NoneType' object has no attribute 'start'
```

</details>

I'm still curious to see if the camera could work with the emitter disabled but nothing urgent, I can wait for v4 to be out :)

EmixamPP, 2022-06-20:
> start_pos = re.search("[0-9]", device).start()
AttributeError: 'NoneType' object has no attribute 'start'

This error at the end is so strange, no one has ever reported such a bug, anyway the v4 is out for all distros. 

You can try again the same manipulation; answer 'n' until the camera is disabled and the yes. 

Some tips :
1. You can add the `--verbose` flag to the command : `sudo linux-enable-ir-emitter -v configure`, it prints debug information about the searching, Roughly speaking, it prints the instruction (called "control") executed on the camera. The algorithm starts from an initial instruction, incrementing it to the maximum possible. If that doesn't work, it starts again from another initial instruction, etc.
2. As you can see, the software in your case tests a lot of possibilities. If you enable the debug  as mentioned in point 1, it prints something at each new initial instruction. Sometimes you will have 1,2 instructions to test from this one, but sometimes it can go up to 256 (instructions have 8-bit values) ! That's why I included a limit of 5 tests for each group of instructions. I'll skip the details, but I've noticed that this limit is usually more than enough. If your camera is extremely modifiable, then a limit of 5 might be a bit too high and make the search too long. You can lower it by using the `--limit` option: `sudo linux-enable-ir-emitter -v configure -l 2` for example. Or increase it if you want. 
3. I think that the instruction that would deactivate the ir should be the initial. So using `-l 1` looks good to me, in order to quickly go through everything that is possible.
4. Finally, if you find that the flash time is too long, you can reduce it by modifying the constant `CAMERA_TRIGER_TIME` of the file `sources/driver/driver-generator.cpp`.

I hardly think I can do a better research than v4. Better in the sense of intelligent, a simple technique would be to test all possible combinations, but clearly this brute force would take years to be tested manually by the user.

Normally, v5 will not require human interaction. I will implement an image analysis algorithm that will determine whether the emitter is working. I may also add an option to disable the emitter, but I'm not sure if this feature will be used.

Well, I've talked a lot again, I hope it works out for you with v4. If I haven't been clear on some point, or you'd like me to detail on what the debug shows, feel free to ask me.

EmixamPP, 2022-11-30:
Feel free to comment this issue if anyone wants to reopen it.

closed 2022-11-30 by EmixamPP