EmixamPP/linux-enable-ir-emitter #70: IR camera stopped working on Windows after trying to set up

rogervn, 2022-05-30:
Hello, I dual boot Windows and Arch linux and I was trying to make the camera work with howdy on Arch, but after installing linux-enable-ir-emitter I tried to configure it and after several times it trying to configure the camera looping on both the infrared and the regular camera, it stopped working to open the camera device.

After rebooting on Windows now Windows can't access the camera as well. It says it can't use Windows Hello.

Running linux-enable-ir-emitter again now always times out on select for /dev/video0 and even the basic test of capturing data with gst-launch now returns an Internal data stream error.

Is there anything that happens during the emitter setup that can change the camera behaviour? Is there any way that I can restore it?

I've tried uninstalling and reinstalling the camera on Windows too hoping it would come back to defaults, but so far nothing helped.

EmixamPP, 2022-05-30:
Firstly, you can shut down and plug power cord and then camera settings will be reset. 
 
Secondly, are you sure that your infrared camera is on /dev/video0 ? (this is not common, usually it is on /dev/video2, but if you sure, no problem with that)
 
I see in your pastbin that what are you doing is unsafe, there is message which ask to do not kill the execution `Warning to do not kill the process !`. And you used ctrl-c, you should wait the end of the searching. It can take a long time indeed, you have to be patient.

EmixamPP, 2022-05-30:
Normally, if you do things correctly, everything should go correctly and without causing problems. 
It is possible that no driver is found for the camera, the v4 of the software is under development to improve it. But I think that the current version should be sufficient in most case.

rogervn, 2022-05-30:
Yes, I'm sure. First time that I tried to use gst-launch on it /dev/video2 shows color picturs and /dev/video0 showed grayscale.

The first time I've configured it it went all the way through until in the middle the video0 simply stopped opening. Now /dev/video0 is not even opening anything anymore, that's why I've ctrl-c it.

rogervn, 2022-05-30:
I've also shutdown the laptop and even disconnected the battery in the BIOS, so that still stuck after it.

EmixamPP, 2022-05-30:
The software open  /dev/video0 and raise an error in case of access is denied. This is what you mean by "the middle the video0 simply stopped opening." ?
If not, it is weird.

Execute `sudo linux-enable-ir-emitter fix driver`, in order to remove every thing generated, and then  `sudo linux-enable-ir-emitter boot disable`.

Try again a cold boot (shut down etc). And first boot on Windows, in order to be sure that the problem is only cause by Linux.

rogervn, 2022-05-30:
What I meant was that the tests were always lighting up either the red sensors, the camera led or both in a rotation manner (for several times) until the script simply failed saying that it was unable to open /dev/video0. After this /dev/video0 does not show anymore on /dev/v4l/by-id:

`ls -lash /dev/v4l/by-id                                                                                                                     
total 0
0 drwxr-xr-x 2 root root 100 May 30 22:02 .
0 drwxr-xr-x 4 root root  80 May 30 22:02 ..
0 lrwxrwxrwx 1 root root  12 May 30 22:02 usb-Chicony_Electronics_Co._Ltd._Integrated_Camera-video-index0 -> ../../video2
0 lrwxrwxrwx 1 root root  12 May 30 22:02 usb-Chicony_Electronics_Co._Ltd._Integrated_Camera-video-index1 -> ../../video3
0 lrwxrwxrwx 1 root root  12 May 30 22:02 usb-SunplusIT_Inc_SPCA2085_PC_Camera_01.00.00-video-index1 -> ../../video1
`

I've tried cleaning the driver, disabling the boot, cold booting into windows and nothing. Still says there's no windows hello camera detected and Linux still doesn't show /dev/video0.

Opening the camera app in Windows or cheese in linux both show two cameras, one which is color and lights the camera led and another one which lights the red sensors and show nothing but black. Calling linux-enable-ir-emitter on /dev/video0 still lights the red sensors solid but it always fails with that timeout.

Is it possible that the hardware was damaged?

EmixamPP, 2022-05-30:
It's really strange, I don't think it's damaged. I suspect that due to the camera access problem that occurred, the current instruction may have disabled something. And the camera memory is may be persistent, hence doing a cold boot doesn't change anything.

The v4 version on the dev branch could fix this problem. 
1. Uninstall the current version
2. Clone the repo and switch to the dev branch
3. Install the software using the bash installer
4. run `sudo linux-enable-ir-emitter configure -d /dev/video0`
5. With each question asked, test the camera to see if it works again.

EmixamPP, 2022-05-30:
I have trouble understanding what could have happened, it's really strange.

rogervn, 2022-05-30:
I'm picking the dev branch now and it seems that makepkg is not working on it. It starts to build 4.0.0, but quickly offloads to 3.2.5:
``` ~/git/linux-enable-ir-emitter/AUR/git  dev●  0m  makepkg -csi                                                  
==> Making package: linux-enable-ir-emitter-git 4.0.0.r7.g294ac8d-1 (Mon 30 May 2022 11:04:46 PM BST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning linux-enable-ir-emitter git repo...
Cloning into bare repository '/home/rogervn/git/linux-enable-ir-emitter/AUR/git/linux-enable-ir-emitter'...
remote: Enumerating objects: 1491, done.
remote: Counting objects: 100% (307/307), done.
remote: Compressing objects: 100% (166/166), done.
remote: Total 1491 (delta 194), reused 227 (delta 139), pack-reused 1184
Receiving objects: 100% (1491/1491), 276.35 KiB | 4.32 MiB/s, done.
Resolving deltas: 100% (834/834), done.
==> Validating source files with sha256sums...
    linux-enable-ir-emitter ... Skipped
==> Extracting sources...
  -> Creating working copy of linux-enable-ir-emitter git repo...
Cloning into 'linux-enable-ir-emitter'...
done.
Switched to a new branch 'makepkg'
==> Starting pkgver()...
==> Updated version: linux-enable-ir-emitter-git 3.2.5.r30.gf7fb0c8-1
==> Starting build()...
make: Entering directory '/home/rogervn/git/linux-enable-ir-emitter/AUR/git/src/linux-enable-ir-emitter/sources/driver'
g++ driver-generator.cpp getquery.h lenquery.h setquery.h executequery.h driver.hpp -O3 -std=c++17 -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wcast-align -Wwrite-strings -Wcast-qual -Wconversion -Wunreachable-code -Winit-self -I/usr/include/opencv4 -lopencv_gapi -lopencv_stitching -lopencv_alphamat -lopencv_aruco -lopencv_barcode -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_cvv -lopencv_dnn_objdetect -lopencv_dnn_superres -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_hfs -lopencv_img_hash -lopencv_intensity_transform -lopencv_line_descriptor -lopencv_mcc -lopencv_quality -lopencv_rapid -lopencv_reg -lopencv_rgbd -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_superres -lopencv_optflow -lopencv_surface_matching -lopencv_tracking -lopencv_highgui -lopencv_datasets -lopencv_text -lopencv_plot -lopencv_videostab -lopencv_videoio -lopencv_viz -lopencv_wechat_qrcode -lopencv_xfeatures2d -lopencv_shape -lopencv_ml -lopencv_ximgproc -lopencv_video -lopencv_xobjdetect -lopencv_objdetect -lopencv_calib3d -lopencv_imgcodecs -lopencv_features2d -lopencv_dnn -lopencv_flann -lopencv_xphoto -lopencv_photo -lopencv_imgproc -lopencv_core  -o driver-generator 
g++ execute-driver.cpp executequery.h driver.hpp -O3 -std=c++17 -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wcast-align -Wwrite-strings -Wcast-qual -Wconversion -Wunreachable-code -Winit-self -o execute-driver
make: Leaving directory '/home/rogervn/git/linux-enable-ir-emitter/AUR/git/src/linux-enable-ir-emitter/sources/driver'
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Stripping unneeded symbols from binaries and libraries...
  -> Compressing man and info pages...
==> Checking for packaging issues...
==> Creating package "linux-enable-ir-emitter-git"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Adding install file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: linux-enable-ir-emitter-git 3.2.5.r30.gf7fb0c8-1 (Mon 30 May 2022 11:04:51 PM BST)
==> Installing package linux-enable-ir-emitter-git with pacman -U...
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) linux-enable-ir-emitter-git-3.2.5.r30.gf7fb0c8-1

Total Installed Size:  0.07 MiB

:: Proceed with installation? [Y/n] 

```

EmixamPP, 2022-05-30:
Yes please use the bash installer

rogervn, 2022-05-30:
Cool, I had to comment the selinux stuff as archlinux don't enable it, but now it's complaining that it doesn't recognize /dev/video0:

```
sudo linux-enable-ir-emitter configure -d /dev/video0                
usage: linux-enable-ir-emitter [-h] [-v] [-V] [-d device] {run,configure,delete,boot} ...
linux-enable-ir-emitter: error: unrecognized arguments: -d /dev/video0

```

EmixamPP, 2022-05-30:
Ah yes sorry it is `sudo linux-enable-ir-emitter -d /dev/video0 configure` now

rogervn, 2022-05-30:
It's running now, but so far it has been a few minutes without any timeout and no questions. Just the red sensors constantly on.

rogervn, 2022-05-30:
Still hanging, pstree shows this:

```
  |   |-zsh
  |   |   `-sudo linux-enable-ir-emitter -d /dev/video0 configure
  |   |       `-linux-enable-ir /usr/bin/linux-enable-ir-emitter -d /dev/video0 configure
  |   |           `-driver-generato /dev/video0 5 /etc/linux-enable-ir-emitter/video0 0
  |   |               `-{driver-generato}

```

EmixamPP, 2022-05-30:
You do have the latest commit from the repo? 
git pull && git fetch && sudo bash installer.sh update

Because I did a commit 40min ago an hot fix

rogervn, 2022-05-30:
Yes, it seems it's already up to date.

rogervn, 2022-05-30:
strace shows a ppol stuck at fd 3:

```sudo strace -f -p 18975                                              
strace: Process 18975 attached
ppoll([{fd=-1}, {fd=3, events=POLLIN}], 2, NULL, NULL, 8

```

EmixamPP, 2022-05-30:
I really don't understand what is going on, can you enable the debug `-v` ? To see if something is printed ?

rogervn, 2022-05-30:
Nothing as well. I've straced he whole thing and this is what my default buffer got:
https://pastebin.com/BrTC5Ts6

rogervn, 2022-05-30:
I have to go now, but can continue the troubleshoot tomorrow evening, if you have a set of things I can try I can get them out of the way quickly tomorrow.

EmixamPP, 2022-05-30:
Ok, it's safe to kill the execution before the debugging has displayed anything, so don't worry with that.
But why does it loop endlessly, I created this bug by mistake during the evening, but I fixed it in my last commit. 

I won't be able to spend the night, sorry! I'll look at it again in the morning. 
I'm really sorry that my program created these problems for you. Unfortunately, this has never happened to anyone else, so I have almost no information to help you.

rogervn, 2022-05-30:
No worries. I'll also only be able to come back to it tomorrow evening. Thursday and Friday are holidays in the UK so I might be more available.

Thanks for the support, I'm completely lost on this.

rogervn, 2022-05-31:
Sorry for the delay. I'm on it now and I realized I recloned it but realized I didn't put it in the dev branch but I'm midway on the process. Can I ctrl-c and restart the PC to try the dev one?

EmixamPP, 2022-05-31:
Yes you can

rogervn, 2022-05-31:
Cool, it's still hanging with the IR sensors enabled and the program is stuck.

It just included the message not to use the camera during execution, but otherwise it's still the same as yesterday:

```
$ sudo linux-enable-ir-emitter -v -d /dev/video0 configure                                                
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
```

rogervn, 2022-05-31:
Attaching strace still shows it stuck in a ppoll:

```
sudo strace -f -p 1258                                                                                  
strace: Process 1258 attached
ppoll([{fd=-1}, {fd=3, events=POLLIN}], 2, NULL, NULL, 8
```

EmixamPP, 2022-05-31:
So there is no first question ...
Hum I think opencv is waiting for access to your camera
This is embarrassing, I'll make you a temporary code

EmixamPP, 2022-05-31:
Change the code of `sources/driver/driver-generator.cpp` with https://pastebin.com/e3txHenV

And then execute `sudo bash installer.sh update`

rogervn, 2022-05-31:
Now I got a question and a debug message, but the question is not moving on with no as an answer:

```
sudo linux-enable-ir-emitter -v -d /dev/video0 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
DEBUG: unit: 4, selector: 2, cur control: 3 32, first control to test: 0 0, res control: 1 0, max control: 255 255
Check if your ir camera is fix. Yes/No ? no
Check if your ir camera is fix. Yes/No ? No
Check if your ir camera is fix. Yes/No ? n

```

I've also tried to use fst-launch to get pictures from it but still with Internal data stream error:

```
gst-launch-1.0 v4l2src device=/dev/video0 num-buffers=10 ! image/jpeg ! multifilesink location="frame-%02d.jpg"
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../gstreamer/subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.000635812
Setting pipeline to NULL ...
Freeing pipeline ...

```

EmixamPP, 2022-05-31:
> Now I got a question and a debug message, but the question is not moving on with no as an answer:
Yes that is expected

> I've also tried to use fst-launch to get pictures from it but still with Internal data stream error:
hum ... so I don't know how to fix your camera if you can't check it ...
May be, run `sudo linux-enable-ir-emitter -v -d /dev/video0 configure` 
and input `n` to all questions, until the executions stop
Then send me the outputs, I will analyse the debug in order to find something which seems interesting

But I'll need some time, like until tomorrow night or Thursday

rogervn, 2022-05-31:
You mean with the master code? The dev code won't move while being stuck on ppol. The master code seems that all commands are timing out in the similar way

rogervn, 2022-05-31:
Cool, this is the output: https://pastebin.com/16H08B7d

Not a lot of interesting things though.

rogervn, 2022-05-31:
So I actually got the camera back working after reinstalling the firmware on Windows. So let me share some weird things here.

It is indeed video0:
```
v4l2-ctl --list-devices                                                                            
Integrated IR Camera: Integrate (usb-0000:00:14.0-5):
	/dev/video0
	/dev/video1
	/dev/media0

Integrated Camera: Integrated C (usb-0000:00:14.0-8):
	/dev/video2
	/dev/video3
	/dev/media1
```

However the Chicony camera which is supposed to be the IR one is linked to video2:

```
ls -lash /dev/v4l/by-id                                                                            
total 0
0 drwxr-xr-x 2 root root 120 May 31 22:21 .
0 drwxr-xr-x 4 root root  80 May 31 22:21 ..
0 lrwxrwxrwx 1 root root  12 May 31 22:21 usb-Chicony_Electronics_Co._Ltd._Integrated_Camera-video-index0 -> ../../video2
0 lrwxrwxrwx 1 root root  12 May 31 22:21 usb-Chicony_Electronics_Co._Ltd._Integrated_Camera-video-index1 -> ../../video3
0 lrwxrwxrwx 1 root root  12 May 31 22:21 usb-SunplusIT_Inc_Integrated_IR_Camera-video-index0 -> ../../video0
0 lrwxrwxrwx 1 root root  12 May 31 22:21 usb-SunplusIT_Inc_Integrated_IR_Camera-video-index1 -> ../../video1
```

Calling `st-launch-1.0 v4l2src device=/dev/video0 num-buffers=10 ! image/jpeg ! multifilesink location="frame-%02d.jpg"` will show grayscale pictures on video0 and color pictures on video2.

And `cheese -d /dev/video0` will start the IR camera correctly with the IR blinking even without running linux-ir-enable-emitter, so I think it's not needed?

The reason I got to try it was because on trying howdy on video0 it doesn't start the IR emitters, so it doesn't recognize a face. I've tried a bit more and it does work after I set the frame dimensions. So I think I'm good?

Do you want me to run something to get some diagnostics?

EmixamPP, 2022-05-31:
Fine ! Yes indeed, you don't need to use my software.
More fear than pain ! 
And "thanks" to your problem, I was able to improve several things.
If even after a cold boot it still works under Linux, then that's fine, you can close the issue

rogervn, 2022-05-31:
Haha, yes, it does work after a clean reboot. Thanks for the support!

Glad that you could use that for something good and sorry about the hassle!

closed 2022-05-31 by rogervn