EmixamPP/linux-enable-ir-emitter #72: AUR package fails to build

winux1, 2022-07-04:
Yes
**Describe the bug**

both of the AUR packages fail to build with this error:
```bash
make: Entering directory '/home/winux/.cache/paru/clone/linux-enable-ir-emitter-git/src/linux-enable-ir-emitter-git/sources/driver'
Package opencv was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv.pc'
to the PKG_CONFIG_PATH environment variable
Package 'opencv', required by 'virtual:world', not found
g++ driver-generator.cpp getquery.h lenquery.h setquery.h executequery.h driver.hpp -O3 -std=c++17 -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wcast-align -Wwrite-strings -Wcast-qual -Wconversion -Wunreachable-code -Winit-self -I/usr/include/opencv4 -lopencv_gapi -lopencv_stitching -lopencv_alphamat -lopencv_aruco -lopencv_barcode -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_cvv -lopencv_dnn_objdetect -lopencv_dnn_superres -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_hfs -lopencv_img_hash -lopencv_intensity_transform -lopencv_line_descriptor -lopencv_mcc -lopencv_quality -lopencv_rapid -lopencv_reg -lopencv_rgbd -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_superres -lopencv_optflow -lopencv_surface_matching -lopencv_tracking -lopencv_highgui -lopencv_datasets -lopencv_text -lopencv_plot -lopencv_videostab -lopencv_videoio -lopencv_viz -lopencv_wechat_qrcode -lopencv_xfeatures2d -lopencv_shape -lopencv_ml -lopencv_ximgproc -lopencv_video -lopencv_xobjdetect -lopencv_objdetect -lopencv_calib3d -lopencv_imgcodecs -lopencv_features2d -lopencv_dnn -lopencv_flann -lopencv_xphoto -lopencv_photo -lopencv_imgproc -lopencv_core  -o driver-generator 
/usr/bin/ld: warning: libfmt.so.8, needed by /usr/lib/libvtkFiltersGeneral.so.1, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/lib/libvtkFiltersGeneral.so.1: undefined reference to `fmt::v8::vformat[abi:cxx11](fmt::v8::basic_string_view<char>, fmt::v8::basic_format_args<fmt::v8::basic_format_context<fmt::v8::appender, char> >)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:14: driver-generator] Error 1
make: Leaving directory '/home/winux/.cache/paru/clone/linux-enable-ir-emitter-git/src/linux-enable-ir-emitter-git/sources/driver'
```

**Who to reproduce**
Try installing the AUR package

**Additional info**
 - Distro: Arch Linux
 - Version: -
 - Version of linux-enable-ir-emitter: latest
 - How did you install linux-enable-ir-emitter:  AUR

EmixamPP, 2022-07-04:
I think the fmt lib dependency is missing. Can you try to install it ?

winux1, 2022-07-04:
Works with fmt installed. Closing issue

closed 2022-07-04 by winux1

tuanlc, 2022-10-16:
Hello @EmixamPP 

I've encountered the same issue even I installed `fmt` package
```
collect2: error: ld returned 1 exit status
make: *** [Makefile:14: driver-generator] Error 1
make: Leaving directory '/home/tuanlc/.cache/yay/linux-enable-ir-emitter-git/src/linux-enable-ir-emitter-git/sources/driver'
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: linux-enable-ir-emitter-git
```

Would you mind taking a look?

My environment: (Archline)
- OS: Linux archlinux 5.15.74-1-lts #1 SMP Sat, 15 Oct 2022 06:43:47 +0000 x86_64 GNU/Linux
- Thinkpad X1 (Gen 6)

Thanks in advance!

EmixamPP, 2022-10-16:
@tuanlc can you please open a new issue and provide the full output ? Because what you show gives no information. Thanks

ckissane, 2022-11-25:
@EmixamPP vtk was needed for me, I believe that may be what they were referring to as well...