EmixamPP/linux-enable-ir-emitter #73: V4 config fails on Thinkpad X390

Bakunin-san, 2022-07-17:
**Ouput of `linux-enable-ir-emitter -v -d /dev/videoX configure`**

<details><summary>configure command output</summary>

```
[~]$ sudo linux-enable-ir-emitter -v -d /dev/video2 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 9, selector: 10, cur control: 0 8 0 187 187 187 187 187, first control to test: 0 0 0 0 0 0 0 0, res control: 1 0
0 0 0 0 0 0, max control: 255 255 255 255 255 255 255 255
DEBUG: Negative answer limit exceeded, skipping the pattern.
ERROR: Impossible to reset the control.
INFO: Please keep this debug in case of issue :
DEBUG: unit: 9, selector: 10, control: 0 8 0 187 187 187 187 187
DEBUG: unit: 10, selector: 2, cur control: 1 0 0 0 0 0 0 16 39 0 0 0 0 0 0, first control to test: 0 0 0 0 0 0 0 16 39 0 0 0 0
0 0, res control: 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0, max control: 7 0 0 0 0 0 0 200 20 5 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Computing the resolution control.
DEBUG: unit: 10, selector: 6, cur control: 1 3 1 0 0 0 0 0 0, first control to test: 1 3 2 0 0 0 0 0 0, res control: 0 0 1 0 0
0 0 0 0, max control: 1 3 3 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 10, selector: 9, cur control: 0 0 0 0, first control to test: 1 0 0 0, res control: 1 0 0 0, max control: 1 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 14, selector: 14, cur control: 0 0, first control to test: 1 10, res control: 1 10, max control: 2 100
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 14, selector: 16, cur control: 0, first control to test: 1, res control: 1, max control: 1
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
paste here
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
[~]$ v4l2-ctl --list-devices
Integrated Camera: Integrated C (usb-0000:00:14.0-8):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/media0
        /dev/media1
```
</details>

**Additional info**
 - Distro: Arch Linux
 - Version: latest
 - Version of linux-enable-ir-emitter: 4.1.2
 - How did you install linux-enable-ir-emitter: AUR
 - Device: Lenovo Thinkpad X390

Possibly related to Issue #71 
Until V4 it was working without any issues but after updating to V4 the program broke. I have tried reinstalling it several times both from the AUR and directly from github and it fails the same way every time. Installing V3 from git creates a functioning configuration but upgradig to V4.1.2  and running the script to migrate the config breaks the program and subsequent attempts to reconfigure the program fails.

EmixamPP, 2022-07-17:
Theoretically, I've fixed the installation script for Arch since v4.1.2 which . Therefore, it should not be necessary to run the migration script manually. So I don't understand why the update v3->v4.1.2 broke the software only on Arch but not on Fedora.
 
And I don't understand why on Arch the configuration does not trigger the emitter. I think this is related to OpenCV which does not work proprely ...

> Installing V3 from git creates a functioning configuration but upgradig to V4.1.2 and running the script to migrate the config breaks the program and subsequent attempts to reconfigure the program fails.

Did you use the `bash installer.sh update` command ? Because I realize that this command cannot update major release (I will remove it).

--- 
1. Can you install the v3.2.5 from git (with bash installer)
2. Configure your emitter
3. reboot
4. Download the v4.2.1 from git, install it (with bash installer)
5. Past the output of `cat /etc/linux-enable-ir-emitter.yaml` 
6. Past the output of `cat /etc/linux-enable-ir-emitter/video2` 
7. Past the output of `sudo linux-enable-ir-emitter boot status` 
8. Execute `sudo linux-enable-ir-emitter boot enable` 
9. Past the output of `sudo linux-enable-ir-emitter boot status` 

Thank you !

Bakunin-san, 2022-07-17:
- cat /etc/linux-enable-ir-emitter.yaml
Doesn't exist anymore since the 4.1.2 installation removes this file, I reinstalled 3.2.5 and backed up the file before installing 4.1.2 and here's the original file:
```
#Caution: any manual modification of this file may corrupt the operation of the program! You must therefore be very careful.
#Please consult https://github.com/EmixamPP/linux-enable-ir-emitter/wiki/Manual-configuration before.
#If you currupt the driver file: execute 'linux-enable-ir-emitter fix driver' to reset the file.

!!python/object:driver.Driver.Driver
_control:
- 1
- 3
- 3
- 0
- 0
- 0
- 0
- 0
- 0
_device: /dev/video2
_selector: 6
_unit: 10
```

/etc/linux-enable-ir-emitter/video2 contains:
```
device=/dev/video2
unit=10
selector=9
size=4
control0=1
control1=0
control2=0
control3=0
```
systemctl status (it looks the same both before and after it is enabled at boot):
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/usr/lib/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: inactive (dead) since Sun 2022-07-17 14:29:13 CEST; 12min ago
    Process: 792 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 793 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 793 (code=exited, status=0/SUCCESS)
        CPU: 303ms

Jul 17 14:29:13 host systemd[1]: Starting enable the infrared emitter...
Jul 17 14:29:13 host systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 17 14:29:13 host systemd[1]: Finished enable the infrared emitter.
```
Don't put too much thought into what systemctl says either, after installing and configuring 3.2.5 the IR emitter continues to work for a while even without any version of linux-enable-ir-emitter installed.

Bakunin-san, 2022-07-17:
Regarding opencv, interestingly enough on 3.2.5 I get these error messages when running the configuration but I do not get them on 4.1.2

```
[ WARN:0@7.610] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@7.610] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@7.610] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

EmixamPP, 2022-07-17:
/etc/linux-enable-ir-emitter/video2 is the result of a configuration with the v4 ? 
Because I don't think the migration script (automatically executed during the installation of the v4) could have produced that.

>Regarding opencv, interestingly enough on 3.2.5 I get these error messages when running the configuration but I do not get them on 4.1.2

Yes, I've disabled OpenCV log

Bakunin-san, 2022-07-17:
/etc/linux-enable-ir-emitter/video2 is the result of the configuration with 3.2.5 but after installing 4.1.2
Configuration fails during 4.2.5 but if I tell it it passed successfully after the second attempt (same as 3.2.5) I get this:

´´´
device=/dev/video2
unit=10
selector=2
size=15
control0=0
control1=0
control2=0
control3=0
control4=0
control5=0
control6=0
control7=16
control8=39
control9=0
control10=0
control11=0
control12=0
control13=0
control14=0
´´´

EmixamPP, 2022-07-17:
Ok, things become complicated to explains ^^
The problem is that I cannot find a fix if I don't have precisely the state of the two mentioned files before and after update in a "regular" situation.

1. Uninstall the v3 and v4 (no more files in  /usr/lib/linux-enable-ir-emitter/ and /etc/linux-enable-ir-emitter/ and /etc/linux-enable-ir-emitter.yaml)
2. Shut down your computer
3. remove the AC adaptater
4. boot, install the v3 and configure the emitter
5. reboot, is everythings working ?
6. install the v4 
7. past the content of /etc/linux-enable-ir-emitter/video2 and linux-enable-ir-emitter boot status

I won't have time to try debugging today. So I think it's better for you to go back to v3. And probably tomorrow I'll install a VM with Arch to do some tests ...

EmixamPP, 2022-07-17:
@supdrewin, sorry to bother you, did you have any problems with v4 ? Thanks

Bakunin-san, 2022-07-17:
At the moment the IR emitter just works without anything installed, this has happened once before and it stops working again fairly quickly. I'll update with the yaml and video2 files after it has stopped working and I've reconfigured V3 again.

If I have enough time on Monday or Tuesday I'll boot a fedora live-usb and see if I have the same issue there

Bakunin-san, 2022-07-17:
video2 file from 3.2.5 config after 4.1.2 install:
```
device=/dev/video2
unit=10
selector=6
size=9
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
```

boot status after 4.1.2 install but before reboot
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/usr/lib/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: inactive (dead) since Sun 2022-07-17 15:19:18 CEST; 3min 36s ago
    Process: 796 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 798 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 798 (code=exited, status=0/SUCCESS)
        CPU: 327ms

Jul 17 15:19:17 host systemd[1]: Starting enable the infrared emitter...
Jul 17 15:19:18 host systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 17 15:19:18 host systemd[1]: Finished enable the infrared emitter.
```

boot status after 4.1.2 install but after reboot
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/usr/lib/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: inactive (dead) since Sun 2022-07-17 15:23:51 CEST; 16s ago
    Process: 640 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 641 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 641 (code=exited, status=0/SUCCESS)
        CPU: 80ms

Jul 17 15:23:51 host systemd[1]: Starting enable the infrared emitter...
Jul 17 15:23:51 host systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 17 15:23:51 host systemd[1]: Finished enable the infrared emitter.
```

For now it seems to be working with the 3.2.5 config but 4.1.2 install, I'll leave it as is and update if it stops working

EmixamPP, 2022-07-17:
(Miss click on close) 
Everything seems good, therefore I think there is an issue with AUR, not with the migration script.

Bakunin-san, 2022-07-19:
I'll boot a fedora liveusb tomorrow and see what that does, could also be something weird going on with my specific arch install.

EmixamPP, 2022-08-14:
This problem may be corrected in v4.1.4. I forced v4l2 as opencv backend.

EmixamPP, 2022-08-14:
Let me know if you have the time to test the v4.1.4

1. uninstall it
2. shutdown and boot (in order to reset the camera driver)
3. install the new version
4. configure again and see if the problem still there 

Thanks

Bakunin-san, 2022-08-20:
Sorry for not updating, had a very busy few weeks.
I tried v4.1.4 and configuration is now working, thank you so much! :)

closed 2022-08-20 by Bakunin-san