EmixamPP/linux-enable-ir-emitter #74: Alienware 17R4

4Wheels, 2022-07-18:
**If you had 'ERROR: Impossible to reset the control' message, give the DEBUG info displayed**
```
DEBUG:
```

**Ouput of `linux-enable-ir-emitter -v -d /dev/videoX configure`**

<details><summary>script cannot access the device (Yes, I used sudo)</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
CRITICAL: Cannot access to /dev/video1
ERROR: The configuration has failed.

```
</details>

**Output of `v4l2-ctl --list-devices`**
```
paste here
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-7):
        /dev/video0
        /dev/video1
        /dev/media0

```
</details>

**Additional info**
 - Distro:  Arch
 - Version: 5.18.12-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 15 Jul 2022 15:33:02 +0000 x86_64 GNU/Linux
 - Version of linux-enable-ir-emitter: 1:4.1.2-1
 - How did you install linux-enable-ir-emitter:  AUR 

Please NOTE: /dev/video0 is my regular (non-IR) integrated camera, /dev/video1 is the IR camera.  If I try to configure /dev/video0, it just flashes my regular camera light, not the IR (which is red) so I know it's not working.

EmixamPP, 2022-07-19:
/dev/video1 is not your infrared camera, in fact, this is nothing which can be used.
Are you sure that your computer has an infrared camera ? If so, this is a Kernel compatibility issue. It does not recognize your infrared camera.

4Wheels, 2022-07-19:
Thanks for your reply. Yes I'm positive that my computer has an IR camera as I use it in Windows. In Windows it uses a Realtek IR camera driver. and Tobii Aware uses it also

I'm curious, how are you certain that /dev/video1 is not the IR camera? What is it then? Seems strange that the output recognizes "Video capture" even though it doesn't show any modes.

EmixamPP, 2022-07-19:
Example with my infrared camera 
```
❯ v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
```
As you can see, it as grey scale mode. 

Often the kernel create two entry for each camera, one is usable, the other one not. I think this is due to the usb port (but I'm not sure), because the kernel see two entry at the usb subsystem level.
If you execute `udevadm info -a /dev/video0` and `udevadm info -a /dev/video1` you can see the device hierarchy and compare them.

EmixamPP, 2022-07-19:
#9 has the same problem with his Dell computer.

Personally, I think the problem is that the camera does not use a "standard protocol" to identify itself to the system. It needs its own driver, and unfortunately, except from managing to get the code developed by the manufacturer for Windows, and then to "translate" it to Linux, I'm afraid that the camera will never be correctly recognized by the kernel.

My software is looking for a way to activate the IR emitter, but it needs access to the IR camera.

EmixamPP, 2022-07-19:
May be try some other distros like Fedora or Ubuntu ?

EmixamPP, 2023-06-21:
Closing because we can't do anything until this infrared camera is recognized by the kernel :confused: