EmixamPP/linux-enable-ir-emitter #75: Razer Blade 15 Advanced Late 2019 - Cannot run the configuration, configuration failed

GeoRCode, 2022-07-31:
**If you had 'ERROR: Impossible to reset the control' message, give the DEBUG info displayed**
```
DEBUG:
```

**Ouput of `linux-enable-ir-emitter -v -d /dev/videoX configure`**

<details><summary>configure command output</summary>

```
**I am using video2 for this command**
/usr/lib/linux-enable-ir-emitter/driver/driver-generator: error while loading shared libraries: libopencv_videoio.so.4.5d: cannot open shared object file: No such file or directory
ERROR: The configuration has failed.
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Integrated Camera: Integrated C (usb-0000:00:14.0-7):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
**I am using video2 for this command**
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 360x360
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Additional info**
 - Distro: Ubuntu
 - Version: Focal Fossa (20.04 LTS)
 - Version of linux-enable-ir-emitter: 4.2.1
 - How did you install linux-enable-ir-emitter: PPA

EmixamPP, 2022-07-31:
Opencv is not found on the system, please install it. This is weird because it is in the requiremement of the .deb package. I will look at this tomorrow.

GeoRCode, 2022-08-01:
In fact yesterday I noticed that OpenCV was not found.  The thing is, when I search for `llibopencv_videoio.so.4.` on my system, it actually show me a `libopencv_videoio.so.4.6`. I'm not sure if is then a problem with a line in the driver generator with an outdated form of libopencv_videoio.so
Check it out and let me know ^^

EmixamPP, 2022-08-01:
In order to fix the problem I have to release a new linux-enable-ir-emitter. Unfortunately, I have nothing to release at the moment. I had a small modification I wanted to do, so I will work on it. 

So I advise you to install linux-enable-ir-emitter using the bash script, it will compile the software directly on your system.

GeoRCode, 2022-08-01:
Alright! I be patiently waiting for the new linux-enable-ir-emitter then. I'll also will give it a try to the one with the bash script and see how it goes. Thanks!

EmixamPP, 2022-08-03:
Sorry but I don't understand 

> it gets stuck in the configuration command

which normaly, should not imply 

> On the emitter part, it begins to work and blinks 

Can you past some outputs of the configure command ? (with the answer y/n that you provide)

GeoRCode, 2022-08-04:
I'm gonna write for you the steps I made in order for you to make conclusions and stuff:
1. I opened a terminal and installed the bash script using the "For other distro:" steps found in the README.md
2. I continued by opening a new terminal and wrote `sudo linux-enable-ir-emitter configure`
3. It showed me this (as expected):
```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ?
```
4. I answered 'no' to the question
5. Then it showed me this:
```
can't get debug descriptor: Resource temporarily unavailable
ERROR: Impossible to reset the control.
INFO: Please keep this debug in case of issue :
DEBUG: unit: 9, selector: 10, control: 0 8 0 187 187 187 187 187
Is the ir emitter flashing (not just turn on) ? Yes/No ?
```
6. I continued answering to the question 'no' 6 more times
7. Then, no more questions appeared on console and the emitter actually began to flash
8. I let it run for 5-6 hours with nothing new on the console log, then killed the process
9. I searched for anything created on the supposed driver folder (/etc/linux-enable-ir-emitter) but it was empty
10. I searched in the other folders created by the installation in the whole system but there was nothing new
11. I tested the camera + emitter using `ffplay /dev/video2` and/or `sudo howdy test`. It worked as intended
12. I powered off the whole pc. Then I powered on again, went to Windows, on the lock screen I performed a reboot and then I went to Ubuntu
13. The camera works, the emitter doesn't

Hopefully this help you understand my process

EmixamPP, 2022-08-04:
Interesting... Can you execute `sudo linux-enable-ir-emitter --verbose configure`, and give all the outputs ? When the process seems to be stuck you can kill it.

GeoRCode, 2022-08-04:
Ok.
First I have to uninstall the script and delete all the remaining folders (i.e the git clone one basically) and do the same as before: power off -> Windows -> reboot -> Ubuntu. This is because if I execute `sudo linux-enable-ir-emitter --verbose configure` with the emitter "working as supposed to be", as the script left it before, the console just shows this before getting stucked:
```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
```
So, after installing the script again, `sudo linux-enable-ir-emitter --verbose configure` shows me this:
```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
can't get debug descriptor: Resource temporarily unavailable
DEBUG: unit: 9, selector: 10, cur control: 0 8 0 187 187 187 187 187, first control to test: 0 0 0 0 0 0 0 0, res control: 1 0 0 0 0 0 0 0, max control: 255 255 255 255 255 255 255 255
DEBUG: Negative answer limit exceeded, skipping the pattern.
ERROR: Impossible to reset the control.
INFO: Please keep this debug in case of issue :
DEBUG: unit: 9, selector: 10, control: 0 8 0 187 187 187 187 187
DEBUG: unit: 10, selector: 2, cur control: 1 0 0 0 0 0 0 20 39 0 0 0 0 0 0, first control to test: 0 0 0 0 0 0 0 16 39 0 0 0 0 0 0, res control: 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0, max control: 7 0 0 0 0 0 0 128 150 152 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 10, selector: 3, cur control: 16 0 0 0 0 0 0 100 0 0 0, first control to test: 16 0 0 0 0 0 0 14 6 0 0, res control: 16 0 0 0 0 0 0 0 0 0 0, max control: 16 0 0 0 0 0 0 14 6 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Computing the resolution control.
DEBUG: unit: 10, selector: 6, cur control: 1 3 1 0 0 0 0 0 0, first control to test: 1 3 2 0 0 0 0 0 0, res control: 0 0 1 0 0 0 0 0 0, max control: 1 3 3 0 0 0 0 0 0
```
Same as before, the emitter begins to flash and the process gets stuck again (obviously)

EmixamPP, 2022-08-04:
From what I understand, I guess that opencv is waiting for access to /dev/video2, which freezes the process. 
The emitter works because it just found the right driver, but because opencv is stuck, it can't ask the user if the driver works.

I will manually create the config file for you, in order to bypass test and the file generation. Please create the file `/etc/linux-enable-ir-emitter/video2`, which *exactly* contains :
```
device=/dev/video2
unit=10
selector=6
size=9
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
```

Then execute `sudo linux-enable-ir-emitter boot enable`. 
After that, everything should work like a charm !

But I don't understand why opencv works 6 times, but not the 7th time. For many reasons, I would like to use another library than opencv.

GeoRCode, 2022-08-05:
Yeah, that sounds pretty pretty weird.
Anyways, thanks for taking the time to help! As you said, it now apparently works like a charm. Have a good day/night ^^

EmixamPP, 2022-08-05:
You're welcome ! If you like the project, do not hesitate to star the repository to support me, thanks. 

I will let the issue open until the Ubuntu repo is fixed.

EmixamPP, 2022-08-16:
The PPA repo is now fixed (or will be soon, if the builds are not yet complete).

closed 2022-08-16 by EmixamPP