EmixamPP/linux-enable-ir-emitter #77: Configuration Error: "OSError: [Errno 8] Exec format error"

CanopusFalling, 2022-08-14:
**Ouput of `linux-enable-ir-emitter -v -d /dev/videoX configure`**

<details><summary>configure command output</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Traceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 82, in <module>
    configure.execute(device, args.limit[0])
  File "/usr/lib/linux-enable-ir-emitter/command/configure.py", line 21, in execute
    exit_code = subprocess.call([DRIVER_GENERATOR_PATH, device, str(neg_answer_limit), driver_path, str(log_level)])
  File "/usr/lib/python3.10/subprocess.py", line 345, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/usr/lib/linux-enable-ir-emitter/driver/driver-generator'
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-6):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Additional info**
 - Distro: Manjaro Linux
 - Version: 21.3.6
 - Version of linux-enable-ir-emitter: 4.1.4
 - How did you install linux-enable-ir-emitter: AUR & bash

EmixamPP, 2022-08-14:
When you say that you installed the software with "AUR & bash", you want to mean that you tried both ? And you have this same problem with the AUR package script and the bash install script ?

I've never seen this error before, so I think it's a compiling problem on your machine. 
Uninstall, then reinstall it using the bash script, and give me the outputs. I will see if something wrong happen during the compile phase.

EmixamPP, 2022-11-30:
Feel free to comment on this issue if anyone wants to reopen it.

closed 2022-11-30 by EmixamPP