EmixamPP/linux-enable-ir-emitter #78: Error configuring systemd service when using custom video custom

CharlesNRU, 2022-09-03:
I used the following command to try to configure the service: sudo linux-enable-ir-emitter -v -d /dev/video0 configure
I get the following message when I am done configuring:
```
INFO: The driver has been successfully generated.
ERROR: Error with the systemd boot service.
```

When I check the logs in journalctl, I saw the following:
```
linux-enable-ir-emitter[10416]: CRITICAL: The device /dev/video2 does not exists.
linux-enable-ir-emitter[10416]: INFO: Please choose among this list: /dev/video1 /dev/video0
```

It seems like it's not using the video0 device specified with the configure command.

EmixamPP, 2022-09-03:
Do you see other files than video0 in `/etc/linux-enable-ir-emitter/` ?

If a file video2 is in the dir, execute `sudo linux-enable-ir-emitter -d /dev/video2 delete` and then `sudo linux-enable-ir-emitter boot enable`. I think this should solve the issue.

Otherwise, can you provide the output of `linux-enable-ir-emitter boot status` and ` cat /usr/lib/systemd/system/linux-enable-ir-emitter.service`

CharlesNRU, 2022-09-03:
There is no other file than video0 in `/etc/linux-enable-ir-emitter/`

Output of `linux-enable-ir-emitter boot status`
```
CRITICAL: The device /dev/video2 does not exists.
INFO: Please choose among this list: /dev/video1 /dev/video0
```
Output of `cat /usr/lib/systemd/system/linux-enable-ir-emitter.service`
```
[Unit]
Description = enable the infrared emitter
Requires =  dev-video0.device
After = multi-user.target suspend.target hybrid-sleep.target hibernate.target suspend-then-hibernate.target dev-video0.device

[Service]
Type = oneshot
ExecStartPre = /sbin/modprobe uvcvideo
ExecStart = /usr/bin/linux-enable-ir-emitter run

[Install]
WantedBy = multi-user.target suspend.target hybrid-sleep.target hibernate.target suspend-then-hibernate.target
```

EmixamPP, 2022-09-04:
Please execute `linux-enable-ir-emitter -d /dev/video0 boot enable`, it should work

CharlesNRU, 2022-09-05:
Hi sorry for the late reply,
I get the same error message:
`ERROR: Error with the systemd boot service.`

The output of `journalctl -u linux-enable-ir-emitter.service` is the following:
```
systemd[1]: Starting linux-enable-ir-emitter.service - enable the infrared emitter...
linux-enable-ir-emitter[28624]: CRITICAL: The device /dev/video2 does not exists.
linux-enable-ir-emitter[28624]: INFO: Please choose among this list: /dev/video1 /dev/video0
systemd[1]: linux-enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: linux-enable-ir-emitter.service: Failed with result 'exit-code'.
systemd[1]: Failed to start linux-enable-ir-emitter.service - enable the infrared emitter.
```

EmixamPP, 2022-09-06:
I've committed a fix. But I don't know when all the distro repo's will be updated. 

So either you use the bash script to reinstall the software.
Or you can change the contents of `/usr/bin/linux-enable-ir-emitter` with <https://raw.githubusercontent.com/EmixamPP/linux-enable-ir-emitter/master/sources/linux-enable-ir-emitter.py>

CharlesNRU, 2022-09-06:
Hello,
I just tried the new version and it works correctly!
Thank you for the quick fix.

EmixamPP, 2022-09-14:
The 4.1.5 is published on every distro repo

closed 2022-09-14 by EmixamPP