EmixamPP/linux-enable-ir-emitter #80: Dell XPS 13 plus

epgeroy, 2022-10-06:
Hi there I'm trying to make my IR sensor working. I'm running Ubuntu 22.04 and this model is supposed to be "hardware certified"

version
```
❯ linux-enable-ir-emitter -V
linux-enable-ir-emitter 4.1.5
Developped by Maxime Dirksen - EmixamPP
MIT License
```

this is the output for the device that looks like the IR sensor
```
❯ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
```

This is what happens with I try to enable it
```
❯ linux-enable-ir-emitter -v -d /dev/video0 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```

Any help will be appreciated. Thanks in advance

epgeroy, 2022-10-06:
I just found this after

**Output of `linux-enable-ir-emitter boot status`**
```
ERROR: The boot service does not exists.
```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
no matches found: /etc/linux-enable-ir-emitter/*
```

**Give more information if you have**
check original post

**Additional info**
 - Distro: Ubuntu
 - Version: 22.04
 - Version of linux-enable-ir-emitter: 4.1.5
 - How did you install linux-enable-ir-emitter: PPA

EmixamPP, 2022-10-06:
I think I've found the problem. I've pushed a fix.

Please uninstall your current version. Then clone the git and install linux-enable-ir-emitter using the bash script, i.e. `sudo bash installer.sh install`. (g++ and libopencv-dev are required as dependencies).

Feel free to ask me any questions

EmixamPP, 2022-10-25:
Have you been able to test? Does it work? Thanks

davwheat, 2022-11-25:
Hi, I'm not the OP, but I'm trying to use this for my XPS 13 9315.

I cloned the git repo, ran the bash install script but it still results in the same error for me.

Let me know if I can help.

EmixamPP, 2022-12-05:
@davwheat Sorry for the late reply. Unfortunately, I won't have time to debug that problem until February. Which is may be caused by opencv and not by linux-enable-ir-emitter itself.

epgeroy, 2022-12-05:
I just got a notification about this. Sorry I haven't had the time to try again. I completely crashed my drivers moving stuff around, I try again after a fresh reinstall... may at the end of the month, I'll post here whatever I find.

Thank you.

EmixamPP, 2023-02-08:
Could you test the last update on the dev branch ? You have to install the OpenCV C++ libraries (`libopencv-dev` package on Ubuntu I think) in order to be able to build the software:
```
git clone https://github.com/EmixamPP/linux-enable-ir-emitter.git
cd linux-enable-ir-emitter
git checkout dev
sudo bash installer.sh install
sudo linux-enable-ir-emitter configure
```

Don't hesitate to ask me any question if you need help

epgeroy, 2023-02-08:
I did try, but it didn't work (same error). I have a fresh install OS, I'm going to try to make it work today.
I'll let you know

Thank you

EmixamPP, 2023-02-17:
The v4.4.0 is ready for release. I still have some details to work out regarding Arch. I will drop all the distributions repositories so, I've changed the way how the dependencies are managed.

You can test in advance the deb package: [downloadable here](https://github.com/EmixamPP/linux-enable-ir-emitter/releases/tag/4.4.0)

Let me know if it fix your issue, thanks !

EmixamPP, 2023-02-24:
#93 found that the command `linux-enable-ir-emitter` may be not found. 
Here is the fix: `sudo chmod +x /usr/lib/x86_64-linux-gnu/linux-enable-ir-emitter/linux-enable-ir-emitter.py`

EmixamPP, 2023-03-21:
I assume that the last release fix this problem (and I also tested it in a VM). Feel free to comment to reopen the issue

closed 2023-03-21 by EmixamPP