EmixamPP/linux-enable-ir-emitter #83: IR emitter is not flashing, just turn on. Is this normal on Linux, or it should blink like window?

ankworld, 2022-11-18:
**Ouput of `linux-enable-ir-emitter -v -d /dev/videoX configure`**

<details><summary>configure command output</summary>

```
λ sudo linux-enable-ir-emitter -v -d /dev/video2 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 4, selector: 2, cur control: 2 0, first control to test: 0 0, res control: 1 0, max control: 255 255
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 3, cur control: 0, first control to test: 1, res control: 1, max control: 255
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 4, cur control: 45 0, first control to test: 0 0, res control: 1 0, max control: 255 255
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 5, cur control: 255 0, first control to test: 0 0, res control: 1 0, max control: 255 255
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 6, cur control: 1, first control to test: 0, res control: 1, max control: 1
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 4, selector: 7, cur control: 1, first control to test: 2, res control: 1, max control: 13
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 8, cur control: 3, first control to test: 0, res control: 1, max control: 255
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 9, cur control: 0, first control to test: 1, res control: 1, max control: 1
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 4, selector: 12, cur control: 0, first control to test: 1, res control: 1, max control: 1
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 4, selector: 13, cur control: 0 0 0 0, first control to test: 1 0 0 0, res control: 1 0 0 0, max control: 255 255 255 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 14, cur control: 0 0 0 0, first control to test: 1 0 0 0, res control: 1 0 0 0, max control: 255 255 255 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
[... 199 lines cut ...]
	/dev/media1
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Additional info**
 - Distro: Ubuntu
 - Version: 22.04
 - Version of linux-enable-ir-emitter: 4.1.5
 - How did you install linux-enable-ir-emitter: PPA
 - Device: T14s Gen3 AMD
 - Info: Video is too dark.

EmixamPP, 2022-11-23:
It should blink like on Windows.

In fact, what I see is that the firmware of your camera gives almost no information on the possible values to configure it, and therefore, my software must make you test everything. 

What happens is that it doesn't offer you to test everything, otherwise it would take you a week to answer 'no'. 

At the moment, I confess that I don't know what to suggest as a solution. Maybe you could tell me after which line, i.e. `DEBUG: unit: <>, selector: <>, cur control: <>, first control to test: <>, res control: <>, max control: <>`, you see that your emitter turns on but without flashing?

In v5 of linux-enable-ir-emitter, I would like to make it 100% automatic, which would make the kind of problem you have less painful.

ankworld, 2022-11-23:
I try to test it but after run this command my webcam hang and need to reboot to make it can access again. 

```
sudo linux-enable-ir-emitter -v -d /dev/video2 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
DEBUG: unit: 4, selector: 2, cur control: 0 32, first control to test: 0 0, res control: 1 0, max control: 255 255
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 3, cur control: 0, first control to test: 1, res control: 1, max control: 255
CRITICAL: Cannot access to /dev/video2
ERROR: Impossible to reset the control.
INFO: Please keep this debug in case of issue :
DEBUG: unit: 4, selector: 3, control: 0
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```

I think I notice that in test suite there is some `DEBUG: unit: x, selector: x, cur control: x, first control to test: x, res control: x, max control: xxx` set which emitter turns on without flashing when I answer 'no'. Then, I assume that you want that value, right? 

For now, I need to fix that hang first. I can't remember how to make it run next test set like before I clean install my OS. After that, I will report debug line to you. Thanks for your attention.

EmixamPP, 2022-11-24:
If linux-enable-ir-emitter messes up your webcam you should not use it. 
But what is strange is that, I think in the past, people with a T14s Gen3 AMD had managed to set up the camera.

MK73DS, 2022-12-14:
Can it damage the IR LED if it's not blinking? I purposefully set it to **not** blink because it helps a ton with Howdy from my experience, but I don't want to burn my LED or damage my device just for that.

EmixamPP, 2022-12-14:
> Can it damage the IR LED if it's not blinking?
No don't worry.

In fact, if it doesn't blink, it has the same effect as if it had blinked only once.

EmixamPP, 2023-02-07:
You can test the last update (it may fix your issue !) on the dev branch, you have to install the OpenCV C++ libraries (`opencv-devel` package on Fedora distro based) in order to be able to build the software:
```
git clone https://github.com/EmixamPP/linux-enable-ir-emitter.git
git checkout dev
sudo bash installer.sh install
sudo linux-enable-ir-emitter -v configure
```
The software will ask you to shut down and boot and retry at some point

I could not test as I cannot reproduce this problem, to send me the outputs in case of issue

Don't hesitate to ask me any question if you need help

EmixamPP, 2023-02-08:
> INFO: Please shutdown your computer, boot and retry.
Yes you have to shut down and retry

ankworld, 2023-02-08:
Wow! It doesn't hang anymore. After long time to answer [NO] this is log when I see IR blink but not fast as windows.
```
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: minimum: 0 3 3 0 0 0 0 0 161
DEBUG: Computing the resolution control.
DEBUG: resolution: 0 0 1 0 0 0 0 0 0
DEBUG: new current: 1 3 2 0 0 0 0 0 0
DEBUG: unit: 7 selector: 6
Is the ir emitter flashing (not just turn on) ? Yes/No ? yes
INFO: The driver has been successfully generated.
Unknown device "/dev/v4l/by-path/pci-0000:34:00.0-usb-0:1:1.2-video-index0.exclude": No such device
Traceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 90, in <module>
    configure.execute(device, args.emitters[0], args.limit[0])
  File "/usr/lib64/linux-enable-ir-emitter/command/configure.py", line 28, in execute
    boot.execute("enable")
  File "/usr/lib64/linux-enable-ir-emitter/command/boot.py", line 28, in execute
    if systemd.enable():
       ^^^^^^^^^^^^^^^^
  File "/usr/lib64/linux-enable-ir-emitter/Systemd.py", line 55, in enable
    self._create_udev()
  File "/usr/lib64/linux-enable-ir-emitter/Systemd.py", line 94, in _create_udev
    kernels = get_kernels(device)
              ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/linux-enable-ir-emitter/globals.py", line 83, in get_kernels
    return subprocess.check_output("udevadm info -a %s | grep -m 1 -oP 'KERNELS==\\K.*'" % (device), shell=True).decode("utf-8").strip()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'udevadm info -a /dev/v4l/by-path/pci-0000:34:00.0-usb-0:1:1.2-video-index0.exclude | grep -m 1 -oP 'KERNELS==\K.*'' returned non-zero exit status 1
```

ankworld, 2023-02-08:
Oh, I see. 

In /etc/linux-enable-ir-emitter/ I found pci-0000:34:00.0-usb-0:1:1.2-video-index0.excluded and pci-0000:34:00.0-usb-0:1:1.2-video-index0_emitter0.driver.

- pci-0000:34:00.0-usb-0:1:1.2-video-index0.exclude is file that you create and write some content for exclude guess pattern, right? (fix hang issue)
- pci-0000:34:00.0-usb-0:1:1.2-video-index0_emitter0.driver is generated driver.

I will go to test with howdy again. Hope everything will work. Thank you for dedicating the time to this!

EmixamPP, 2023-02-08:
I will fix that very soon, this is just a problem with the boot service

EmixamPP, 2023-02-08:
The fix has been pushed, you can fetch and pull the dev branch, everything should be fine now. 
If you like the project, do not hesitate to star the repository to support me, thank you !

EmixamPP, 2023-02-09:
Feel free to comment if needed

closed 2023-02-09 by EmixamPP