EmixamPP/linux-enable-ir-emitter #84: Cannot configure on Lenovo (Yoga) Slim 7 Pro X 14ARH7

Adaoh2, 2022-12-11:
**If you had 'ERROR: Impossible to reset the control' message, give the DEBUG info displayed**
```
DEBUG: unit: 4, selector: 3, control: 0
```

**Ouput of `linux-enable-ir-emitter configure`** (used the default without the -d parameter. After the first fail I it does not ask me if I see the flashing, it just waits and gives the same error message)

<details><summary>configure command output</summary>

```
[maxime@laptop-mxm ~]$ sudo linux-enable-ir-emitter configure
[sudo] Mot de passe de maxime : 
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
CRITICAL: Cannot access to /dev/video2
ERROR: Impossible to reset the control.
INFO: Please keep this debug in case of issue :
DEBUG: unit: 4, selector: 3, control: 0
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
paste here
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
[maxime@laptop-mxm ~]$ v4l2-ctl --list-devices
Integrated RGB Camera: Integrat (usb-0000:05:00.0-1):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
```
</details>

**Additional info**
 - Distro: Fedora 37 kernel 6.0.11-300
 - Version: 4.15
 - Version of linux-enable-ir-emitter:
 - How did you install linux-enable-ir-emitter: COPR
 - Camera model: `Bus 005 Device 002: ID 30c9:0057 Luxvisions Innotech Limited Integrated RGB Camera`
 - On the first try, I get asked 6 times if I see the flasing IR blaster like on windows, (I do not, it is a continuous light on all 6 prompts) After the 6 tries when I do the configure command again it only shows me the error message that it cannot access /dev/video2 (same for video0, on this laptop it is video0 and video2 that point to the IR camera, and video1/3 are the RBG camera)
 -

EmixamPP, 2022-12-11:
Please first:
1. Shutdown your computer
2. Remove the AC adapter and if possible the battery
3. Wait one minute
4. Boot

Then, can you provide the outputs of `v4l2-ctl -d /dev/video2 --list-formats-ext`.

As well as the outputs of `linux-enable-ir-emitter -v configure`. The `-v` option is very important in order to debug the problem.

Adaoh2, 2022-12-11:
Hello, 

Here is the output of all 4 video devices:
<details><summary>/dev/video0</summary>

```bash
[maxime@laptop-mxm ~]$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1920x1080
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
```
</details>

<details><summary>/dev/video1</summary>

```bash
[maxime@laptop-mxm ~]$ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```
</details>

<details><summary>/dev/video2</summary>

```bash
[maxime@laptop-mxm ~]$ v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
```
</details>

<details><summary>/dev/video3</summary>

```
[maxime@laptop-mxm ~]$ v4l2-ctl -d /dev/video3 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```
</details>

And here is the output of `sudo linux-enable-ir-emitter -v configure`
<details><summary>configure command with verbose parameter</summary>

```bash
[maxime@laptop-mxm ~]$ sudo linux-enable-ir-emitter -v configure
[sudo] Mot de passe de maxime : 
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
DEBUG: unit: 4, selector: 2, cur control: 0 32, first control to test: 0 0, res control: 1 0, max control: 255 255
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: unit: 4, selector: 3, cur control: 0, first control to test: 1, res control: 1, max control: 255
CRITICAL: Cannot access to /dev/video2
ERROR: Impossible to reset the control.
INFO: Please keep this debug in case of issue :
DEBUG: unit: 4, selector: 3, control: 0
ERROR: The configuration has failed.
```
</details>

EmixamPP, 2022-12-13:
Ok I see. The problem is that at the moment linux-enable-ir-emitter has no "memory" to ignore the problematic instructions that disabled (or crashed) the camera.

I will think about implementing this feature. This is the first time I've seen this case. 

Unfortunately, I am going to be very busy until February.

Adesin-fr, 2022-12-21:
Hi
I also have this laptop, and have the same issue ;)
Can I help implementing this feature  ?
What kind of "memory" should be implemented ?

EmixamPP, 2023-02-07:
You can test it on the dev branch, you have to install the OpenCV C++ libraries (`opencv-devel` package on Fedora distro based) in order to be able to build the software:
```
git clone https://github.com/EmixamPP/linux-enable-ir-emitter.git
git checkout dev
sudo bash installer.sh install
sudo linux-enable-ir-emitter -v configure
```
The software will ask you to shut down and boot and retry at some point

I could not test as I cannot reproduce this problem, to send me the outputs in case of issue

Don't hesitate to ask me any question if you need help

mxmvncnt, 2023-02-07:
Thanks!

Here is the output for the initial run in the dev branch:
```INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 0 32
DEBUG: maximum: 255 255
DEBUG: resolution: 1 0
DEBUG: new current: 1 32
DEBUG: unit: 4 selector: 2
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 2 32
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 3 32
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 4 32
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 5 32
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 0
DEBUG: maximum: 255
DEBUG: resolution: 1
DEBUG: new current: 1
DEBUG: unit: 4 selector: 3
ERROR: Impossible to reset the camera.
INFO: Please shutdown your computer, boot and retry.
CRITICAL: Cannot access to /dev/v4l/by-path/pci-0000:05:00.0-usb-0:1:1.2-video-index0
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```

After rebooting, it asks me infinitely, and none of them work. Here is the output of where I am at. I did not close it yet, so if I should continue to do the answers, tell me.

<details>
    <summary>output after reboot</summary>
    
```
make : on entre dans le répertoire « /home/maxime/linux-enable-ir-emitter/sources/driver »
make: rien à faire pour « all ».
make : on quitte le répertoire « /home/maxime/linux-enable-ir-emitter/sources/driver »
'/usr/lib64/linux-enable-ir-emitter/globals.py' supprimé
'sources/globals.py' -> '/usr/lib64/linux-enable-ir-emitter/globals.py'
'/usr/lib64/linux-enable-ir-emitter/linux-enable-ir-emitter.py' supprimé
'sources/linux-enable-ir-emitter.py' -> '/usr/lib64/linux-enable-ir-emitter/linux-enable-ir-emitter.py'
'/usr/lib64/linux-enable-ir-emitter/Systemd.py' supprimé
'sources/Systemd.py' -> '/usr/lib64/linux-enable-ir-emitter/Systemd.py'
'/usr/lib64/linux-enable-ir-emitter/command/boot.py' supprimé
'sources/command/boot.py' -> '/usr/lib64/linux-enable-ir-emitter/command/boot.py'
'/usr/lib64/linux-enable-ir-emitter/command/configure.py' supprimé
'sources/command/configure.py' -> '/usr/lib64/linux-enable-ir-emitter/command/configure.py'
'/usr/lib64/linux-enable-ir-emitter/command/delete.py' supprimé
'sources/command/delete.py' -> '/usr/lib64/linux-enable-ir-emitter/command/delete.py'
'/usr/lib64/linux-enable-ir-emitter/command/__init__.py' supprimé
'sources/command/__init__.py' -> '/usr/lib64/linux-enable-ir-emitter/command/__init__.py'
'/usr/lib64/linux-enable-ir-emitter/command/run.py' supprimé
'sources/command/run.py' -> '/usr/lib64/linux-enable-ir-emitter/command/run.py'
'/usr/lib64/linux-enable-ir-emitter/driver/driver-generator' supprimé
'sources/driver/driver-generator' -> '/usr/lib64/linux-enable-ir-emitter/driver/driver-generator'
'/usr/lib64/linux-enable-ir-emitter/driver/execute-driver' supprimé
'sources/driver/execute-driver' -> '/usr/lib64/linux-enable-ir-emitter/driver/execute-driver'
ValueError: Le contexte du fichier pour /usr/lib/linux-enable-ir-emitter/driver/execute-driver n'est pas défini
ValueError: Le contexte du fichier pour /usr/lib/linux-enable-ir-emitter/driver/driver-generator n'est pas défini
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 0 32
DEBUG: maximum: 255 255
DEBUG: resolution: 1 0
[... 348 lines cut ...]
DEBUG: new current: 1 0 0 0
DEBUG: unit: 4 selector: 14
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 2 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 3 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 4 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 5 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 0
DEBUG: maximum: 3
DEBUG: resolution: 1
DEBUG: new current: 1
DEBUG: unit: 4 selector: 16
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 2
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 3

```

</details>

mxmvncnt, 2023-02-07:
OK! So little update 2 minutes after, I was very near the end of the script and it ended by itself, so here is the full output after the reboot, you can discard the one from before.

<details>
    <summary>Output after reboot</summary>
    
```
sudo bash installer.sh install
sudo linux-enable-ir-emitter -v configure
Déjà sur 'dev'
Votre branche est à jour avec 'origin/dev'.
[sudo] Mot de passe de maxime : 
make : on entre dans le répertoire « /home/maxime/linux-enable-ir-emitter/sources/driver »
make: rien à faire pour « all ».
make : on quitte le répertoire « /home/maxime/linux-enable-ir-emitter/sources/driver »
'/usr/lib64/linux-enable-ir-emitter/globals.py' supprimé
'sources/globals.py' -> '/usr/lib64/linux-enable-ir-emitter/globals.py'
'/usr/lib64/linux-enable-ir-emitter/linux-enable-ir-emitter.py' supprimé
'sources/linux-enable-ir-emitter.py' -> '/usr/lib64/linux-enable-ir-emitter/linux-enable-ir-emitter.py'
'/usr/lib64/linux-enable-ir-emitter/Systemd.py' supprimé
'sources/Systemd.py' -> '/usr/lib64/linux-enable-ir-emitter/Systemd.py'
'/usr/lib64/linux-enable-ir-emitter/command/boot.py' supprimé
'sources/command/boot.py' -> '/usr/lib64/linux-enable-ir-emitter/command/boot.py'
'/usr/lib64/linux-enable-ir-emitter/command/configure.py' supprimé
'sources/command/configure.py' -> '/usr/lib64/linux-enable-ir-emitter/command/configure.py'
'/usr/lib64/linux-enable-ir-emitter/command/delete.py' supprimé
'sources/command/delete.py' -> '/usr/lib64/linux-enable-ir-emitter/command/delete.py'
'/usr/lib64/linux-enable-ir-emitter/command/__init__.py' supprimé
'sources/command/__init__.py' -> '/usr/lib64/linux-enable-ir-emitter/command/__init__.py'
'/usr/lib64/linux-enable-ir-emitter/command/run.py' supprimé
'sources/command/run.py' -> '/usr/lib64/linux-enable-ir-emitter/command/run.py'
'/usr/lib64/linux-enable-ir-emitter/driver/driver-generator' supprimé
'sources/driver/driver-generator' -> '/usr/lib64/linux-enable-ir-emitter/driver/driver-generator'
'/usr/lib64/linux-enable-ir-emitter/driver/execute-driver' supprimé
'sources/driver/execute-driver' -> '/usr/lib64/linux-enable-ir-emitter/driver/execute-driver'
ValueError: Le contexte du fichier pour /usr/lib/linux-enable-ir-emitter/driver/execute-driver n'est pas défini
ValueError: Le contexte du fichier pour /usr/lib/linux-enable-ir-emitter/driver/driver-generator n'est pas défini
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 0 32
DEBUG: maximum: 255 255
DEBUG: resolution: 1 0
DEBUG: new current: 1 32
DEBUG: unit: 4 selector: 2
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 2 32
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 3 32
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 4 32
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 5 32
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 0 0
DEBUG: maximum: 255 255
DEBUG: resolution: 1 0
DEBUG: new current: 1 0
DEBUG: unit: 4 selector: 4
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 2 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 3 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 4 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 5 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 255 0
[... 426 lines cut ...]
DEBUG: new current: 29
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 30
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 82 0
DEBUG: maximum: 255 0
DEBUG: resolution: 1 0
DEBUG: new current: 83 0
DEBUG: unit: 5 selector: 4
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 84 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 85 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 86 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 87 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```

</details>

EmixamPP, 2023-02-07:
> After rebooting, it asks me infinitely

There is always an ending, since it have to explore all the possibilities, and your camera has a lot.

I've made a modification, could you please retry :
```
git fetch && git pull
sudo linux-enable-ir-emitter -v configure -l 1
```

I've added the parameter `-l 1` which will skip the pattern if you answered no one time, by default this is 4, and so the research will be faster. But maybe the solution could be missed, with 4 also by the way, only using 256 can ensure to miss nothing. However, from my experience, explore everything is useless. The v5 I'm working on will do a 100% automatic search, and so allow exploring everything. 

Although, your camera may require a more complex configuration, such as a combination of several instructions. This kind of case could also be done using a 100% automatic technique.

mxmvncnt, 2023-02-08:
That new version also did not work. BTW Did you see my second reply where I say I was able to go through all the possibilities one by one? I got the same result but it might be more precise than the new one.

Anyways, here is the new output

 <details>
  <summary>`sudo linux-enable-ir-emitter -v configure -l 1` output</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 0 32
DEBUG: maximum: 255 255
DEBUG: resolution: 1 0
DEBUG: new current: 1 32
DEBUG: unit: 4 selector: 2

Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 0 0
DEBUG: maximum: 255 255
DEBUG: resolution: 1 0
DEBUG: new current: 1 0
DEBUG: unit: 4 selector: 4
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 255 0
DEBUG: maximum: 255 255
DEBUG: resolution: 1 0
DEBUG: new current: 255 255
DEBUG: unit: 4 selector: 5
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 1
DEBUG: maximum: 1
DEBUG: resolution: 1
DEBUG: current: 1
DEBUG: maximum: 13
DEBUG: resolution: 1
DEBUG: new current: 2
DEBUG: unit: 4 selector: 7
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 3
DEBUG: maximum: 255
DEBUG: resolution: 1
DEBUG: new current: 4
DEBUG: unit: 4 selector: 8
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 0
DEBUG: maximum: 1
DEBUG: resolution: 1
DEBUG: new current: 1
DEBUG: unit: 4 selector: 9
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 0
DEBUG: maximum: 1
DEBUG: resolution: 1
DEBUG: new current: 1
DEBUG: unit: 4 selector: 12
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 0 0 0 0
DEBUG: maximum: 255 255 255 0
DEBUG: resolution: 1 0 0 0
DEBUG: new current: 1 0 0 0
DEBUG: unit: 4 selector: 13
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
[... 220 lines cut ...]
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 0 0 0 0
DEBUG: maximum: 255 255 255 255
DEBUG: resolution: 1 0 0 0
DEBUG: new current: 1 0 0 0
DEBUG: unit: 4 selector: 27
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 25
DEBUG: maximum: 100
DEBUG: resolution: 1
DEBUG: new current: 26
DEBUG: unit: 5 selector: 2
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 92 0
DEBUG: maximum: 255 0
DEBUG: resolution: 1 0
DEBUG: new current: 93 0
DEBUG: unit: 5 selector: 4
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
ERROR: The configuration has failed.
```
</details>

EmixamPP, 2023-02-08:
With the current version I can't do anything more for your problem. But I keep your issue in a corner of my head. When v5 will be ready, I hope it will be able to activate your camera

EmixamPP, 2023-08-01:
Hey, I'm coming back to you with a new version that might be able to configure your camera.

The installation method has changed, I no longer maintain the Fedora COPR, but a unique tarball.
There is a comment from a bot in pull request https://github.com/EmixamPP/linux-enable-ir-emitter/pull/121, the installation instructions for this pre-version are specified in it.

To configure your camera, run the command `sudo linux-enable-ir-emitter -v configure -l -1`, and past the output here. 
This runs a fully exhaustive automatic configuration. If the tool thinks it has detected that the transmitter is blinking, it will ask you to confirm.

This can be a long process, but, since all you have to do is confirm whether it works, you don't have to be in the terminal all the time. You can let it search on the side, and cross your fingers, because this time (unless there's a bug in this pre-version) the tool will try everything it can try. And as a result, I don't think I'll be able to make a future version that will activate your emitter. 

If the transmitter is flashing and the terminal hasn't asked for confirmation, you'll have to switch to manual mode with `-m`.

If you no longer wish to configure your infrared camera, no problem, please just close the issue.

mxmvncnt, 2023-08-01:
Hey, I am currently trying this, it seems it will take a very long time, doing 0-256 256 (I assume it is 256) times. Do you have a Matrix channel (or other ways of contact)? For now I am at the 200 2 attempt, still nothing but I guess there is hope!

EmixamPP, 2023-12-30:
Hey, please try the v6 https://github.com/EmixamPP/linux-enable-ir-emitter/pull/154. There is a bot comment with a link to the tarball. Please uninstall any existing version first:
```
sudo rm -rf /usr/lib64/linux-enable-ir-emitter \
/usr/bin/linux-enable-ir-emitter \
/usr/lib/systemd/system/linux-enable-ir-emitter.service \
/etc/udev/rules.d/99-linux-enable-ir-emitter.rules \
/usr/share/bash-completition/completitions/linux-enable-ir-emitter \
/etc/linux-enable-ir-emitter
```

Execute `sudo linux-enable-ir-emitter -v configure`
And past here the content of the file which is inside `/etc/linux-enable-ir-emitter`, from this point I can explain you how to manually tweak any instructions of your camera.

mxmvncnt, 2023-12-31:
Hello! I have just ran the software again, and at the step before the last one, I saw flashing! This is certainly a good sign. I have pasted below both the output of the software and the file in /etc/

<details><summary>/etc/linux-enable-ir-emitter/pci-0000\:05\:00.0-usbv2-0\:1\:1.2-video-index0</summary>

```
- corrupted: false
  unit: 4
  selector: 2
  current:
    - 39
    - 0
  maximum:
    - 255
    - 255
  minimum:
    - 0
    - 0
- corrupted: false
  unit: 4
  selector: 3
  current:
    - 54
  maximum:
    - 255
  minimum:
    - 0
- corrupted: false
  unit: 4
  selector: 4
  current:
    - 0
    - 0
  maximum:
    - 255
    - 255
  minimum:
    - 0
    - 0
- corrupted: false
  unit: 4
  selector: 5
  current:
    - 255
    - 0
  maximum:
    - 255
    - 255
  minimum:
    - 0
    - 0
- corrupted: false
  unit: 4
  selector: 6
  current:
    - 1
  maximum:
    - 1
  minimum:
    - 0
- corrupted: false
  unit: 4
  selector: 7
  current:
    - 1
  maximum:
    - 13
  minimum:
    - 1
- corrupted: false
  unit: 4
[... 2200 lines cut ...]
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 24 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 25 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 26 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 27 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 28 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 29 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 30 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 31 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 32 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 33 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 34 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 35 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 3, control: 16 0 0 0 0 0 0 36 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 6, control: 1 3 2 0 0 0 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 6, control: 1 3 3 0 0 0 0 0 0
DEBUG: Instruction applied: unit: 7, selector: 9, control: 1 0 0 0
ERROR: The configuration has failed.
INFO: Please retry in manual mode by adding the '-m' option.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

</details> 

I am pretty sure the `unit: 7, selector: 6, control: 1 3 3 0 0 0 0 0 0` instruction is what made the IR light blink, but it could also be the one before.

EmixamPP, 2023-12-31:
> I am pretty sure the unit: 7, selector: 6, control: 1 3 3 0 0 0 0 0 0 instruction is what made the IR light blink, but it could also be the one before.

Oh nice!
Please first update the tool again (I made a fix commit this morning).
Execute `sudo linux-enable-ir-emitter tweak`, you will see a line `x) unit: 7, selector: 6, control: 1 3 1 0 0 0 0 0 0`.
Input the corresponding number, then type `1 3 3 3 0 0 0 0 0 0` and press enter. Then you should see the ir emitter flashing.

> Also, do you have any idea if the drivers generated from your project could in any way be upstreamed if they are confirmed to work and have been tested? This could allow to unlock many laptops for many installs if only one person uses this program

This will work for a few people with **exactly** the same computer; so maybe another Lenovo (Yoga) Slim 7 Pro X 14ARH7 will not work using your configuration file, often the unit and selector are not the same.

At first, since the configuration was done by sniffing a Windows VM, I asked people to bring up their configuration, but because of the problem I mentioned, I created this tool which normally enables any emitter with a single command.

mxmvncnt, 2023-12-31:
Thanks for the help! It turns out, the right instruction was `1 3 2 0 0 0 0 0 0`. However, it seems like the driver does not persist after a few minutes. I was able to setup Howdy, but after a few minutes or after a reboot, the camera stops flashing which is a little bit weird.

EmixamPP, 2024-01-01:
Execute `sudo linux-enable-ir-emitter boot enable`

mxmvncnt, 2024-01-01:
Thanks! This seems to have worked to make the config persistent. Since this works, I think I will close the issue. Happy new year!

closed 2024-01-01 by mxmvncnt

EmixamPP, 2024-01-02:
Nice to hear! Happy new year!