EmixamPP/linux-enable-ir-emitter #86: Ir emitter stopped working on Lenovo X1 gen 9.

spiftire, 2023-01-03:
**Output of `linux-enable-ir-emitter boot status`**
```
× linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/lib/systemd/system/linux-enable-ir-emitter.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-01-03 08:24:15 CET; 1min 2s ago
    Process: 13024 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 13025 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=1/FAILURE)
   Main PID: 13025 (code=exited, status=1/FAILURE)
        CPU: 31ms

Jan 03 08:24:15 steffen-popos systemd[1]: Starting enable the infrared emitter...
Jan 03 08:24:15 steffen-popos linux-enable-ir-emitter[13025]: ERROR: Impossible to execute the driver at /etc/linux-enable-ir-emitter/video2.
Jan 03 08:24:15 steffen-popos systemd[1]: linux-enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE
Jan 03 08:24:15 steffen-popos systemd[1]: linux-enable-ir-emitter.service: Failed with result 'exit-code'.
```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
device=/dev/video2
unit=14
selector=6
size=9
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
```

**Give more information if you have**

Ir emitting have been working perfectly till it suddenly dident. I had just done a firmware update when it stopped working, but after downgrading the firmware the problem seemed to remain.

**Additional info**
 - Distro: pop_OS
 - Version: 22.04
 - Version of linux-enable-ir-emitter: linux-enable-ir-emitter 4.1.5
 - How did you install linux-enable-ir-emitter:  ppa

EmixamPP, 2023-01-07:
Did you try to reconfigure the camera? Can you paste the outputs of `sudo linux-enable-ir-emitter -v -d /dev/video2 configure`
And also, ensure that the infrared camera still on /dev/video2

spiftire, 2023-01-10:
I don't know what fixed the problem, but it seem to be working as expected now

closed 2023-01-10 by spiftire