EmixamPP/linux-enable-ir-emitter #87: Cannot Install on OpenSUSE Tumbleweed

meechgalhuquot, 2023-01-14:
I can add the repo to Tumbleweed from the .repo file that is on the COPR page when I manually put in the url from that file, [https://download.copr.fedorainfracloud.org/results/emixampp/linux-enable-ir-emitter/opensuse-tumbleweed-$basearch/](https://download.copr.fedorainfracloud.org/results/emixampp/linux-enable-ir-emitter/opensuse-tumbleweed-$basearch/), however when I try to install the package  I get the attached error. I have installed a 
[conflicts.txt](https://github.com/EmixamPP/linux-enable-ir-emitter/files/10417288/conflicts.txt)
I can install opencv through the repos on my device, but it specifically wants that libopencv_aruco.so.406. When I attempt to build the package manually it also fails, just in a different way

EmixamPP, 2023-01-16:
I'm not sure to understand

Could you first remove the repo: `sudo rm /etc/yum.repos.d/_copr:copr.fedorainfracloud.org:emixampp:linux-enable-ir-emitter.repo`

Then, could you give me the outputs of the following commands: 
1. `sudo dnf copr enable emixampp/linux-enable-ir-emitter`
2. `sudo dnf --refresh install linux-enable-ir-emitter`

meechgalhuquot, 2023-01-16:
OpenSUSE does not use `dnf`, while it uses RPM packages the package manager is `zypper` which does not have `copr` built into the command

EmixamPP, 2023-01-16:
Yes indeed! I forgot.
Maybe try to first install `opencv` (as you already did) but also `opencv-devel`. Then install `linux-enable-ir-emitter`, it could be that the dependency is in the devel package.

Otherwise, just build and install it manually using the bash script of the git repo; everything is explained in the readme.

meechgalhuquot, 2023-01-17:
[errorlog.txt](https://github.com/EmixamPP/linux-enable-ir-emitter/files/10435688/errorlog.txt)
[log.txt](https://github.com/EmixamPP/linux-enable-ir-emitter/files/10435689/log.txt)
Rather than trying with the repo again since it wasn't working I tried to build and install manually after ensuring that `opencv` and `opencv-devel` were installed and these are the logs I got when running `sudo bash installer.sh install 2> errorlog.txt > log.txt`

EmixamPP, 2023-01-21:
May be, you could try to find on your computer if a file `libopencv_aruco.so*` exists (i.e. `find / -name libopencv_aruco.so* 2> /dev/null`). If so, you could try to create a symbolic link named : `libopencv_aruco.so.406` to that file, which in fact is from another version. Most likely, this will not be a problem.

Otherwise, I can do nothing for you, this seems to be an issue of the opencv package on Tumbleweed.

EmixamPP, 2023-02-01:
Does this trick work?

meechgalhuquot, 2023-02-03:
I cannot find the file with find, however I can see that it is installed in the software list. I wonder if adding the packages to the openSUSE Build Service would resolve issues as then it may auto solve dependencies and could be installed by using the command `opi <packagename>`
[image]

EmixamPP, 2023-02-05:
> I cannot find the file with find

This is strange because, if the aruco package in version 407 is installed, the file `libopencv_aruco.so407` should be present on the system 

Maybe try to install the libopencv3_4 package ?

Honestly, I can't do much about this compilation problem with opencv.

EmixamPP, 2023-02-06:
I ran a new build on COPR, but the build fails because of the server, I will try again later.

You will only have to download again the .rpm.  I hope it will work!

EmixamPP, 2023-02-17:
The v4.4.0 is ready for release. I still have some details to work out regarding Arch. I will drop all the distributions repositories so, I've changed the way how the dependencies are managed.

You can test in advance the rpm package [downloadable here](https://github.com/EmixamPP/linux-enable-ir-emitter/releases/tag/4.4.0)

Let me know if it fix your issue, thanks !

EmixamPP, 2023-02-20:
The v4.4.0 has been released and should fix this issue. 
Feel free to comment if needed

meechgalhuquot, 2023-02-21:
[image]

meechgalhuquot, 2023-02-21:
I see I have libopencv 407 packages instead of 406, could that be the issue?, is it calling for 406 specifically? Or is this just a quirk of OpenSUSE package management do you think 
[image]

EmixamPP, 2023-02-21:
Hum, the package contains all the needed dependencies. Can you add `--nodeps` to the rpm command ?

meechgalhuquot, 2023-02-21:
[image]

EmixamPP, 2023-02-22:
This time it should be fine :crossed_fingers:

meechgalhuquot, 2023-02-22:
[image]
Progress!!

EmixamPP, 2023-02-22:
Can you past the outputs of `sudo linux-enable-ir-emitter boot status` ?

EmixamPP, 2023-02-22:
miss click the close

meechgalhuquot, 2023-02-22:
[ir-emitter.log](https://github.com/EmixamPP/linux-enable-ir-emitter/files/10804385/ir-emitter.log)

EmixamPP, 2023-02-22:
Thanks, it is fixed

meechgalhuquot, 2023-02-22:
Fantastic, it appears to be working now!! Thank you for working on this!

closed 2023-02-22 by meechgalhuquot

EmixamPP, 2023-02-22:
You're welcome. If you like the project, do not hesitate to star the repository to support me, thank you !