EmixamPP/linux-enable-ir-emitter #89: Dell Latitude 7280 Configuration has failed

WillMatthews, 2023-01-19:
Hello!
Thank you for your work on this repo. I'm very keen to use it, however I'm struggling to make the IR LED on my laptop work, I get stuck at the configuration stage sadly.
I hope I've provided all the information needed - if there's anything I can do to help or experiment with please let me know and I will respond promptly.
Thank you :)

```
DEBUG:
```

**Ouput of `sudo linux-enable-ir-emitter -v -d /dev/video2 configure`**
<details><summary>configure command output</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit: 10, selector: 10, cur control: 0 8 0 187 187 187 187 187, first control to test: 0 0 0 0 0 0 0 0, res control: 1 0 0 0 0 0 0 0, max control: 255 255 255 255 255 255 255 255
DEBUG: Negative answer limit exceeded, skipping the pattern.
ERROR: Impossible to reset the control.
INFO: Please keep this debug in case of issue :
DEBUG: unit: 10, selector: 10, control: 0 8 0 187 187 187 187 187
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki

```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-5):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/media0
        /dev/media1
```

**Ouput of `v4l2-ctl -d /dev/video2 --list-formats-ext`**
<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 340x374
                        Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Additional info**
 - Distro: Kubuntu 
 - Version: 22.04.1 LTS x86_64
 - Version of linux-enable-ir-emitter: 4.1.5
 - How did you install linux-enable-ir-emitter: PPA
 
Attempted with multiple kernel versions. Currently using: Linux 6.1.7-060107-generic x86_64. Same output for all cases.

Laptop is a Dell Latitude 7280. IR LED was checked with a camera tested and known to show IR LED light.
linux-enable-ir-emitter when tested on /dev/video0 caused the visible white indicator LED to flash, but when repeated for the IR camera, the IR LED never lit (using /dev/video2).

**Relevant `lsusb` output**
<details><summary>Click to expand</summary>

```
Bus 001 Device 002: ID 0bda:58c7 Realtek Semiconductor Corp. Integrated_Webcam_HD
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
</details>

Tested all devices in /dev/videoX (and even /dev/mediaX out of blind hope). /dev/video2 is the only case where there is an IR camera which opens with ffplay.

Followed all wiki instructions - even disconnecting the battery for 1 minute and attempting again.

EmixamPP, 2023-01-19:
Hello, I'm currently quite busy, but I will look into it in two weeks! (I think I have an idea for a fix)

WillMatthews, 2023-02-07:
Hi there!
Thanks for getting back to me :)

I tried to get it working by including opencv in the pkg-config search path;
pkg-config --cflags opencv4
pkg-config --libs opencv4
and then running the install.sh

This failed as opencv,pc weirdly doesn't exist on my computer.
This might be because it's opencv4, which is the only version of opencv from apt on Ubuntu 22.04. (?)

I'll poke around a bit more in the meantime, but do you have any suggestions on how to proceed?

I'll put an error log up soon.
Thanks!

WillMatthews, 2023-02-07:
Hi!
Thanks for finding some time to work on this. 
I did as you suggested, and I get the following log.

<details><summary>Click to expand</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 0 8 0 187 187 187 187 187
DEBUG: maximum: 255 255 255 255 255 255 255 255
DEBUG: resolution: 1 0 0 0 0 0 0 0
DEBUG: new current: 1 8 0 187 187 187 187 187
DEBUG: unit: 10 selector: 10
DEBUG: new current: 2 8 0 187 187 187 187 187
DEBUG: new current: 3 8 0 187 187 187 187 187
DEBUG: new current: 4 8 0 187 187 187 187 187
DEBUG: new current: 5 8 0 187 187 187 187 187
DEBUG: Negative answer limit exceeded, skipping the pattern.
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```
</details>

I've been trying to figure out how it works so I can have a try too when time permits.

EmixamPP, 2023-02-08:
Um, it looks like your infrared camera doesn't give any information on how it can be configured. 

Here we can observe that your camera indicates only one possible instruction. However, 4 times in a row it refused to apply a modification on the instruction; in order to find the right one, which enable the emitter.

May be try with this command: `sudo linux-enable-ir-emitter -v configure -l 256`
The option `-l 256` will avoid the "Negative answer limit exceeded, skipping the pattern.", and perform a full exploration of the possibilities. 

Otherwise, unfortunately, I don't think I can do anything about it.

WillMatthews, 2023-02-11:
Whoops, thought I sent this comment earlier but I didn't.
I tried again with -l 256 but unfortunately for the instructions it tried the LED didn't light.

WillMatthews, 2023-07-20:
Hi, thanks for your work, I had no issues getting linux-enable-ir-emitter installed.

Unfortunately, I still couldn't get the IR emitter to flash or even turn on.

<details><summary>Here is the dump from my terminal</summary>
<p>

```
sudo linux-enable-ir-emitter -v -d /dev/video2 configure -m -x
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on)? Yes/No? n
DEBUG: Extension units: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 197 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 
DEBUG: current: 0 8 0 187 187 187 187 187
DEBUG: maximum: 255 255 255 255 255 255 255 255
DEBUG: minimum: 0 0 0 0 0 0 0 0
DEBUG: resolution: 1 0 0 0 0 0 0 0
DEBUG: unit: 10 selector: 10
DEBUG: new current: 0 0 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 0
DEBUG: Negative answer limit exceeded, skipping the pattern.
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

</p>
</details> 

Thank you once again, when I have time I might try digging more but I'm stretched very thin at the moment.
I'll be more than happy to test any future versions :)

While I'm here, is there an uninstaller?

EmixamPP, 2023-07-20:
`sudo rm -rf /usr/lib64/linux-enable-ir-emitter \ /usr/libexec/linux-enable-ir-emitter \ /usr/bin/linux-enable-ir-emitter \ /usr/lib/systemd/system/linux-enable-ir-emitter.service \ /etc/udev/rules.d/99-linux-enable-ir-emitter.rules \ /usr/share/bash-completition/completitions/linux-enable-ir-emitter \ /etc/linux-enable-ir-emitter`

EmixamPP, 2023-08-01:
I've just finished the exhaustive search, please refer again to the pr for installation.

To configure your camera, run the command `sudo linux-enable-ir-emitter -v configure -l -1`, and past the output here. 
This runs a fully exhaustive automatic configuration. If the tool thinks it has detected that the transmitter is blinking, it will ask you to confirm.

This can be a long process, but, since all you have to do is confirm whether it works, you don't have to be in the terminal all the time. You can let it search on the side, and cross your fingers, because this time (unless there's a bug in this pre-version) the tool will try everything it can try. And as a result, I don't think I'll be able to make a future version that will activate your emitter. 

If the transmitter is flashing and the terminal hasn't asked for confirmation, you'll have to switch to manual mode with `-m`.

EmixamPP, 2023-10-09:
I'm closing the issue, but if you still want to try configuring your camera, don't hesitate to leave a comment. I'll be happy to help you further.

WillMatthews, 2023-10-09:
Thanks,I tried the exhaustive search but it ended up erroring out. I’m quite strapped for time at the moment so I might have another look in a few months.Many thanks again - sorry we didn’t get anywhere :'(William__________________________      ▸   William A. Matthews   MEng (Oxon)    ▸   DPhil Candidate, University College, Oxford     ▸   mob: +44 7516 175554     ▸   ***@***.***     ▸   web: willmatthews.xyzSent From M0

EmixamPP, 2023-12-30:
Hey, please try the v6 https://github.com/EmixamPP/linux-enable-ir-emitter/pull/154. There is a bot comment with a link to the tarball. Please uninstall any existing version first:
```
sudo rm -rf /usr/lib64/linux-enable-ir-emitter \
/usr/bin/linux-enable-ir-emitter \
/usr/lib/systemd/system/linux-enable-ir-emitter.service \
/etc/udev/rules.d/99-linux-enable-ir-emitter.rules \
/usr/share/bash-completition/completitions/linux-enable-ir-emitter \
/etc/linux-enable-ir-emitter
```

Execute `sudo linux-enable-ir-emitter -v configure`
And past here the content of the file which is inside `/etc/linux-enable-ir-emitter`

closed 2024-02-02 by EmixamPP