EmixamPP/linux-enable-ir-emitter #90: Supporting multiple IR emitters

OwenJPage, 2023-01-21:
My laptop has two IR emitters (either side of the IR camera), however currently, only the left emitter will flash when the tool is configured. Is there a way to enable both IR emitters to work at once? If not, could support for this be added in the future?

EmixamPP, 2023-01-21:
Can you past the output of `cat /dev/video*` ?

OwenJPage, 2023-01-24:
```
cat: /dev/video0: Invalid argument
cat: /dev/video1: Invalid argument
cat: /dev/video2: Invalid argument
cat: /dev/video3: Invalid argument
```
`/dev/video2` is the IR camera itself; testing using `cheese -d /dev/videoX`, `video0`, `video2`, and `video3` appear to be normal cameras.

EmixamPP, 2023-01-24:
Hum ok I see. I will look into it in the coming weeks, but I can't make any promises (especially since I cannot test solutions).

OwenJPage, 2023-01-24:
Thanks! No worries if you can't find a solution, it does mostly work (with howdy) using just the one emitter, I just imagine it'd be somewhat better to have both working.

OwenJPage, 2023-01-24:
I used `linux-enable-ir-emitter`, it wasn't working natively.

EmixamPP, 2023-02-05:
If you perform the configuration twice and answer no every time until the second emitter blink.
Do you find a configuration where the second one is working ? Or does it terminate without success ?

OwenJPage, 2023-02-06:
It is able to enable the other transmitter, but as far as I could tell, not both at once

EmixamPP, 2023-02-07:
Perfect, the next update will support multi-emitter cameras.
I will probably release a first beta version at the end of the week

OwenJPage, 2023-02-07:
That's amazing, thank you!

EmixamPP, 2023-02-07:
I leave the issue open so that we can discuss it in case of problems (unfortunately I can't guarantee that it will work).
I'll let you know when you can test the update

EmixamPP, 2023-02-07:
In fact, I think everything is good.
You can test it on the dev branch, you have to install the OpenCV C++ libraries in order to be able to build the software:
```
git clone https://github.com/EmixamPP/linux-enable-ir-emitter.git
git checkout dev
sudo bash installer.sh install
sudo linux-enable-ir-emitter configure -e 2
```
Answer yes every time you see one emitter (or both at the same time) blinking.
Reboot and let me know if everything works 

Don't hesitate to ask me any question if you need help

EmixamPP, 2023-02-17:
The v4.4.0 is ready for release. I still have some details to work out regarding Arch. If you wish, I can provide you a .deb or .rpm package in advance.

EmixamPP, 2023-02-20:
The v4.4.0 has been released and should support that feature. 
Feel free to comment in case of issue

closed 2023-02-20 by EmixamPP