EmixamPP/linux-enable-ir-emitter #91: Emitter does not work after successful configuration on Lenovo P14s Gen2 AMD

Johannbr, 2023-01-29:
**Output of `linux-enable-ir-emitter boot status`**
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/lib/systemd/system/linux-enable-ir-emitter.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2023-01-29 23:41:54 CET; 3min 13s ago
    Process: 76633 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 76634 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 76634 (code=exited, status=0/SUCCESS)
        CPU: 62ms

janv. 29 23:41:54 johann systemd[1]: Starting enable the infrared emitter...
janv. 29 23:41:54 johann systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
janv. 29 23:41:54 johann systemd[1]: Finished enable the infrared emitter.

```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
device=/dev/video2
unit=14
selector=6
size=9
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0

```

**Give more information if you have**
Output of ls -l /dev/v4l/by-path:
```
lrwxrwxrwx 1 root root 12 janv. 29 23:41 pci-0000:06:00.0-usb-0:2:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 janv. 29 23:41 pci-0000:06:00.0-usb-0:2:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 janv. 29 23:41 pci-0000:06:00.0-usb-0:2:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 janv. 29 23:41 pci-0000:06:00.0-usb-0:2:1.2-video-index1 -> ../../video3
```

Output of v4l2-ctl --list-devices:
```
Integrated Camera: Integrated C (usb-0000:06:00.0-2):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```

**Additional info**
 - Distro: Ubuntu
 - Version: 22.04.1
 - Version of linux-enable-ir-emitter: 4.1.5
 - How did you install linux-enable-ir-emitter: PPA

EmixamPP, 2023-01-31:
If you execute `linux-enable-ir-emitter run` after the boot, does the emitter work ?

Johannbr, 2023-01-31:
Hi, thanks for the reply.

I just did: 
`sudo linux-enable-ir-emitter boot enable`
Output
`INFO: The boot service have been enabled.`

And `linux-enable-ir-emitter run` with no output

Then, `sudo howdy test `
=> camera is working but the IR emitter does not work.

EmixamPP, 2023-02-01:
Is howdy using the /dev/video2 camera ?

Johannbr, 2023-02-01:
The device_path used by howdy is:
`device_path = /dev/v4l/by-path/pci-0000:06:00.0-usb-0:2:1.0-video-index0`
which is: 
`/dev/video0`

EmixamPP, 2023-02-01:
Please change the variable and use /dev/video2. In fact there is no issue :)
/dev/video0 is your regular camera

Johannbr, 2023-02-01:
Indeed, the emitter is on now.
Thanks for your time.

closed 2023-02-01 by Johannbr