EmixamPP/linux-enable-ir-emitter #93: linux-enable-ir-emitter: command not found

zq418, 2023-02-24:
**Describe the bug**

**How to reproduce**

**Additional info**
 - Distro: Ubuntu
 - Version: 22.04.2LTS
 - Version of linux-enable-ir-emitter: The latest deb
 - How did you install linux-enable-ir-emitter: 

After installation, I ran `sudo linux-enable-ir-emitter configure` but got `sudo: linux-enable-ir-emitter: command not found` error. It turned out that `/usr/lib/x86_64-linux-gnu/linux-enable-ir-emitter/linux-enable-ir-emitter.py` was not executable. Running the following command fixed it.

```bash
sudo chmod +x /usr/lib/x86_64-linux-gnu/linux-enable-ir-emitter/linux-enable-ir-emitter.py
```

EmixamPP, 2023-02-24:
Thanks for the discovery and the fix, strangely it didn't happen during the tests. Maybe because of the version of Meson on your system.

closed 2023-02-24 by EmixamPP

Woofweasel, 2024-04-07:
hey! I'm on fedora silverblue 39, and I have this same issue, except when manually building the package I had to use meson to change the installation directory to $HOME/.local rather than /usr/local, as suggested here. therefore, the command suggested here does not work for me. I cannot find the equivalent file within the /.local/lib64/linux-enable-ir-emitter directory on my own pc. can anyone suggest an alternate solution for the issue?

EmixamPP, 2024-04-07:
@Woofweasel, please open a new issue for your case, I will help you to find out the problem. 
By the way, when you execute `ninja install -C build`, you can see where are installed the different files.