EmixamPP/linux-enable-ir-emitter #94: Lenovo 510 webcam

rdeng4, 2023-02-25:
**Ouput of `linux-enable-ir-emitter -v -d /dev/videoX configure`**

<details><summary>configure command output</summary>

```
sudo linux-enable-ir-emitter -v -d /dev/video2 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: resolution: 0 0 0 0 0 0 0 0 0
DEBUG: new current: 1 3 1 0 0 0 0 0 0
DEBUG: unit: 2 selector: 6
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 1 3 1 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 1 3 1 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 1 3 1 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 1 3 1 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Negative answer limit exceeded, skipping the pattern.
DEBUG: current: 16 0 0 0
DEBUG: maximum: 16 0 0 0
DEBUG: resolution: 0 0 0 0
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Lenovo 510 Camera: Lenovo 510 R (usb-0000:00:14.0-11.4):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
```
</details>

**Additional info**
 - Distro:  Pop OS
 - Version:  22.04
 - Version of linux-enable-ir-emitter: 4.4.0
 - How did you install linux-enable-ir-emitter:  deb

svenmang2, 2023-02-26:
I also have this issue on /dev/video2 

I have the same web cam mentioned above

v4l2-ctl -d /dev/video2 --list-ctrls                                                 1 ✘  13s  

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=100 value=100
                            hue 0x00980903 (int)    : min=-64 max=64 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=1 max=100 step=1 default=50 value=50
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=2 (60 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4000 value=4000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=12 step=1 default=6 value=6
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=1 value=1

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=3 max=500 step=1 default=20 value=20 flags=inactive
                   pan_absolute 0x009a0908 (int)    : min=-57600 max=57600 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-57600 max=57600 step=3600 default=0 value=0
                  zoom_absolute 0x009a090d (int)    : min=100 max=400 step=1 default=100 value=100
    ~ 

EmixamPP, 2023-02-26:
@svenmang2 did you install the 4.5.0 ?

svenmang2, 2023-02-26:
yes i did

sudo linux-enable-ir-emitter -V                                                         ✔  4s  
linux-enable-ir-emitter 4.5.0
Developped by Maxime Dirksen - EmixamPP
MIT License

EmixamPP, 2023-02-26:
Can you past give the outputs of `linux-enable-ir-emitter -v configure` ? Thanks

svenmang2, 2023-02-26:
sudo linux-enable-ir-emitter -v configure                                                      ✔ 
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
DEBUG: Computing the resolution control.
DEBUG: Minimum not consistent, it will be ignored.
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: resolution: 0 0 1 0 0 0 0 0 0
DEBUG: unit: 2 selector: 6
DEBUG: new current: 1 3 2 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
DEBUG: new current: 1 3 3 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki

EmixamPP, 2023-02-26:
Ok so the fix I made worked. But it doesn't seem to configure the camera.

Honestly, I don't really know what else to do.
Maybe, the v5 which will introduce the exhaustive search of all possibilities, will be able to configure the camera.
But I still haven't started the development, and I can't give you a date.

EmixamPP, 2023-02-26:
@rdeng4 you should also try the 4.5.0, we never know :crossed_fingers:

rdeng4, 2023-02-26:
> 

Hi, just tested quickly and I had the same result as @svenmang2 .
Thanks for the work :smile:

svenmang2, 2023-02-26:
I was configuring mouse buttons using x input and this caught my attention
  xinput list                                                                                      1 ✘ 
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver Keyboard            id=13   [slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver                     id=15   [slave  pointer  (2)]
⎜   ↳ Logitech G903 LS                          id=16   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Power Button                              id=8    [slave  keyboard (3)]
    ↳ Lenovo 510 Camera: Lenovo 510 R           id=9    [slave  keyboard (3)]
    ↳ Lenovo 510 Camera: Lenovo 510 I           id=10   [slave  keyboard (3)]
    ↳ Asus WMI hotkeys                          id=11   [slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                        id=12   [slave  keyboard (3)]
    ↳ Logitech USB Receiver Keyboard            id=14   [slave  keyboard (3)]
    ↳ Logitech G903 LS                          id=17   [slave  keyboard (3)]

Dk if its usefull but it sees buttons on the device? maby that's supposed to be there

EmixamPP, 2023-08-01:
Hey, I'm coming back to you with a new version that might be able to configure your camera.

The installation method has changed, I no longer maintain a deb package, but a unique tarball.
There is a comment from a bot in pull request https://github.com/EmixamPP/linux-enable-ir-emitter/pull/121, the installation instructions for this pre-version are specified in it.

To configure your camera, run the command `sudo linux-enable-ir-emitter -v configure -l -1`, and past the output here. 
This runs a fully exhaustive automatic configuration. If the tool thinks it has detected that the transmitter is blinking, it will ask you to confirm.

This can be a long process, but, since all you have to do is confirm whether it works, you don't have to be in the terminal all the time. You can let it search on the side, and cross your fingers, because this time (unless there's a bug in this pre-version) the tool will try everything it can try. And as a result, I don't think I'll be able to make a future version that will activate your emitter. 

If the transmitter is flashing and the terminal hasn't asked for confirmation, you'll have to switch to manual mode with `-m`.

If you no longer wish to configure your infrared camera, no problem, please just close the issue.

svenmang2, 2023-08-01:
Hey man,

Thanks for getting back to me unfortunately iam no longer using Linux on
the pc the games I wanted to play just refused to play unfortunately

Thank you for the update,

I love to see people like yourself help out the linux community!

rdeng4, 2023-08-02:
Hi,

Thanks for the update. Unfortunately my Lenovo ir camera was lost when I moved, so I have no hardware to test on. I'm sure it will help other people out there though. Thanks for sharing again.

EmixamPP, 2023-08-02:
No problem!