EmixamPP/linux-enable-ir-emitter #95: IR emitter stops working every couple of boots on Dell XPS 9520.

nikfilippas, 2023-02-27:
**Output of `linux-enable-ir-emitter boot status`**
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/lib/systemd/system/linux-enable-ir-emitter.service; enabled; vendor preset: enabled)
     Active: inactive (dead)

Feb 27 13:47:55 canopus systemd[1]: Dependency failed for enable the infrared emitter.
Feb 27 13:47:55 canopus systemd[1]: linux-enable-ir-emitter.service: Job linux-enable-ir-emitter.service/start failed with result 'dependency'.
```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
device=/dev/video2
unit=4
selector=6
size=9
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
```

**Give more information if you have**

IR blinkers only work with Howdy after I run `linux-enable-ir-emitter`.
Then, I ran `linux-enable-ir-emitter boot enable`, and changed `Type = oneshot` to `Type = forking` in `/usr/lib/systemd/system/linux-enable-ir-emitter.service` as per the suggestion on #48, because out of the box it only worked for the current session and stopped working at boot.
This works perfectly, with Howdy configured as needed, and the IR blinkers work on boot. However, every couple of boots, they stop working and I have to manually re-configure the service from the start. I have no idea what is causing the issue. This seems to happen on its own - I don't touch any config file or settings, or update my system in the meantime. It just intermittently fails.

When it is properly configured, I test howdy and the pictures it takes are clear and sharp, so almost certainly this is not a problem with howdy's configuration. The issue is with the IR emitter failing to turn on after some boots.

**Additional info**
 - Computer (or camera) model: Dell XPS 9520
 - Version of linux-enable-ir-emitter:  4.1.5

EmixamPP, 2023-02-27:
Try removing the line `ExecStartPre = /sbin/modprobe uvcvideo` in the service file.

What is your distro ?

nikfilippas, 2023-02-27:
I re-configured it using your suggestion with `Type = forking` and commenting out the line `ExecStartPre`, and it's working again. Now, I have to wait a couple of days to see if the problem comes up on its own again.

I am using Ubuntu 22.04 LTS.

EmixamPP, 2023-02-27:
Theoretically, it should not be changed to fork, since it does not execute anything in background though a child process

Tips: you do not need to reconfigure to apply changes of the systemd service, just re-execute `linux-enable-ir-emitter boot enable`

nikfilippas, 2023-02-28:
**Update**: I reverted it back to `oneshot` and the problem persists after a few reboots. Same with `forking`. Here is my current service file:
```
[Unit]
Description = enable the infrared emitter
Requires =  dev-video2.device
After = multi-user.target suspend.target hybrid-sleep.target hibernate.target s>

[Service]
Type = oneshot
#ExecStartPre = /sbin/modprobe uvcvideo
ExecStart = /usr/bin/linux-enable-ir-emitter run

[Install]
WantedBy = multi-user.target suspend.target hybrid-sleep.target hibernate.targe>
```

What I will do now, is update to the newest version to see if the problem is still there.

nikfilippas, 2023-02-28:
I installed v4.5.0 from the tarball, after clearing every trace of `linux-enable-ir-emitter` from my system, and re-configured it.
`linux-enable-ir-emitter boot status` now returns:
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/lib/systemd/system/linux-enable-ir-emitter.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Tue 2023-02-28 15:34:00 EET; 2min 8s ago
    Process: 24665 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 24667 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 24667 (code=exited, status=0/SUCCESS)
        CPU: 28ms

Feb 28 15:34:00 canopus systemd[1]: Starting enable the infrared emitter...
Feb 28 15:34:00 canopus systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Feb 28 15:34:00 canopus systemd[1]: Finished enable the infrared emitter.
```

and my service file is
```
[Unit]
Description = enable the infrared emitter
After = multi-user.target suspend.target hybrid-sleep.target hibernate.target suspend-then-hibernate.target

[Service]
Type = oneshot
ExecStartPre = /sbin/modprobe uvcvideo
ExecStart = /usr/bin/linux-enable-ir-emitter run

[Install]
WantedBy = multi-user.target suspend.target hybrid-sleep.target hibernate.target
```

Let's see if updating from 4.1 to 4.5 fixes things.

EmixamPP, 2023-02-28:
> Version of linux-enable-ir-emitter: 4.1.5

OK, you said your version was in 4.5.0, but it wasn't. I hope the change I made in version 4.4.0 will solve your problem

EmixamPP, 2023-02-28:
> Version of linux-enable-ir-emitter: 4.1.5

OK! I didn't realize you were in version 4.1.5 and not 4.5.0
Yes, normally it should fix your issue

nikfilippas, 2023-03-01:
After a couple of reboots and cold starts it looks like the issue disappeared. Will be closing, and if it ever comes up I will be re-opening. Thanks for all your hard work @EmixamPP !

closed 2023-03-01 by nikfilippas