EmixamPP/linux-enable-ir-emitter #96: CRITICAL: Impossible to write the driver in /etc/linux-enable-ir-emitter/...

fufexan, 2023-03-07:
**Describe the bug**

The driver cannot be written to `/etc/linux-enable-ir-emitter`

```
 › sudo result/bin/linux-enable-ir-emitter -v -d /dev/video2 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
YeDEBUG: Computing the resolution control.
DEBUG: Minimum not consistent, it will be ignored.
DEBUG: current: 1 25
DEBUG: maximum: 2 100
DEBUG: resolution: 0 37
DEBUG: unit: 13 selector: 14
DEBUG: new current: 1 62
Is the ir emitter flashing (not just turn on) ? Yes/No ? Yes
Yes/No ? Yes
terminate called after throwing an instance of 'std::ios_base::failure[abi:cxx11]'
  what():  CRITICAL: Impossible to write the driver in /etc/linux-enable-ir-emitter//pci-0000:03:00.4-usb-0:4:1.2-video-index0_emitter0.driver: iostream error
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```

I have checked the perms of `/etc`, there should be no problem writing to it.
```
drwxr-xr-x - root  7 Mar 23:22  etc
```

**How to reproduce**

Run the above command.

**Additional info**
 - Computer (or camera) model: Lenovo Yoga Slim 7 13ACN5, `Bus 003 Device 003: ID 13d3:5419 IMC Networks Integrated Camera`
 - Version of linux-enable-ir-emitter: 4.5.0

EmixamPP, 2023-03-08:
What are the perms of /etc/linux-enable-ir-emitter ? 
Did you try a second time ?

fufexan, 2023-03-08:
> What are the perms of /etc/linux-enable-ir-emitter ? Did you try a second time ?

It does not exist. Shouldn't it be auto-created?
Yes, I've been running manual reconfigure as root every time I booted, ever since this started happening.

EmixamPP, 2023-03-08:
> It does not exist.

Ok, this is the problem. To fix your issue, you can create it  : `sudo mkdir /etc/linux-enable-ir-emitter`

How did you install linux-enable-ir-emitter ? 
Using the command `sudo tar -C / -h -xzf linux-enable-ir-emitter-*.tar.gz` ?
Because it should create this directory.

fufexan, 2023-03-08:
I wrote a [Nix package](https://github.com/NixOS/nixpkgs/pull/216245/files#diff-61851b65ca3c461d7bc8c13689a1038e1d59d4643d60eca244b3d7d13af10b41) that uses meson only to build `linux-enable-ir-emitter`. I will instruct the respective [NixOS module](https://github.com/NixOS/nixpkgs/pull/216245/files#diff-99b421611a6d4aad99b8a4226a165cb47dc59507ccee15b15c8d9c5d46832bc7) to create that directory then.

EmixamPP, 2023-03-08:
Lines 78-81 of the meson; it creates that directory when the install target is called:
```
install_subdir(
    meson.project_name(),
    install_dir: get_option('sysconfdir'),
)
```

fufexan, 2023-03-08:
> Lines 78-81 of the meson; it creates that directory when the install target is called:
> 
> ```
> install_subdir(
>     meson.project_name(),
>     install_dir: get_option('sysconfdir'),
> )
> ```

Right. That path gets generated inside the `linux-enable-ir-emitter` package, but it does not affect the global `/etc`. This is due to the way NixOS works.

Anyway, it's an easy fix. Thank you!

closed 2023-03-08 by fufexan

EmixamPP, 2023-03-08:
Ok ok, fine then

With pleasure!