EmixamPP/linux-enable-ir-emitter #97: Configuration works - Howdy works after running "sudo linux-enable-ir-emitter run" manualy. Not working after reboot

londeril, 2023-03-08:
**Output of `linux-enable-ir-emitter boot status`**
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/usr/lib/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: inactive (dead)

Mar 08 15:34:47 Nova systemd[1]: Dependency failed for enable the infrared emitter.
Mar 08 15:34:47 Nova systemd[1]: linux-enable-ir-emitter.service: Job linux-enable-ir-emitter.service/start failed with result 'dependency'.

```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
device=/dev/video2
unit=4
selector=6
size=9
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0

```

**Give more information if you have**
Everything works flawlessly until I reboot. Running "linux-enable-ir-emitter run" solves the problem - until I reboot.
linux-enable-ir-emitter boot enable was run and didn't give an error msg.
the failed dependency is weird... because it works if I run I do a manual run...

**Additional info**
 - Computer (or camera) model: Dell XPS 13 2-in-1 (9310)
 - Linux distribution: Arch Linux
 - Version of linux-enable-ir-emitter: linux-enable-ir-emitter 4.1.5 (from AUR https://aur.archlinux.org/packages/linux-enable-ir-emitter)

EmixamPP, 2023-03-08:
Similar issue to https://github.com/EmixamPP/linux-enable-ir-emitter/issues/95. You should install linux-enable-ir-emitter 4.5.0.  Please refer to the README (The AUR support has been dropped)

Or if you do not want to reinstall it (which I could quite understand), change the content of `/usr/lib/systemd/system/linux-enable-ir-emitter.service` with https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/sources/systemd/linux-enable-ir-emitter.service

londeril, 2023-03-08:
Yeah - I just saw that the AUR package is old... I reinstalled from your tarball. Works now! thanks!

closed 2023-03-08 by londeril