EmixamPP/linux-enable-ir-emitter #98: Bluetooth mouse stops working after applying the driver: Lenovo Yoga Slim 7 ProX

this-username-has-been-taken, 2023-03-28:
**Describe the bug**
After successful installation and reboot the wifi and bluetooth (mouse) stop working. Wifi connection will restore after manual reconnect (no need to delete the saved network or enter password again - just click connect button), but bluetooth mouse won't work. It will pair successfully, but has no control over the cursor. Unpairing/pairing again, deleting/creating connection, enabling/disabling adapter do not fix the issue. Deleting the driver (```sudo linux-enable-ir-emitter delete```) and rebooting do not fix the issue. The only way I found to fix the bug is to restore the system from backup (Timeshift) and reboot.

**How to reproduce**
1. Install the driver as described in the manual.
2. Reboot the PC.
3. Log in.
4. Observe the issue: wifi won't connect automatically, bluetooth mouse will connect, but will not work.

**Additional info**
 - Computer (or camera) model: Lenovo Yoga Slim 7 ProX 14IAH7 aka Yoga Slim 7i
 - Version of linux-enable-ir-emitter: 4.5.0
 - Linux distro: Kubuntu
 - Kernel version: 5.19.0-38-generic
 - KDE Plasma version: 5.27.3

EmixamPP, 2023-03-28:
Your issue is not related to linux-enable-ir-emitter, it does not execute anything directly and indirectly related to bluetooth or wifi. I am pretty sure that there is no correlation between these two events.

this-username-has-been-taken, 2023-03-28:
I have double checked:
1. Rebooted the notebook.
2. Made sure that wifi is connected and bluetooth mosue is working.
3. Made a Timeshift snapshot.
4. Installed the driver.
5. Rebooted the notebook again.
6. Got wifi and bluetooth issues just as described in the initial post.
7. Rolled back to the saved snapshot.
8. Rebooted.
9. Got wifi and bluetooth working again.

I didn't do anything except a Timeshift backup and installing the driver. I even didn't download it - used the previously saved archive.
It looks like the issue is related to the linux-enable-ir-emitter. Should I may be check the files that have been modified during/after install (how to do it)?

EmixamPP, 2023-03-29:
I'll investigate, but it's extremely strange, since linux-enable-ir-emitter only access the video device. The strange thing is that it breaks your wireless card only after the reboot.

I don't really want to make you play with the danger of not being able to reactivate your wifi and bluetooth. But if you feel confident with timeshit, you could try this to help me to debug your problem:
1. `sudo linux-enable-ir-emitter -v configure`
2. `sudo linux-enable-ir-emitter boot disable`
3. reboot

And give me the output of the two commands. The second command just disable to load the driver loading after boot. Like that, we can see if the issue is due to the boot service or something else.

this-username-has-been-taken, 2023-03-29:
Thank you for the reply! The issue indeed looks weird and not related to the linux-enable-ir-emitter utility. But it reproduces every time. I appreciate your willingness to help and dig into the issue.

I am totally OK with running tests or whatever helps to find the reason of such strange behavior.
Please find attached the output of the commands. After reboot the issue reappeared.

`user@LENOVO-nb:~$ sudo linux-enable-ir-emitter -v configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Computing the resolution control.
DEBUG: Minimum not consistent, it will be ignored.
DEBUG: current: 0 0
DEBUG: maximum: 2 100
DEBUG: resolution: 1 50
DEBUG: unit: 13 selector: 14
DEBUG: new current: 1 50
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 2 100
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 0
DEBUG: maximum: 1
DEBUG: minimum: 0
DEBUG: resolution: 1
DEBUG: unit: 13 selector: 16
DEBUG: new current: 1
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 1 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: maximum: 7 0 0 0 0 0 0 200 20 5 0 0 0 0 0
DEBUG: minimum: 0 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: resolution: 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0
DEBUG: unit: 14 selector: 2
DEBUG: new current: 0 0 0 0 0 0 0 16 39 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 7 0 0 0 0 0 0 200 20 5 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: current: 16 0 0 0 0 0 0 0 0 0 0
DEBUG: maximum: 16 0 0 0 0 0 0 232 3 0 0
DEBUG: minimum: 16 0 0 0 0 0 0 0 0 0 0
DEBUG: resolution: 16 0 0 0 0 0 0 0 0 0 0
DEBUG: unit: 14 selector: 3
DEBUG: new current: 16 0 0 0 0 0 0 232 3 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: Computing the resolution control.
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: resolution: 0 0 1 0 0 0 0 0 0
DEBUG: unit: 14 selector: 6
DEBUG: new current: 1 3 2 0 0 0 0 0 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? y
INFO: The driver has been successfully generated.
INFO: The boot service have been enabled.
user@LENOVO-nb:~$ sudo linux-enable-ir-emitter boot disable
INFO: The boot service have been disabled.`

I also tried to pair the mouse again and got the following errors in system log:
`29.03.2023 13:10	bluetoothd	src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
29.03.2023 13:10	bluetoothd	src/service.c:service_accept()  for D7:35:3E:1E:77:9A
29.03.2023 13:10	kernel	Bluetooth: hci0: Bad flag given (0x2) vs supported (0x1)
29.03.2023 13:10	bluetoothd	src/adapter.c:set_device_privacy_complete() Set device flags return status: Invalid Parameters
29.03.2023 13:16	bluetoothd	src/service.c:service_accept() input-hog profile accept failed for D7:35:3E:1E:77:9A
29.03.2023 13:16	bluetoothd	src/service.c:service_accept() input-hog profile accept failed for D7:35:3E:1E:77:9A`

After reverting to a previous snapshot the mouse paired without problems and the system log looked like this:
`29.03.2023 13:27	systemd	Reached target Bluetooth.
29.03.2023 13:27	bluetoothd	src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
29.03.2023 13:27	bluetoothd	src/adapter.c:set_device_privacy_complete() Set device flags return status: Invalid Parameters
29.03.2023 13:27	kernel	Bluetooth: hci0: Bad flag given (0x2) vs supported (0x1)
29.03.2023 13:27	kernel	hid-generic 0005:2B89:6621.0005: input,hidraw2: BLUETOOTH HID v0.01 Mouse [UGREEN BLE Mouse] on 3c:e9:f7:30:7c:16`

this-username-has-been-taken, 2023-03-29:
An update to the issue.
Looks like it is not related to the linux-enable-ir-emitter utility directly, but to the emitter itself.

I found a few posts with possible solutions and wanted to try them.
After reverting to the previous snapshot I ran the linux-enable-ir-emitter again and my IR diode started to flash immediately. The utility said that the emitter is already configured and there is nothing to do.
After reboot the mouse stopped working again.
It looks like not the utility causes the issue, but the first IR-diode run. Which is sad because it looks like the emitter is now configured and any software which enables it will cause the mouse to stop working after reboot.
Suggested solutions (changing parameters in the /etc/bluetooth/input.conf file) didn't work either.

EmixamPP, 2023-04-22:
Sorry for the late reply, unfortunately I can't do anything for you :/. I close the issue

closed 2023-04-22 by EmixamPP