EmixamPP/linux-enable-ir-emitter #99: Install command issue

dingwen07, 2023-04-14:
**Describe the bug**

The install command 

```
sudo tar -C / -h -xzf linux-enable-ir-emitter-*.tar.gz
```

incorrectly set several system file owner to the current user including:

```
/
/etc
/usr
...

```

Basically all files and directories in the path  of those files in the zip file will be changed.

**How to reproduce**

Execute install command 

```
sudo tar -C / -h -xzf linux-enable-ir-emitter-*.tar.gz
```

**Additional info**
 - Computer (or camera) model: 
 - Version of linux-enable-ir-emitter:

EmixamPP, 2023-04-14:
When executing the command, the current user should be root, because of the sudo command.

I have not observed this bug myself

EmixamPP, 2023-04-14:
Thank you so much for pointing me this critical bug ! 

This can be avoided by using the option `--no-same-owner` : `sudo tar -C / --no-same-owner -h -xzf linux-enable-ir-emitter-*.tar.gz`

I've updated the readme. I didn't have this problem, because I go directly through meson, of course ... I should have thought about it.

Thank you again for your vigilance

closed 2023-04-14 by EmixamPP