EmixamPP/linux-enable-ir-emitter #101: Stuck while configuring and stops working after sleep on Lenovo YOGA14s 2021 ARH

AkinoKaede, 2023-05-01:
**Describe the bug**

After the program successfully generated driver, the configure program didn't stop.
So I just kill the program by  pressing `Control+C`, and it works even I reboot my laptop.

But when the laptop wake up from S3 Sleep State, the ir emitter cannot work correctly. I must use  `sudo linux-enable-ir-emitter run` to re run it.

**How to reproduce**

```
> sudo linux-enable-ir-emitter -v -d /dev/video2 configure                      
    
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? No 
DEBUG: Computing the resolution control.
DEBUG: Minimum not consistent, it will be ignored.
DEBUG: current: 0 0
DEBUG: maximum: 2 100
DEBUG: resolution: 1 50
DEBUG: unit: 13 selector: 14
DEBUG: new current: 1 50
Is the ir emitter flashing (not just turn on) ? Yes/No ? Yes
INFO: The driver has been successfully generated.
```

**Additional info**
 - Computer (or camera) model: `Bus 001 Device 002: ID 13d3:5419 IMC Networks Integrated Camera`
 - Version of linux-enable-ir-emitter: `4.5.0`

Status of systemd service:

```
linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/usr/lib/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: inactive (dead)
```

AkinoKaede, 2023-05-01:
It seems it is caused by systemd, systemd cannot start linux-enable-ir-emitter.service.

EmixamPP, 2023-05-01:
Unfortunately, I really don't have time to look at that now, it will only be in July

AkinoKaede, 2023-05-01:
> Unfortunately, I really don't have time to look at that now, it will only be in July

It doesn't matter, I will try my best to solve the problem by myself.

AkinoKaede, 2023-05-01:
For workaround, I create and script:

```
# > /lib/systemd/system-sleep/linux-enable-ir-emitter

#!/bin/sh

case $1 in
    post) /usr/bin/linux-enable-ir-emitter run  ;;
esac
```

It's works for me.

EmixamPP, 2023-05-05:
What is your distro ? 

And could you please execute: `sudo /sbin/modprobe uvcvideo`. Does the command terminate ?

AkinoKaede, 2023-05-05:
> What is your distro ?
> 
> And could you please execute: `sudo /sbin/modprobe uvcvideo`. Does the command terminate ?

Hmm, I am using Arch Linux, the command are not terminated.
Actually, I tried to remove ExecStartPre from service, but it still terminate.

EmixamPP, 2023-05-05:
So even without `ExecStartPre = /sbin/modprobe uvcvideo`  in `linux-enable-ir-emitter.service`, the command `sudo linux-enable-ir-emitter boot enable` command still hangs?

AkinoKaede, 2023-05-05:
> sudo linux-enable-ir-emitter boot enable

Yes

AkinoKaede, 2023-06-23:
> I've tried something in #115. In the PR there is a comment with a link to download the tarball that contains the (possible) fix.
> 
> Could you please test this version? Thank you

systemd hangs is fixed, but it not start after suspend.

AkinoKaede, 2023-06-23:
> > I've tried something in #115. In the PR there is a comment with a link to download the tarball that contains the (possible) fix.
> > Could you please test this version? Thank you
> 
> systemd hangs is fixed, but it not start after suspend.

I deleted `multi-user.target` in `After`, it is fixed.

```
[Unit]
Description = enable the infrared emitter
After = suspend.target hybrid-sleep.target hibernate.target suspend-then-hibernate.target

[Service]
Type = oneshot
ExecStartPre = /sbin/modprobe uvcvideo
ExecStart = /usr/bin/linux-enable-ir-emitter run

[Install]
WantedBy = suspend.target hybrid-sleep.target hibernate.target suspend-then-hibernate.target
```

EmixamPP, 2023-06-23:
Could you also test this one : 
```
[Unit]
Description = enable the infrared emitter
After=suspend.target sleep.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target

[Service]
Type = oneshot
ExecStartPre = /sbin/modprobe uvcvideo
ExecStart = /usr/bin/linux-enable-ir-emitter run

[Install]
WantedBy=multi-user.target
```
It is more "correct"

AkinoKaede, 2023-06-23:
> That makes sens, there were a dependency loop, I think.
> 
> Could you also test this one please :
> 
> ```
> [Unit]
> Description = enable the infrared emitter
> After=suspend.target sleep.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
> 
> [Service]
> Type = oneshot
> ExecStartPre = /sbin/modprobe uvcvideo
> ExecStart = /usr/bin/linux-enable-ir-emitter run
> 
> [Install]
> WantedBy=multi-user.target
> ```
> 
> It is more "correct"

It not works.

AkinoKaede, 2023-06-23:
I add `multi-user.target` to `WantedBy`, I think it is correct.

```
[Unit]
Description = enable the infrared emitter
After = suspend.target hybrid-sleep.target hibernate.target suspend-then-hibernate.target

[Service]
Type = oneshot
ExecStartPre = /sbin/modprobe uvcvideo
ExecStart = /usr/bin/linux-enable-ir-emitter run

[Install]
WantedBy = multi-user.target suspend.target sleep.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
```

EmixamPP, 2023-06-23:
So it was hanging because `multi-user.target` was also in `After=`

Great! Thank you so much for your contribution to the project!

EmixamPP, 2023-06-23:
Also, I see that you removed `sleep.target` from `After=` is this on purpose?

AkinoKaede, 2023-06-23:
> Also, I see that you removed `sleep.target` from `After=` is this on purpose?

This does not appear to be in the `After`.

<https://github.com/EmixamPP/linux-enable-ir-emitter/blob/7eb24b4fe65328c2520ac169c41825e9a96c9273/sources/boot_service/systemd/linux-enable-ir-emitter.service>

AkinoKaede, 2023-06-23:
I removed udev rule and reboot my laptop, it works.

EmixamPP, 2023-06-23:
> This does not appear to be in the After.

Ok mistakes from my part. 

> I removed udev rule and reboot my laptop, it works.

Ok then, I think I can merge the fix

Thanks again :smile:

AkinoKaede, 2023-06-23:
You are welcome.

closed 2023-06-23 by EmixamPP

AkinoKaede, 2023-07-05:
I found out the reason, it was caused by the udev process being stuck, which was caused by the Linux kernel, linux-enable-ir-emitter was fine. Changes should probably be revert.