EmixamPP/linux-enable-ir-emitter #102: IR Camera appearing as different device and not working after using tool

AndyKolski, 2023-05-08:
**Ouput of `linux-enable-ir-emitter -v -d /dev/video0 configure`**
<details><summary>configure command output</summary>

```
$ linux-enable-ir-emitter -v -d /dev/video0 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
CRITICAL: Cannot access to /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
$ v4l2-ctl --list-devices
SPCA2085 PC Camera: PC Camera (usb-0000:00:14.0-5):
        /dev/video0
        /dev/video1
        /dev/media0

Integrated Camera: Integrated C (usb-0000:00:14.0-8):
        /dev/video2
        /dev/video3
        /dev/media1
```

**Ouput of `v4l2-ctl -d /dev/video0 --list-formats-ext`**
<details><summary>v4l2 device output</summary>

```
$ v4l2-ctl -d /dev/video0 --list-formats-ext                                                                                                                               
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Additional info**
 - Computer (or camera) model: Integrated IR camera in Lenovo Thinkpad X1 Carbon Gen 6
 - Version of linux-enable-ir-emitter: 
 ```
$ linux-enable-ir-emitter --version
linux-enable-ir-emitter 4.5.0
Developped by Maxime Dirksen - EmixamPP
MIT License
```

The IR camera previously appeared in lsusb as the following:
```
$ lsusb
[...]
Bus 001 Device 004: ID 04f2:b614 Chicony Electronics Co., Ltd Integrated Camera (The visible light camera)
[...]
Bus 001 Device 002: ID 04f2:b615 Chicony Electronics Co., Ltd Integrated IR Camera (The IR camera, with its original appearance) 
[...]
```
Since running the script, it has started appearing as the following:
```
$ lsusb
[...]
Bus 001 Device 004: ID 04f2:b614 Chicony Electronics Co., Ltd Integrated Camera (The visible light camera, unchanged)
[...]
Bus 001 Device 002: ID 1bcf:0b09 Sunplus Innovation Technology Inc. SPCA2085 PC Camera (The IR camera, with its new appearance)
[...]
```

Since the change, the IR camera has stopped working in Windows 10, and while it is detected in Linux, it only reports being capable of one resolution, frame rate, and format, and it does not produce any frames, although the IR emitters are on solid. I have shut off the laptop entirely multiple times, for prolonged periods, and it has made no difference.

The first time I ran the configure command, I used the command `linux-enable-ir-emitter -v -d /dev/video0 configure -e 2`, as the camera has an emitter on each side of the lens. That time it got farther - it repeatedly asked if the emitters were flashing, and since they were staying off, I answered "No". After a few iterations, it told me to shutdown and restart my computer, then try again, so I did that, and it has not been able to get that far again. Unfortunately, I do not have the output from the first time I ran the command.

Note that I have also tried using some of the scripts and C programs from [https://github.com/boltgolt/howdy/issues/269](https://github.com/boltgolt/howdy/issues/269) and [https://github.com/PetePriority/chicony-ir-toggle](https://github.com/PetePriority/chicony-ir-toggle) to no effect. The ioctl on /dev/video0 always fails with errno=2, "No such file or directory", despite the file existing, and my user having permission to access it. I also did make sure I correctly changed the code to point to the right file

At this point I have no interest in getting the IR emitters to stay on (as opposed to flashing as it did by default), I would be happy to have basic functionality again.

Thank you!

EmixamPP, 2023-05-11:
Hum, is it possible to download the camera driver on the support page of your laptop which contains the drivers on the vendor's website? If so, since you have a Windows dual boot, you could run it to reset the configuration.

EmixamPP, 2023-05-11:
If you feel able, one possibility that could reset the configuration of the camera, is to open the laptop and disconnect the battery. 

Once the camera is reset, if you haven't uninstalled `linux-enable-ir-emitter`, you could try to configure the camera again, as the tool has learned what made your camera crash. 

To be sure, you can see if there is a `/etc/linux-enable-ir-emitter/pci*.exclude` file that contains anything.

But I can also understand why you might be afraid to try again.

Sorry it took me a while to reply, unfortunately I am very busy during this period. But I will try to help you and get your camera working again.

EmixamPP, 2023-05-11:
I looked up the camera driver for you on your laptop's support page: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-6th-gen-type-20kh-20kg/downloads/driver-list/component?name=Camera%20and%20Card%20Reader

Or start the download directly using this link: https://download.lenovo.com/pccbbs/mobiles/n22ci10w.exe

EmixamPP, 2023-06-21:
Feel free to comment if you wish to re-open the issue.

closed 2023-06-21 by EmixamPP