EmixamPP/linux-enable-ir-emitter #103: Configuration Fails With Critical Error (Lenovo Thinkpad x1 Yoga gen 3)

moss-cf, 2023-05-20:
**Ouput of `linux-enable-ir-emitter -v -d /dev/video1 configure`**

<details><summary>configure command output</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
CRITICAL: Cannot access to /dev/v4l/by-path/pci-0000:00:14.0-usb-0:8:1.0-video-index1
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki

```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Integrated Camera: Integrated C (usb-0000:00:14.0-8):
        /dev/video0
        /dev/video1
        /dev/media0

```

**Ouput of `v4l2-ctl -d /dev/video1 --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

```
</details>

**Additional info**
 - Computer (or camera) model: Lenovo Thinkpad x1 Yoga gen 3
 - Version of linux-enable-ir-emitter:

EmixamPP, 2023-05-22:
It seems that your infrared camera is not recognized by the kernel. It is not listed in `v4l2-ctl --list-devices`, the one you see is the regular one

EmixamPP, 2023-06-21:
Closing because we can't do anything until this infrared camera is recognized by the kernel :confused:

closed 2023-06-21 by EmixamPP